bluekitchen/btstack #293: Should we change the function call "sm_passkey_used" to "sm_passkey_entry"?

LeslieChe, 2020-06-16:
https://github.com/bluekitchen/btstack/blob/651e51d4f5313fe6a2ba7626bf49cc641c75b0f9/src/ble/sm.c#L3800-L3804

According to the comment, the judgment condition should be "JUST WORK/NC in initiator role" or "**passkey entry**".
The Line 3801, should we change the function call "sm_passkey_used" to "sm_passkey_entry"?

My test environment:
local device is "STM32F429 + CC2564", remote device is my phone.
STM32F429 is Responder, displays passkey.
phone is initiator, inputs passkey.

When I Check the log of STM32F429, I can not find the operation of CHECK_CONFIRMATION,
but According to Bluetooth protocol, there should be CHECK_CONFIRMATION operation.

when I change the function call "sm_passkey_used" to "sm_passkey_entry",there will be CHECK_CONFIRMATION operation.

Thank you.

mringwal, 2020-06-19:
Hi! Thanks for reporting. What problem do you see? Does pairing fail for you, or, is the confirmation value not checked?

LeslieChe, 2020-06-20:
> Hi! Thanks for reporting. What problem do you see? Does pairing fail for you, or, is the confirmation value not checked?

Pairing is successful，but the confirmation value is not checked，that is，ignore the check on the confirmation value

mringwal, 2020-06-22:
Thanks. You're right. The comment really was right. It's fixed on develop branch now and will be merged into master beginning of July.

closed 2020-06-22 by mringwal