bluekitchen/btstack #294: Segmentation fault

CSparn, 2020-06-19:
It occurs a segmentation fault while running/debugging example mesh_node_demo.exe with bluetooth dongle (Asus USB-BT400) in port/windows-winusb. 

The error is in mesh.c line 1084
(`    uint32_t prov_len = btstack_tlv_singleton_impl->get_tag(btstack_tlv_singleton_context, mesh_tag_for_prov_data, (uint8_t *) &persistent_provisioning_data, sizeof(mesh_persistent_provisioning_data_t));
`). 

**To Reproduce**
Steps to reproduce the behavior:
1. Follow Readme.md in port/windows-winusb
1.1 Use Zadic to replace USB driver
1.2 Install Msys2 + MinGW64 
1.3 Make port/windows-winusb with Msys2 + MinGw64

2. Run/debug mesh_node_demo.exe in port/windows-winusb

3. Program crashes before console input/ Debugging shows segmentation fault

**HCI Packet Logs**
https://file.io/JEevooHs

**Environment:  (please complete the following information):**
 - Current BTstack branch: master
 - Bluetooth Controller [Asus USB-BT400]

mringwal, 2020-06-22:
Thanks for reporting. The mesh code uses the BTstack TLV storage, however, it is not setup for the Windows ports. I'll try to update them  soon (similar to libusb or posix-h* ports)

mringwal, 2020-06-22:
Done. Please switch to latest version on develop and try again. It doesn't seem to crash :)
There's a major rewrite pending on mesh-message, but we didn't get around to merge. Feel  free to play with that as well.

CSparn, 2020-06-23:
Wow that was fast. Thank you it is working now:)

closed 2020-06-23 by CSparn

mringwal, 2020-06-23:
Thanks for testing. It was just that the windows ports did use the older implementations to store link key and LE pairing data, which have been replaced by a generic storage approach (but haven't been updated to use them).