bluekitchen/btstack #299: wii gamepads stop working after commit

ricardoquesada, 2020-07-05:
**Describe the bug**
- Wii gamepads (Wii Mote / Wii U Controller Pro) are failing to establish a connection
- It seems that the culprit commit is https://github.com/bluekitchen/btstack/commit/bbc80a48cd3e8446adbd88810be85e4051af78dc

**To Reproduce**
- I tested using this project: https://gitlab.com/ricardoquesada/unijoysticle2 using both ESP32 and libusb ports.
- I tested using Wii Mote and a Wii U Controller Pro.
-Works Ok with other controllers like DS4, Xbox One S, Android, Nintendo Switch, iOS (only fails with Wii).

**Expected behavior**
- Wii Mote / Wii U Controller Pro should connect.

**HCI Packet Logs**
attached hci_dump.pklg.zip

**Environment:  (please complete the following information):**
Tested on both with ESP32 v4.0 / v4.1-beta2
Tested with libusb

**Additional context**
used to work fine until commit bbc80a48cd3e8446adbd88810be85e4051af78dc

ricardoquesada, 2020-07-05:
Logs attached in this msg:

[hci_dump.pklg.zip](https://github.com/bluekitchen/btstack/files/4874327/hci_dump.pklg.zip)

mringwal, 2020-07-06:
Hi. The issue here is that pairing with the BD_ADDR fails. The commit fixes a bug where BTstack would not enforce authentication in some scenarios. You may call gap_set_security_level(0) to disable security for testing.

mringwal, 2020-07-06:
What do you mean with 'breaks the rest'?

The correct fix would be get the pairing to work. Maybe you need to flip the BD ADDR (once more)?

gap_security_level is consulted for all non-SDP L2CAP connections, both incoming and outgoing. The outgoing level is checked just before the connection is opened.

ricardoquesada, 2020-07-06:
> What do you mean with 'breaks the rest'

I meant, it breaks the rest of the gamepads. I tried it with the Nintendo Switch controller, and if security==level_0, it fails to connect. So I was wondering if there is "correct" place to set "security==level0" that doesn't affect the non Wii gamepads.

The use case that I have is:  2 gamepads can be connected at the same time. You could have a Nintendo Wii and a Nintendo Switch Pro connected at the same time.

I'm trying to figure out how to have both of them running again (it was working before https://github.com/bluekitchen/btstack/commit/bbc80a48cd3e8446adbd88810be85e4051af78dc ).

Many thanks.

mringwal, 2020-07-07:
Can you post a log from before with a working outgoing connection to Wii Controller and a log that shows Nintendo Switch controller failing with security = level 0? I still think that the correct solution is to fix pairing with Wii Controller.

ricardoquesada, 2020-07-09:
Here are the logs:

* nintendo_switch.bad  (security level == 0)
* nintendo_switch.ok    (without security level)
* nintendo_wii.bad       (without security level)
* nintendo_wii.ok         (security level == 0)

Interesting is that DualShock4 works ok both with and without `gap_set_security_level(0)`.

Thanks!
[nintendo_controllers_logs.zip](https://github.com/bluekitchen/btstack/files/4897644/nintendo_controllers_logs.zip)

mringwal, 2020-07-13:
thanks for the logs. some comments:
- nintendo_switch.bad (security level == 0): the controller disconnects us as we try to open HID Control while sec leve 0 (encryption not actitve)
- nintendo_switch.ok (security level default): works as expected
- nintendo_wii.ok (security level == 0): works as Wii Remote don't mind no encryption
- nintendo_wo_bad (security level default. Pairing fails as mentioned before. I can see that your code uses the BD_ADDR in reverse as described [https://wiibrew.org/wiki/Wiimote#Bluetooth_Pairing](here). Do you start pairing by holding 1+2 down?

again, if we're able to get Wii Mote pairing to work, your code will work correctly with all controllers.

ricardoquesada, 2020-07-13:
Thanks for the answer. I'm not using "1+2". Instead I'm using sync. From the wiki:

> The newer Wiimote RVL-CNT-01-TR shuts down immediately upon receiving any HID output report if it has been turned on using the 1 + 2 method, although it works using the sync button. It is possible that authentication is now mandatory for the 1 + 2 temporary sync[check].

> If connecting by holding down the 1+2 buttons, the PIN is the bluetooth address of the wiimote backwards, if connecting by pressing the "sync" button on the back of the wiimote, then the PIN is the bluetooth address of the host backwards.

I'll double check everything again. But I know this used to work before. Thanks.

mringwal, 2020-07-14:
Thanks for pasting the right snippet. Please try to use the reverse BD_ADDR of the host (i.e., the ESP32) when connecting using the "sync" button.

Before, it worked as a) BTstack failed to start pairing and b) the Wiimote allowed HID without authentication.

ricardoquesada, 2020-07-14:
Thanks Matthias. 
There was an error in my code. I was using the WiiMote reversed address, instead of the ESP32 reversed-address as the PIN.
I fixed it. But I'm still having the same issue.

I guess I might be missing setting something (?)

The Auth PIN code is pretty simple, and the logs mention that Auth is Ok:
```c
        case HCI_EVENT_PIN_CODE_REQUEST: {
          bd_addr_t pin_code;
          bd_addr_t local_addr;
          hci_event_pin_code_request_get_bd_addr(packet, event_addr);
          gap_local_bd_addr(local_addr);
          reverse_bd_addr(local_addr, pin_code);
          hci_send_cmd(&hci_pin_code_request_reply, event_addr,
                       sizeof(pin_code), pin_code);
          break;
        }
```

But after a successful PIN Auth, I'm still getting: `l2cap.c.2398: channel 0x17b2640, cid 0041 - state 6: actual 0 >= required 2?`
and the connection is closed.

 - I tried both with WiiMote and Wii U Pro.
 - Using btstack master as of today
 - tried using both ESP32-IDF 4.1-beta2 and using libusb

Attached the logs of the Wii U controller.

any idea what am I possible missing? Thanks!

[hci_dump-nintendo_wii_u_pro.pklg.zip](https://github.com/bluekitchen/btstack/files/4919588/hci_dump-nintendo_wii_u_pro.pklg.zip)

mringwal, 2020-07-15:
Getting closer. There was one more log message missing for you to understand. After the encryption, BTstack checks the amount of entropy used, which can be 7-16. I thought everything below 16 is suspicious and have set that as minimum. There's another issue open about this. 

please call gap_set_required_encryption_key_size(7) during init and try again.

ricardoquesada, 2020-07-15:
Thanks Matthias!

That worked. Many thanks!

closed 2020-07-15 by ricardoquesada

mringwal, 2020-07-15:
You're welcome. I've also set 7 as default on develop, so you may remove that line eventually (although it doesn't hurt)