bluekitchen/btstack #300: Memory leak

LeslieChe, 2020-07-05:
https://github.com/bluekitchen/btstack/blob/651e51d4f5313fe6a2ba7626bf49cc641c75b0f9/src/btstack_memory.c#L916-L925

When I tried to connect my device to Aligenie through Mesh, I Found that the function **btstack_memory_mesh_network_pdu_get** always returns NULL.

I think that memory leak has occurred in **mesh_network_pdu_pool**. That is, memory is allocated in some places, but forget to release. No matter how big the mesh_network_pdu_pool is, it will eventually be exhausted.

But I can't find where the memory should be released.

Thank you!

mringwal, 2020-07-06:
Hi. Thanks fort the feedback. What's a 'Aligenie'?

We did a major rewrite of the Mesh stack on the 'mesh-message' branch, but it hasn't been merged onto develop yet.
Could you try this branch?

LeslieChe, 2020-07-06:
> Hi. Thanks fort the feedback. What's a 'Aligenie'?
> 
> We did a major rewrite of the Mesh stack on the 'mesh-message' branch, but it hasn't been merged onto develop yet.
> Could you try this branch?

AliGenie is an intelligent voice assistant speaker developed by Alibaba Artificial Intelligence Laboratory，similar to Amazon Echo, Google Home, Apple HomePod.

I will try the 'mesh-message' branch. Thank you very much!

LeslieChe, 2020-07-12:
I have tried.

In the mesh-message branch, memory leaks still exist in mesh_network_pdu_pool.

When mesh_network_pdu_pool is exhausted, the function of turning on and off the lights will be invalid.

mringwal, 2020-07-13:
Could you describe what you're doing to trigger this memory leak such that we would be able to reproduce it?

BTstack generally tries to avoid buffers as good as possible and for both Classic and LE, there aren't any packet buffers (only structs to track connection state), and such problems cannot arise.

In Mesh, that's not possible, as mesh network pdus are passed over multiple layers up and down.

LeslieChe, 2020-07-19:
Software： 
    mesh_node_demo.c
Hardware:   
    STM32F429 + CC2564, as device
    Tmall Genie, as a provisioner

First, tell Tmall Genie to discover the device. Then, Tmall Genie replied:  the light is found, do you want to connect?

And I Tell Tmall Genie, connect with the light. At this time, provisioning starts, Tmall Genie invites the device to join the network.

You can see the entire process of the provisioning from the log, and then there will be network layer packets interaction.

Then, Tmall Genie reports that the connection is successful.

At this time, you can use Tmall Genie to control the node to turn on or off the lights.

I added the allocation and release information of the memory pool in the code, such as

```c
#ifdef MAX_NR_MESH_NETWORK_PDUS
int mesh_network_pdu_storage_get = 0; // added by me
int mesh_network_pdu_storage_free = 0; // added by me
#if MAX_NR_MESH_NETWORK_PDUS > 0
static mesh_network_pdu_t mesh_network_pdu_storage[MAX_NR_MESH_NETWORK_PDUS];
static btstack_memory_pool_t mesh_network_pdu_pool;
mesh_network_pdu_t * btstack_memory_mesh_network_pdu_get(void){
    void * buffer = btstack_memory_pool_get(&mesh_network_pdu_pool);
    if (buffer){
        mesh_network_pdu_storage_get++; // added by me
        memset(buffer, 0, sizeof(mesh_network_pdu_t));
    }
    return (mesh_network_pdu_t *) buffer;
}
void btstack_memory_mesh_network_pdu_free(mesh_network_pdu_t *mesh_network_pdu){
    mesh_network_pdu_storage_free++; // added by me
    btstack_memory_pool_free(&mesh_network_pdu_pool, mesh_network_pdu);
}
#else

```

Every time I turn on or turn off the light, I will print the memory pool information,  the code is:

```c
 static void mesh_state_update_message_handler(uint8_t packet_type, uint16_t channel, uint8_t *packet, uint16_t size){
    UNUSED(channel);
    UNUSED(size);
    int state;

    if (packet_type != HCI_EVENT_PACKET) return;
   
    switch(packet[0]){
        case HCI_EVENT_MESH_META:
            switch(packet[2]){
                case MESH_SUBEVENT_STATE_UPDATE_BOOL:
                    
                    state =  mesh_subevent_state_update_bool_get_value(packet);
                    if(state)
                        led_on();
                    else
                        led_off();

                    printf("State update: %u\n",state);
                    int memory_leak = mesh_network_pdu_storage_get - mesh_network_pdu_storage_free;
                    printf("Memory Leak: %u / %u\n",memory_leak,MAX_NR_MESH_NETWORK_PDUS);
                    
                    break;
                default:
                    break;
            }
            break;
        default:
            break;
    }
}
```		

The log is:

>Transport: received message. SRC 89, SEQ 2c1a
Upper Transport pdu  0x80 0x8C 0x64 0x74 0x98 0x5D 
AKF: 1
AID: 31
KEY_INIT: dst 195b, akf 1, aid 31
AppNonce             0x01 0x00 0x00 0x2C 0x1A 0x00 0x89 0x19 0x5B 0x00 0x00 0x00 0x00 
AppOrDevKey          0x39 0x47 0x4E 0x3B 0xBC 0xA9 0x05 0xEE 0xBD 0x6B 0x90 0x06 0x45 0xC9 0x34 0xF9 
EncAccessPayload     0x80 0x8C 0x64 0x74 0x98 0x5D 
Decrypted PDU        0x82 0x02 0x01 0x13 0x41 0x00 
TransMIC             0x2D 0xD1 0x27 0x04 
TransMIC matches
MESH Access Message, Opcode = 8202: 0x82 0x02 0x01 0x13 0x41 0x00 
[+] Upper transport, send upper (un)segmented Access PDU - dest 0089, seq 00000c
Access Payload       0x82 0x04 0x00 0x01 0x41 
AppNonce             0x01 0x00 0x00 0x00 0x0C 0x19 0x5B 0x00 0x89 0x00 0x00 0x00 0x00 
AppOrDevKey          0x39 0x47 0x4E 0x3B 0xBC 0xA9 0x05 0xEE 0xBD 0x6B 0x90 0x06 0x45 0xC9 0x34 0xF9 
EncAccessPayload     0xD1 0xCC 0x9F 0xC8 0xE9 
TransMIC             0xFC 0x99 0x29 0x31 
UpperTransportPDU    0xD1 0xCC 0x9F 0xC8 0xE9 0xFC 0x99 0x29 0x31 
[+] Lower transport, unsegmented pdu, sending now 20000c3c
[00:01:59.153] LOG -- adv_bearer.c.190: can send now
[00:01:59.158] LOG -- adv_bearer.c.292: adv bearer message, type 0x2a
[00:01:59.165] LOG -- adv_bearer.c.269: Send ADV Bearer message
[00:01:59.272] LOG -- adv_bearer.c.216: Timeout (state bearer)
[00:01:59.277] LOG -- adv_bearer.c.269: Send ADV Bearer message
[00:01:59.385] LOG -- adv_bearer.c.216: Timeout (state bearer)
[00:01:59.390] LOG -- adv_bearer.c.269: Send ADV Bearer message
[00:01:59.498] LOG -- adv_bearer.c.216: Timeout (state bearer)
State update: 0
Memory Leak: 17 / 26
State update: 1
Memory Leak: 17 / 26

Sometimes，I paused the program through the debugger and added the statistical results to the watch windows of KEIL, as shown below：
[image]

You will see that 15 (0x99-0x8A=0xF) of the mesh_network_pdu have not been released.

Thank you!

mringwal, 2020-07-20:
Thanks for the detailed report. If you toggle the light multiple times, does the the number of allocated but not freed, aka leaked, blocks increase?

I think somewhere in the mesh access layer, the stack buffers enough network pdus to be able to store a max size transport packet. 15 packets might be about the right amount of packets for this.

Do you get the same behaviour if you use one of the Android or iOS Mesh apps from Nordic, Dialog Semi, Silicon Labs or ST Micro?

LeslieChe, 2020-08-02:
> If you toggle the light multiple times, does the the number of allocated but not freed, aka leaked, blocks increase?

I toggle the light multiple times, about 108 times, and plotted the memory usage as the following figure.
The x-axis is the number of operations, and the y-axis is the number of memory occupied.
As you can see, the memory usage varies from 16 to 20.

[image]

mringwal, 2020-08-03:
Thanks for the nice plot. Does it show the current number of allocated network pdus (as it is always going back to 16, this would mean there is no leak), or does it show the number of lost/leaked network pdus for each light toggle?

LeslieChe, 2020-08-04:
Hi,
it shows the current number of allocated network pdus.
Yes, it is always going back to 16.
You mean there is no memory leak? In other words, these 16 pdus are used as buffers and will not be released?

mringwal, 2020-08-05:
Hi. In the current version on mesh-message branch, the mesh upper transport layer tries to reserve MESSAGE_BUILDER_MAX_NUM_NETWORK_PDUS (14) network pdus to be able to build a max size outgoing access message. If the number of allocated network pdus does not go up, it is not leaking. (if it would leak/loose pdus, the number would not go back to 16).

Initially, you've reported that the version on develop/master would run out of buffers even if you provide 100+ buffers. So, it looks like there was a leak in the old code, but at least for your test setup, the memory allocation works correct on mesh-message-branch. Does this make sense?

LeslieChe, 2020-08-05:
Thanks, I understand. I think this issue can be closed.

closed 2020-08-05 by LeslieChe

mringwal, 2020-08-06:
Anyway, I'm glad you can use our Mesh implementation for your device although its not complete. I'll merge the mesh-messages onto the develop branch.