bluekitchen/btstack #301: The mouse pointer on the Thinkpad cannot move

LeslieChe, 2020-07-06:
Hi！

Hardware : my device(STM32 + CC2564) and Thinkpad
Software : hog_mouse_demo.c

Test result: My device and Thinkpad can pair successfully, and the Thinkpad  shows that the mouse is connected.

But when I send "a, s, d, w" to my device through  serial port, the mouse pointer on the Thinkpad does not move.

When I connect my device with some phones (mobile phone brand: Xiaomi or Huawei), the pointer appears on the phone, the movement of the pointer is normal.

Is there a problem with the BTstack code, or is there a problem with my Thinkpad driver?
If there is a problem with my Thinkpad, but the Microsoft mouse can work normally.

Thank you！

The following is a partial log of the connection between my device and ThinkPad.

>[00:00:39.916] LOG -- att_db.c.328: att_validate_security. flags 0xf10f (=> security level 1, key size 16) authorized 0, authenticated 1, encryption_key_size 16, secure connection 1
[00:00:39.933] LOG -- att_server.c.851: Store CCC value 0x0001 for handle 0x0020 of remote E8:D0:FC:D8:C3:00, le device id 3
[00:00:39.970] LOG -- btstack_tlv_flash_bank.c.332: Found tag '4254430e' at position 14444
[00:00:39.981] LOG -- btstack_tlv_flash_bank.c.332: Found tag '4254430f' at position 13332
[00:00:39.991] LOG -- btstack_tlv_flash_bank.c.332: Found tag '42544310' at position 13316
[00:00:40.001] LOG -- btstack_tlv_flash_bank.c.332: Found tag '42544311' at position 12760
[00:00:40.011] LOG -- btstack_tlv_flash_bank.c.332: Found tag '42544312' at position 12744
[00:00:40.021] LOG -- btstack_tlv_flash_bank.c.332: Found tag '42544313' at position 12728
[00:00:40.030] LOG -- att_server.c.913: tag_for_empty 4254430d, tag_for_lowest_seq_nr 42544313
[00:00:40.039] LOG -- btstack_tlv_flash_bank.c.377: write '4254430d', len 8 at 14460
[00:00:40.049] LOG -- hids_device.c.232: Enable Report Input notifications: 1
Report Characteristic Subscribed 1
[00:00:40.060] LOG -- l2cap.c.980: l2cap_send_prepared_connectionless handle 1025, cid 0x04
[00:00:40.146] LOG -- att_server.c.831: ATT Packet, handle 0x0401
[00:00:40.153] LOG -- l2cap.c.1005: L2CAP_EVENT_CHANNEL_CAN_SEND_NOW local_cid 0x4
[00:00:40.161] LOG -- att_db.c.328: att_validate_security. flags 0xf10f (=> security level 1, key size 16) authorized 0, authenticated 1, encryption_key_size 16, secure connection 1
[00:00:40.178] LOG -- att_server.c.851: Store CCC value 0x0001 for handle 0x0028 of remote E8:D0:FC:D8:C3:00, le device id 3
[00:00:40.213] LOG -- btstack_tlv_flash_bank.c.332: Found tag '4254430d' at position 14460
[00:00:40.223] LOG -- btstack_tlv_flash_bank.c.332: Found tag '4254430e' at position 14444
[00:00:40.234] LOG -- btstack_tlv_flash_bank.c.332: Found tag '4254430f' at position 13332
[00:00:40.244] LOG -- btstack_tlv_flash_bank.c.332: Found tag '42544310' at position 13316
[00:00:40.254] LOG -- btstack_tlv_flash_bank.c.332: Found tag '42544311' at position 12760
[00:00:40.264] LOG -- btstack_tlv_flash_bank.c.332: Found tag '42544312' at position 12744
[00:00:40.274] LOG -- btstack_tlv_flash_bank.c.332: Found tag '42544313' at position 12728
[00:00:40.283] LOG -- att_server.c.913: tag_for_empty 4254430c, tag_for_lowest_seq_nr 42544313
[00:00:40.292] LOG -- btstack_tlv_flash_bank.c.377: write '4254430c', len 8 at 14476
[00:00:40.302] LOG -- hids_device.c.244: Enable Report Feature notifications: 1
[00:00:40.310] LOG -- l2cap.c.980: l2cap_send_prepared_connectionless handle 1025, cid 0x04
>LE Connection Update:
 Connection Interval: 15.00 ms
Connection Latency: 0
[00:00:47.159] LOG -- l2cap.c.1005: L2CAP_EVENT_CHANNEL_CAN_SEND_NOW local_cid 0x4
[00:00:47.167] LOG -- l2cap.c.980: l2cap_send_prepared_connectionless handle 1025, cid 0x04
Mouse: -30/0 - buttons: 00
[00:00:48.999] LOG -- l2cap.c.1005: L2CAP_EVENT_CHANNEL_CAN_SEND_NOW local_cid 0x4
[00:00:49.007] LOG -- l2cap.c.980: l2cap_send_prepared_connectionless handle 1025, cid 0x04
Mouse: 30/0 - buttons: 00
[00:00:50.229] LOG -- l2cap.c.1005: L2CAP_EVENT_CHANNEL_CAN_SEND_NOW local_cid 0x4
[00:00:50.237] LOG -- l2cap.c.980: l2cap_send_prepared_connectionless handle 1025, cid 0x04
Mouse: 0/-30 - buttons: 00

mringwal, 2020-07-07:
Hi. Tricky to say whom to blame here. Does it work with other LE-only Bluetooth mouse?
You could try the hid_mouse_demo as well

LeslieChe, 2020-07-15:
Yes, my computer and Microsoft mouse can be connected normally and operate normally.

hid_mouse_demo? It seems to be used with classic Bluetooth, but I need LE-only Bluetooth mouse.

What computer did you use for the test?

If the test passes, could you tell me what brand and type of the computer?