bluekitchen/btstack #302: Link Key Conversion

caleedubya, 2020-07-08:
Within the BTStack manual - [https://bluekitchen-gmbh.com/btstack.pdf] is the following note:

5.7.7. Cross-transport Key Derivation for LE Secure Connections. In a dual- mode configuration, BTstack automatically generates an BR/EDR Link Key from the LE LTK via the Link Key Conversion function h6. It is then stored in the link key db.

My question is - how do we call the Link Key Conversion function h6? Any help, or even a example project would be much appreciated!

mringwal, 2020-07-15:
No bug reported yet, discussion on mailing list.

closed 2020-07-15 by mringwal

drewandre, 2020-07-15:
@mringwal is this a public mailing list or internal? I'd be interesting in hearing more about espressif development is it's public.

mringwal, 2020-07-15:
On the public mailing list / BTstack Google Group: https://groups.google.com/forum/#!topic/btstack-dev/SRz8jYJXcLw

This was with regard to using BTstack in general, which is possible on the ESP32. For questions/support on the ESP-IDF version of Bluedroid, which is used by default on the ESP32, please use the Espressif forum or post issues on the esp-idf project.