bluekitchen/btstack #307: gap_pin_code_response to support non-string chars

ricardoquesada, 2020-07-22:
Currently btstack has the convenience `gap_pin_code_response` used to reply to the `HCI_EVENT_PIN_CODE_REQUEST` 

The problem is that only "strings" chars are supported. The declaration of the function is:

* `int gap_pin_code_response(bd_addr_t addr, const char * pin)`

and somewhere in `hci.c`, `strlen(pin)` is used to determine the pin length. Usually that would be enough, but Nintendo Wii uses Mac Address as the Pin, and a 00 could be used as part of the Pin.

The request is to provide a similar function, but instead of receiving a "C string", one that supports just "data". E.g:

* `int gap_pin_code_response2(bt_addr_t addr, const unsigned char * pin, int pin_size)`

mringwal, 2020-07-22:
Hi. Sorry for ignoring this so long. Please check (my similar) commit aad97216186d8a961b020b0c707f006ace68c167

closed 2020-07-22 by mringwal

ricardoquesada, 2020-07-22:
Thanks for fixing it so quickly.
(I forgot that I had sent a patch for it... I was about to send another patch).