bluekitchen/btstack #309: pan_lwip_http_server.c bug

naliekkez, 2020-08-06:
**Describe the bug**
I had successfully loaded the file into esp32. however, it seems it's not possible to establish the connection between my smartphone (Xiaomi) and esp32. it seems that the link key accepted however it got close due to "link key based on SC, but remote does not support SC".

I am quite new here and did not know much on what to do. sorry if it has been discussed before. additionally, sometimes you might encounter "guru Meditation Error: Core 0 panic'ed (Interrupt wdt timeout on CPU0)"

**To Reproduce**
1. load the pan_lwip_http_server.c to esp32
2. try to pair it normally.

**Expected behavior**
Supposed to be able to connect

**HCI Packet Logs**
`[00:08:51.209] LOG -- hci.c.2249: Connection_incoming: D8:63:75:7D:A3:A6, type 1`
`[00:08:51.210] LOG -- hci.c.193: create_connection_for_addr D8:63:75:7D:A3:A6, type fd`
`[00:08:51.213] LOG -- hci.c.3785: sending hci_accept_connection_request`
`[00:08:51.244] LOG -- hci.c.2273: Connection_complete (status=0) D8:63:75:7D:A3:A6`
`[00:08:51.245] LOG -- hci.c.2293: New connection: handle 129, D8:63:75:7D:A3:A6`
`[00:08:51.248] LOG -- hci.c.4601: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1`
`[00:08:51.303] LOG -- hci.c.2386: HCI_EVENT_LINK_KEY_REQUEST`
`[00:09:14.207] LOG -- hci.c.3827: responding to link key request`
`[00:09:14.207] LOG -- btstack_tlv_esp32.c.54: tag 42544c00 -> 42544C00`
`[00:09:14.208] LOG -- btstack_tlv_esp32.c.68: read tag 42544C00`
`[00:09:14.214] LOG -- btstack_tlv_esp32.c.54: tag 42544c01 -> 42544C01`
`[00:09:14.220] LOG -- btstack_tlv_esp32.c.68: read tag 42544C01`
`[00:09:14.226] LOG -- btstack_tlv_esp32.c.54: tag 42544c02 -> 42544C02`
`[00:09:14.232] LOG -- btstack_tlv_esp32.c.68: read tag 42544C02`
`[00:09:14.238] LOG -- btstack_tlv_esp32.c.54: tag 42544c03 -> 42544C03`
`[00:09:14.244] LOG -- btstack_tlv_esp32.c.68: read tag 42544C03`
`[00:09:14.250] LOG -- btstack_tlv_esp32.c.54: tag 42544c04 -> 42544C04`
`[00:09:14.256] LOG -- btstack_tlv_esp32.c.68: read tag 42544C04`
`[00:09:14.261] LOG -- btstack_tlv_esp32.c.54: tag 42544c05 -> 42544C05`
`[00:09:14.268] LOG -- btstack_tlv_esp32.c.68: read tag 42544C05`
`[00:09:14.273] LOG -- btstack_tlv_esp32.c.54: tag 42544c06 -> 42544C06`
`[00:09:14.280] LOG -- btstack_tlv_esp32.c.68: read tag 42544C06`
`[00:09:14.285] LOG -- btstack_tlv_esp32.c.54: tag 42544c07 -> 42544C07`
`[00:09:14.291] LOG -- btstack_tlv_esp32.c.68: read tag 42544C07`
`[00:09:14.297] LOG -- btstack_tlv_esp32.c.54: tag 42544c08 -> 42544C08`
`[00:09:14.303] LOG -- btstack_tlv_esp32.c.68: read tag 42544C08`
`[00:09:14.309] LOG -- btstack_tlv_esp32.c.54: tag 42544c09 -> 42544C09`
`[00:09:14.315] LOG -- btstack_tlv_esp32.c.68: read tag 42544C09`
`[00:09:14.321] LOG -- btstack_tlv_esp32.c.54: tag 42544c0a -> 42544C0A`
`[00:09:14.327] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0A`
`[00:09:14.333] LOG -- btstack_tlv_esp32.c.54: tag 42544c0b -> 42544C0B`
`[00:09:14.339] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0B`
`[00:09:14.345] LOG -- btstack_tlv_esp32.c.54: tag 42544c0c -> 42544C0C`
`[00:09:14.351] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0C`
`[00:09:14.357] LOG -- btstack_tlv_esp32.c.54: tag 42544c0d -> 42544C0D`
`[00:09:14.363] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0D`
`[00:09:14.368] LOG -- btstack_tlv_esp32.c.54: tag 42544c0e -> 42544C0E`
`[00:09:14.375] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0E`
`[00:09:14.380] LOG -- btstack_tlv_esp32.c.54: tag 42544c0f -> 42544C0F`
`[00:09:14.387] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0F`
`[00:09:14.393] LOG -- btstack_link_key_db_tlv.c.95: tag 42544c0f, addr D8:63:75:7D:A3:A6`
`[00:09:14.400] LOG -- hci.c.3838: Link key based on SC, but remote does not support SC -> disconnect`
`[00:09:14.409] LOG -- hci.c.937: Connection closed: handle 0x81, D8:63:75:7D:A3:A6`
`[00:09:14.416] LOG -- hci.c.4601: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0`
`ASSERT_WARN(1315 257), in ke_task.c at line 252`

**Environment:  (please complete the following information):**
 - BTStack current version
 - ESP32
 - xiaomi redmi note 5

mringwal, 2020-08-06:
Interesting. The pairing was fine but you get the error on reconnect. If you delete the pairing on the phone, you should be able to connect/pair once and try the demo. I’ll check on esp32 when I get a chance whats the problem on reconnect.

naliekkez, 2020-08-06:
I had tried to unpair and pair it. it either go like before or guru Meditation Error: Core 0 panic'ed (Interrupt wdt timeout on CPU0)

naliekkez, 2020-08-07:
Here is the scenario where I get Meditation Error: Core 0 panic'ed (Interrupt wdt timeout on CPU0). 

Seems like core 0 got overloaded..

`BTstack up and running at FC:F5:C4:19:7E:BA
[00:00:19.032] LOG -- hci.c.2249: Connection_incoming: D8:63:75:7D:A3:A6, type 1
[00:00:19.033] LOG -- hci.c.193: create_connection_for_addr D8:63:75:7D:A3:A6, type fd
[00:00:19.036] LOG -- hci.c.3785: sending hci_accept_connection_request
[00:00:19.049] LOG -- hci.c.2273: Connection_complete (status=0) D8:63:75:7D:A3:A6
[00:00:19.050] LOG -- hci.c.2293: New connection: handle 129, D8:63:75:7D:A3:A6
[00:00:19.057] LOG -- hci.c.4601: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1
[00:00:49.170] LOG -- hci.c.3863: IO Capability Request received, stack bondable 1, io cap 3
[00:00:49.170] LOG -- hci.c.4746: gap_mitm_protection_required_for_security_level 0
ASSERT_PARAM(28 0), in lc_task.c at line 3328
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400842ad  PS      : 0x00060b34  A0      : 0x8002d484  A1      : 0x3ffd3280  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffbdbb4  A7      : 0x60008050  A8      : 0x800842ad  A9      : 0x3ffd3260  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffb  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400841e5  LEND    : 0x400841ec  LCOUNT  : 0x00000000  

ELF file SHA256: 7e2836b8201f3886

Backtrace: 0x400842aa:0x3ffd3280 0x4002d481:0x3ffd32a0 0x40024f79:0x3ffd32d0 0x4001a637:0x3ffd32f0 0x40019d11:0x3ffd3320 0x40055b4d:0x3ffd3340 0x4010a057:0x3ffd3360 0x4010a615:0x3ffd3380 0x4008d055:0x3ffd33b0

[0;32mI (13752) esp_core_dump_uart: Press Enter to print core dump to UART...[0m
[0;32mI (13759) esp_core_dump_uart: Print core dump to uart...[0m
[0;32mI (13765) esp_core_dump_common: Found tasks: (6)![0m
================= CORE DUMP START =================
ZBcAAAEAAAAGAAAAdAEAAA==
RDT9P8Ax/T88NP0/
wDH9P9Az/T+ddKqrSBv8P0gb/D9ENP0/QBv8PwIAAAAMJP0/DCT9P0Q0/T8AAAAA
FwAAAEAk/T9idENvbnRyb2xsZXIAYb8AAAAAADw0/T8AAAAAIQAGABcAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAHR2+z/cdvs/RHf7PwAAAAAAAAAA
AQAAAAAAAAAAAAAAAAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA9BDT
/BkIQK1CCEA0CwYAhNQCgIAy/T8BAAAAAAAAAAAAAABUgABgtNv7P1CAAGCtQgiA
YDL9PwQAAAAAAAAATIAAYP////8AAAAA+////wQAAAAFAAAAAAAAAOVBCEDsQQhA
AAAAAEyAAGD/////0BUIQEz++z8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAABQAAAD/////AAAAAPv///98TwKAoDL9P+zX/D8BAQAA
UN/5PwANAAABAAAAyNv7PzqmAYDQMv0/AAAAAOzX/D8gdgFAQIH7PwMAAAAjBAYA
eEEIQNinAUBo/fo/AQAAABSdAYDwMv0/4Nf8P2CD+z8BAQAAKwQAAMwAAAABAAAA
UFsFgCAz/T+UpAFApIH7P1BbBYAgM/0/hHUBQKSB+z8FCAAAaP36P8CC+T8BAAAA
WqAQgEAz/T8AAAAAAAAAAB8AAADA5fs/AQAAAAAAAAAYphCAYDP9PzhbBUDUB/w/
LCT9PwAAAAA0JP0/AOb7P1jQCICAM/0/5Ab8P8wH/D//////JK0QQBAI/D/MB/w/
AAAAALAz/T8woRBAAAAAAAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABAAAA
AAAAANAz/T8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAADcM/0/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAA
tHv7P7C5+z88vvs/
sLn7P9C9+z9X+UEAkBn8P5AZ/D+0e/s/iBn8PxgAAAAFxlvIy3lVSLR7+z8AAAAA
AQAAAECs+z9tYWluAJHZFqX4N9rIfJ8AAAAAADy++z8AAAAAIwQGAAEAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAHR2+z/cdvs/RHf7PwAAAAAAAAAA
AQAAAAAAAAAAAAAAAAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA9+CS
/BkIQPA+CEAwDwYAhI4IgHC6+z8AAAAABwAAAGA2/T8AAAAAFDb9PyMABgDwPgiA
ULr7P4AAAAAjDwYAIA8GAL2+AACqAgAAULv7PxkAAAD+/////BsIQODCAED2wgBA
//////wbCEC9vgAAJPcIQEyI+j8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAABZACIBwuvs/sO37PwAAAADogRGAkLr7P8Qe/T/YzPs/
YDb9PwAAAAAUNv0/IwAGAOmYEICwuvs/AAAAAA0AAAANAAAA0B79PwQAAACmo311
9OIOgNC6+z/MzPs/DQAAAAAAAAACAAAAnB79P/w/CED4gw6A8Lr7PwEAAADNzPs/
[... 75 lines cut ...]
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAA=
GHj7P3Cq+z84rPs/
cKr7P9Cr+z+ZFnbxDBn8PwwZ/D8YePs/BBn8PwMAAACQ//o/kP/6Pxh4+z+I//o/
FgAAADya+z9lc3BfdGltZXIAQ+Q+yT8AAAAAADis+z8AAAAAIQAGABYAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAHR2+z/cdvs/RHf7PwAAAAAAAAAA
AQAAAAAAAAAAAAAAAAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAYoxz
/BkIQPwSCEAwAAYAKNoIgDCr+z8AAAAAAAAAAGgb/D+8e/s/4Mn7P5Q1QT/8EgiA
EKv7P9wA8D8BAAAA2Nv7PwEAAAAwmvs//P///wAAAAClpaWlpaWlpQAAAAAAAAAA
AAAAAPwbCEABAAAAJPcIQEx2+j8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAPwZCEBQ0AhAMAAFAAAAAABTNw2AUKv7P2T/+j8AAAAA
AAAAAEA3DUAAAAAAAAAAAFjQCICQq/s/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
/////wAAAAAAAAAAAAAAALD/+j8AAAAAAQAAAAIAAAAAAAAAsKv7P0A3DUAAAAAA
IwMGAAAAAAAAAAAAAQAAAAAAAADQq/s/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
BhEIgGA7/j/wGPw/9AFAPwAAAAAAAAAA3Kv7PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
================= CORE DUMP END =================
[0;32mI (14496) esp_core_dump_uart: Core dump has been written to uart.[0m
CPU halted.
`

vvhh2002, 2020-08-12:
This situation also occurs in hpf_hf_demo.

mringwal, 2020-08-12:
Hi. There are two issues here. The original, which looks like a false positive detection of a downgrade attack, and the crash. let's focus on the first one. the crash looks like an internal error in ESP32 code.

I just tried to compile current btstack for esp32 on my mac, but so far failed to update/setup the esp-idf + toolchain stuff.

Could you enable HCI Logging as described below and post a new log where the SC downgrade error shows up?

How to get packet log depends on the port:
- For ESP32: BTstack's `port/esp32/integrate.py` script copies the current version of BTstack into the esp-idf folder. To enable packet logging, you need to edit esp-idf/components/btstack/main.c. There, uncomment the call to 'hci_dump_open(..)'

naliekkez, 2020-08-12:
Hi, I have done that. However, I think you need to uncomment it on main.c at the working folder of the example.

CMIIW.

Thanks for the response. The log with HCI debugging enabled.

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5424
ho 0 tail 12 room 4
load:0x40078000,len:14240
load:0x40080400,len:3940
entry 0x40080688
[0;32mI (227) cpu_start: Pro cpu up.[0m
[0;32mI (227) cpu_start: Application information:[0m
[0;32mI (227) cpu_start: Project name:     pan_lwip_http_server[0m
[0;32mI (230) cpu_start: App version:      1[0m
[0;32mI (235) cpu_start: Compile time:     Aug 11 2020 15:51:51[0m
[0;32mI (241) cpu_start: ELF file SHA256:  2b2289496b355c58...[0m
[0;32mI (247) cpu_start: ESP-IDF:          v4.0.1-dirty[0m
[0;32mI (252) cpu_start: Single core mode[0m
[0;32mI (257) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (264) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (270) heap_init: At 3FFB7468 len 00000B98 (2 KiB): DRAM[0m
[0;32mI (276) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (282) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM[0m
[0;32mI (288) heap_init: At 3FFCABD0 len 00015430 (85 KiB): DRAM[0m
[0;32mI (294) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM[0m
[0;32mI (301) heap_init: At 40078000 len 00008000 (32 KiB): IRAM[0m
[0;32mI (307) heap_init: At 40092028 len 0000DFD8 (55 KiB): IRAM[0m
[0;32mI (313) cpu_start: Pro cpu start user code[0m
[0;32mI (331) spi_flash: detected chip: generic[0m
[0;32mI (331) spi_flash: flash io: dio[0m
[0;32mI (331) esp_core_dump_common: Init core dump to UART[0m
[0;31mE (334) esp_core_dump_common: No core dump partition found![0m
[0;32mI (341) cpu_start: Starting scheduler on PRO CPU.[0m
[00:00:00.347] LOG -- btstack_run_loop_freertos.c.309: run loop init, task 0x3ffb7bb4, queue item size 8
[00:00:00.355] LOG -- main.c.272: transport_register_packet_handler
[00:00:00.368] LOG -- btstack_tlv.c.48: TLV Instance 0x3f4095b4
[00:00:00.368] LOG -- btstack_tlv_esp32.c.54: tag 42544400 -> 42544400
[00:00:00.373] LOG -- btstack_tlv_esp32.c.68: read tag 42544400
[00:00:00.379] LOG -- btstack_tlv_esp32.c.54: tag 42544401 -> 42544401
[00:00:00.385] LOG -- btstack_tlv_esp32.c.68: read tag 42544401
[00:00:00.391] LOG -- btstack_tlv_esp32.c.54: tag 42544402 -> 42544402
[00:00:00.397] LOG -- btstack_tlv_esp32.c.68: read tag 42544402
[00:00:00.403] LOG -- btstack_tlv_esp32.c.54: tag 42544403 -> 42544403
[00:00:00.409] LOG -- btstack_tlv_esp32.c.68: read tag 42544403
[00:00:00.415] LOG -- btstack_tlv_esp32.c.54: tag 42544404 -> 42544404
[00:00:00.421] LOG -- btstack_tlv_esp32.c.68: read tag 42544404
[00:00:00.427] LOG -- btstack_tlv_esp32.c.54: tag 42544405 -> 42544405
[00:00:00.433] LOG -- btstack_tlv_esp32.c.68: read tag 42544405
[00:00:00.438] LOG -- btstack_tlv_esp32.c.54: tag 42544406 -> 42544406
[00:00:00.445] LOG -- btstack_tlv_esp32.c.68: read tag 42544406
[00:00:00.450] LOG -- btstack_tlv_esp32.c.54: tag 42544407 -> 42544407
[00:00:00.456] LOG -- btstack_tlv_esp32.c.68: read tag 42544407
[00:00:00.462] LOG -- btstack_tlv_esp32.c.54: tag 42544408 -> 42544408
[00:00:00.468] LOG -- btstack_tlv_esp32.c.68: read tag 42544408
[00:00:00.474] LOG -- btstack_tlv_esp32.c.54: tag 42544409 -> 42544409
[00:00:00.480] LOG -- btstack_tlv_esp32.c.68: read tag 42544409
[00:00:00.486] LOG -- btstack_tlv_esp32.c.54: tag 4254440a -> 4254440A
[00:00:00.492] LOG -- btstack_tlv_esp32.c.68: read tag 4254440A
[00:00:00.498] LOG -- btstack_tlv_esp32.c.54: tag 4254440b -> 4254440B
[00:00:00.504] LOG -- btstack_tlv_esp32.c.68: read tag 4254440B
[00:00:00.510] LOG -- btstack_tlv_esp32.c.54: tag 4254440c -> 4254440C
[00:00:00.516] LOG -- btstack_tlv_esp32.c.68: read tag 4254440C
[... 413 lines cut ...]
[00:00:31.355] LOG -- btstack_tlv_esp32.c.68: read tag 42544C08
[00:00:31.360] LOG -- btstack_tlv_esp32.c.54: tag 42544c09 -> 42544C09
[00:00:31.367] LOG -- btstack_tlv_esp32.c.68: read tag 42544C09
[00:00:31.372] LOG -- btstack_tlv_esp32.c.54: tag 42544c0a -> 42544C0A
[00:00:31.379] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0A
[00:00:31.384] LOG -- btstack_tlv_esp32.c.54: tag 42544c0b -> 42544C0B
[00:00:31.390] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0B
[00:00:31.396] LOG -- btstack_tlv_esp32.c.54: tag 42544c0c -> 42544C0C
[00:00:31.402] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0C
[00:00:31.408] LOG -- btstack_tlv_esp32.c.54: tag 42544c0d -> 42544C0D
[00:00:31.414] LOG -- btstack_tlv_esp32.c.68: read tag 42544C0D
[00:00:31.420] LOG -- btstack_link_key_db_tlv.c.95: tag 42544c0d, addr D8:63:75:7D:A3:A6
[00:00:31.428] LOG -- hci.c.3838: Link key based on SC, but remote does not support SC -> disconnect
[00:00:31.437] CMD => 06 04 03 80 00 05 
[00:00:31.440] EVT <= 05 04 00 80 00 13 
[00:00:31.444] LOG -- hci.c.937: Connection closed: handle 0x80, D8:63:75:7D:A3:A6
[00:00:31.451] LOG -- hci.c.4601: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0
[00:00:31.458] EVT <= 61 01 00 
[00:00:31.460] EVT <= 0F 04 02 05 06 04 
[00:00:31.464] LOG -- btstack_run_loop_freertos.c.247: RL: portMAX_DELAY 4294967295
[00:00:31.471] LOG -- btstack_run_loop_freertos.c.267: RL: wait with timeout 4294967295
[00:00:31.479] EVT <= 6E 00 
[00:00:31.482] LOG -- btstack_run_loop_freertos.c.247: RL: portMAX_DELAY 4294967295
[00:00:31.489] LOG -- btstack_run_loop_freertos.c.267: RL: wait with timeout 4294967295
ASSERT_WARN(1315 1), in ke_task.c at line 252`

mringwal, 2020-08-12:
Good catch. We've moved the main.c into the example recently to make integration with other examples easier.
Could you also disable ENABLE_LOG_DEBUG (it's only useful in special cases for a single file), then delete the pairing on your phone to capture a packet log that includes the initial pairing and also the later reconnect. This could be two files, too

naliekkez, 2020-08-12:
I am sorry, I had disabled the log. however, after I repopulate the BTStack by ./integrate_btstack.py I did not manage to reproduce the SC issue.

I will keep trying. this kind of error happens quite often.

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5424
ho 0 tail 12 room 4
load:0x40078000,len:13592
ho 0 tail 12 room 4
load:0x40080400,len:3952
entry 0x4008068c
[0;32mI (229) cpu_start: Pro cpu up.[0m
[0;32mI (229) cpu_start: Application information:[0m
[0;32mI (229) cpu_start: Project name:     pan_lwip_http_server[0m
[0;32mI (232) cpu_start: App version:      1[0m
[0;32mI (237) cpu_start: Compile time:     Aug 12 2020 17:34:38[0m
[0;32mI (243) cpu_start: ELF file SHA256:  1580e65f005637d9...[0m
[0;32mI (249) cpu_start: ESP-IDF:          v4.0.1-dirty[0m
[0;32mI (254) cpu_start: Single core mode[0m
[0;32mI (258) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (265) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (271) heap_init: At 3FFB7468 len 00000B98 (2 KiB): DRAM[0m
[0;32mI (277) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (284) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM[0m
[0;32mI (290) heap_init: At 3FFCABC8 len 00015438 (85 KiB): DRAM[0m
[0;32mI (296) heap_init: At 3FFE0440 len 0001FBC0 (126 KiB): D/IRAM[0m
[0;32mI (302) heap_init: At 40078000 len 00008000 (32 KiB): IRAM[0m
[0;32mI (308) heap_init: At 40092028 len 0000DFD8 (55 KiB): IRAM[0m
[0;32mI (315) cpu_start: Pro cpu start user code[0m
[0;32mI (332) spi_flash: detected chip: generic[0m
[0;32mI (333) spi_flash: flash io: dio[0m
[0;33mW (333) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
[0;32mI (343) esp_core_dump_common: Init core dump to UART[0m
[0;31mE (349) esp_core_dump_common: No core dump partition found![0m
[0;32mI (355) cpu_start: Starting scheduler on PRO CPU.[0m
[00:00:00.362] LOG -- btstack_run_loop_freertos.c.309: run loop init, task 0x3ffb7bb4, queue item size 8
[00:00:00.370] LOG -- main.c.272: transport_register_packet_handler
[00:00:00.383] LOG -- btstack_tlv.c.48: TLV Instance 0x3f40d4c8
[00:00:00.383] LOG -- btstack_tlv_esp32.c.54: tag 42544400 -> 42544400
[00:00:00.388] LOG -- btstack_tlv_esp32.c.68: read tag 42544400
[00:00:00.394] LOG -- btstack_tlv_esp32.c.54: tag 42544401 -> 42544401
[00:00:00.400] LOG -- btstack_tlv_esp32.c.68: read tag 42544401
[00:00:00.406] LOG -- btstack_tlv_esp32.c.54: tag 42544402 -> 42544402
[00:00:00.412] LOG -- btstack_tlv_esp32.c.68: read tag 42544402
[00:00:00.418] LOG -- btstack_tlv_esp32.c.54: tag 42544403 -> 42544403
[00:00:00.424] LOG -- btstack_tlv_esp32.c.68: read tag 42544403
[00:00:00.430] LOG -- btstack_tlv_esp32.c.54: tag 42544404 -> 42544404
[00:00:00.436] LOG -- btstack_tlv_esp32.c.68: read tag 42544404
[00:00:00.442] LOG -- btstack_tlv_esp32.c.54: tag 42544405 -> 42544405
[00:00:00.448] LOG -- btstack_tlv_esp32.c.68: read tag 42544405
[00:00:00.453] LOG -- btstack_tlv_esp32.c.54: tag 42544406 -> 42544406
[00:00:00.460] LOG -- btstack_tlv_esp32.c.68: read tag 42544406
[00:00:00.465] LOG -- btstack_tlv_esp32.c.54: tag 42544407 -> 42544407
[00:00:00.471] LOG -- btstack_tlv_esp32.c.68: read tag 42544407
[00:00:00.477] LOG -- btstack_tlv_esp32.c.54: tag 42544408 -> 42544408
[00:00:00.483] LOG -- btstack_tlv_esp32.c.68: read tag 42544408
[00:00:00.489] LOG -- btstack_tlv_esp32.c.54: tag 42544409 -> 42544409
[00:00:00.495] LOG -- btstack_tlv_esp32.c.68: read tag 42544409
[00:00:00.501] LOG -- btstack_tlv_esp32.c.54: tag 4254440a -> 4254440A
[00:00:00.507] LOG -- btstack_tlv_esp32.c.68: read tag 4254440A
[00:00:00.513] LOG -- btstack_tlv_esp32.c.54: tag 4254440b -> 4254440B
[00:00:00.519] LOG -- btstack_tlv_esp32.c.68: read tag 4254440B
[00:00:00.525] LOG -- btstack_tlv_esp32.c.54: tag 4254440c -> 4254440C
[00:00:00.531] LOG -- btstack_tlv_esp32.c.68: read tag 4254440C
[... 688 lines cut ...]
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAA=
GHj7P3Cq+z84rPs/
cKr7P9Cr+z8+GgbdLBr8Pywa/D8YePs/JBr8PwMAAACQ//o/kP/6Pxh4+z+I//o/
FgAAADya+z9lc3BfdGltZXIAOg/dCjEAAAAAADis+z8AAAAAIQAGABYAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAHR2+z/cdvs/RHf7PwAAAAAAAAAA
AQAAAAAAAAAAAAAAAAAAAEgdAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAmvkK
mCEIQOATCEAwCQYACPMIgDCr+z8AAAAAAAAAAIgc/D+8e/s/4Mn7PzgpQT/gEwiA
EKv7P9wA8D8BAAAA6Nv7PwEAAAD///8CAAAAAAAAAAClpaWlpaWlpQAAAAAAAAAA
AAAAAJgjCEABAAAA2OgIQKiC+j8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAJghCEBg5whAMAAFAAAAAAAvNg2AUKv7P2T/+j8AAAAA
AAAAABw2DUAAAAAAAAAAAGjnCICQq/s/AAAAAAAAAAAAAAAAfsQAAAAAAAAAAAAA
/////wAAAAAAAAAAAAAAALD/+j8AAAAAAQAAAAIAAAAAAAAAsKv7Pxw2DUAAAAAA
IwMGAAAAAAAAAAAAAQAAAAAAAADQq/s/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
6hEIgGA7/j8QGvw/3ANAPwAAAAAAAAAA3Kv7PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
================= CORE DUMP END =================
[0;32mI (8745) esp_core_dump_uart: Core dump has been written to uart.[0m
CPU halted.
`

mringwal, 2020-08-12:
Thanks for the new logs. There, BTstack sends a 'HCI Read Remote Extended Features" command. The Bluetooth library in the ESP32 should respond with an HCI Command Status indicating that it understands the command and let us know when it is complete. But it doesn't. A bit later, there's a pairing request which BTstack should confirm, but cannot as it is waiting for the Command Status or Command Complete event. Without the confirm, the pairing fails, we get a disconnect and then BTstack sends the pairing confirmation. The Bluetooth lib should respond with a 'connection doesn't exist' but seems to crash at that point.

I need to  get my ESP32 dev environment up and running to properly file two bug reports for this. a) command status is missing, b) crash on user confirm reply without connection if it reproduces.

naliekkez, 2020-08-13:
Alright, I will do an attempt to find a workaround. Please let me know if you got any progress.

mringwal, 2020-08-13:
first step: after setting a symlink from python to python3, I'm was able to setup the esp32 tools.
Then, I'm getting a different error, roughly at the same time. While this is a bug in the ESP32 Bluetooth lib, I think I can add a small workaround to avoid it. More to come....

mringwal, 2020-08-14:
Hi. Please try again either on current master or develop branches. Requesting authentication on an already AES-CCM encrypted connection causes the link to drop. Fortunately, there's AES-CCM shows that the connection was already mutually  authenticated and there's no need to do another authentication.

Your original problem most likely was an artefact from stale link key or having two devices with the same Bluetooth address. Feel free to post again if you get the "Link key based on SC, but remote does not support SC" again.

commit 4a659b0e (2020-08-14): gap: treat AES-CCM encrypted connections as mutually authenticated, fixes #309
commit 13a4533f (2020-08-14): gap: treat AES-CCM encrypted connections as mutually authenticated, fixes #309

# Conflicts:
#	CHANGELOG.md
closed 2020-08-14 by mringwal