bluekitchen/btstack #311: SM compatible

Tsai1005, 2020-08-11:
In the SM profile, there is a default setting about IR & ER
[image]

mringwal, 2020-08-11:
Hi. Could you explain this issue? What did you do? What did you expect? What happened? 
Which side sends the SMP Pairing Failed?

mringwal, 2020-08-11:
btw. it looks like one side is the popular CSR 4.0 dongle, so you're probably on a desktop system. Please also post the /tmp/hci_dump.pklg and or hci_dump.pklg (on windows in local folder).

Tsai1005, 2020-08-11:
Our device is acting as a slave, then connect to the PC, but found that some tablets will terminate the connection after the SM key distribution process, but the situation will change after we modify the IR key.

[keyboard_pc.zip](https://github.com/bluekitchen/btstack/files/5056637/keyboard_pc.zip)

Tsai1005, 2020-08-11:
In my opinion the problem would related to the LTK, it derives from the IR.
[image]

mringwal, 2020-08-12:
What tool do I need to open the keyboard_pc.btt file in the .zip above?

Do you use random bluetooth addresses?
Can you provide btstack's packet log?

The IR and ER should be unique for each device - otherwise, there are issues with using private addresses and also security implications. There's a chance, your table has the IRK derived from  the fixed IR black-listed (I haven't heard about that, but it's quite possible)

In the current version, BTstack will generate random numbers for IR + ER and store it in it's TLV data base (which should be in Flash). If that's working, each device will have unique IR + ER values and the tablet should have no problems with that.

Do you have TLV properly setup such that it can be used by BTstack?

closed 2020-09-13 by Tsai1005