bluekitchen/btstack #312: btstack_crypto unusable after stack restart

jimmo, 2020-08-14:
**Describe the bug**
In MicroPython we provide an API for the user to activate and deactivate the BLE stack. On re-activation after a previous deactivation we can no longer use bstack_crypto functionality.

The underlying cause is that:
 - hci_init will memset `hci_stack`, effectively removing all packet handler registrations.
 - sm_init will call btstack_crypto_init
 - btstack_crypto_init will add a packet handler registration, but then mark in a global variable that it has been run. This variable is never reset

On subsequent startup, btstack_crypto_init will not run and there will be no packet handler registration for the crypto command responses.

From our perspective, we'd be fine with needing to call `btstack_crypto_deinit`, otherwise finding a way for hci shutdown (or hci_init) to automatically clear this flag would be good.

**Environment:  (please complete the following information):**
This is MicroPython, running the embedded run loop on a few different platforms and controllers.
We're at btstack commit `c8b9823f6` but I've checked that the `btstack_crypto_init` behavior is the same at `master`.

mringwal, 2020-08-14:
Hi Jim. So far, BTstack isn't supposed to get de-initialized. 

We expect that hci_power_control(off) is called to shut down the stack when it's not in use, which allows to call hci_power_control(on) later, to bring it up again.

Could this be an option for your setup? If not, please explain :)

jimmo, 2020-08-20:
Thanks @mringwal -- I will investigate and see if this is a possibility for us and get back to you.

One of the scenarios where this is relevant is if the user does a "soft reset" (which in MicroPython means restarting the VM, not the MCU) which results in the heap being invalidated. We currently use btstack with `HAVE_MALLOC=0` so that's fine but I had planned to make it use malloc in the future (to avoid using up bss unnecessarily while BLE isn't in use).

mringwal, 2020-08-20:
I see. When you you call hci_power_control(off), all connections related structs are freed, thus reducing the amount of heap memory. 

I'm open to smaller changes in the core stack code to facilitate a re-init, but it will require careful ordering of the respective init/de-init calls.

mringwal, 2022-06-13:
Hi. While we still need to add a status field to the crypto operation callbacks, the stack supports power off/on, even if it was in the middle of a crypto operation

dpgeorge, 2022-09-08:
Updating to the latest BTstack v1.5.3, this problem has come up again because of this commit: https://github.com/bluekitchen/btstack/commit/2d2d4d3cb13a672c3f1c8f7240f760ba09ebfe54 .  It means that `sm_init()` can now only be called once.

One way to fix that would be to add a `sm_deinit()` function (that sets `sm_initialized` to `false`).

Otherwise, maybe it's cleaner from our side if we have a BTstack pre-init stage which sets up all the things that cannot be reset (eg calling `sm_init()` and relating things).  And then for our soft reset cycle we use hci_power_control off/on.

mringwal, 2022-09-08:
Hi there. I was about to implement sm_deinit(), but then found it already in place since May 2021. So, you could call that.

The de-init function have been added for automated testing. The goal is to get to the exact state after calling all init() function a second time as after the first time. They have not been intended for regular use. (If you do want to use init/deinit, please make sure to call all deinit() functions including btstack_memory_deinit() as the higher layer de-init() functions don't bother to release memory).

So, I'd prefer if you could try to only call hci_power_control(on/off) for your soft reset. This will keep the stack configuration (e.g. gap settings, l2cap/rfcomm services..) all configuration - no idea if that good or bad for MP.

dpgeorge, 2022-09-09:
> I was about to implement sm_deinit(), but then found it already in place since May 2021. So, you could call that.

Oh, I didn't see that function (was just looking at the commit that added the `sm_initialized` flag).  That's good, I tried using `sm_deinit()` and it works.

> If you do want to use init/deinit, please make sure to call all deinit() functions including btstack_memory_deinit() as the higher layer de-init() functions don't bother to release memory).

OK I will try this.

> This will keep the stack configuration (e.g. gap settings, l2cap/rfcomm services..) all configuration - no idea if that good or bad for MP.

We do want services to be reset, so that approach may not work.  I'll try the deinit approach first.

mringwal, 2022-09-09:
Sounds good. The deinit() functions have been added for Classic. If you miss something for LE, please let me know.

closed 2022-09-09 by mringwal