bluekitchen/btstack #314: AVRCP Commands failed in a2dp_sink_demo

jsplyy, 2020-08-24:
**Describe the bug**
AVRCP Commands failed in a2dp_sink_demo

**To Reproduce**
Steps to reproduce the behavior:
1. Run example 'a2dp_sink_demo '
2. Connect from remote device 'my android cell phone MI8'
3. Start action 'play music with one app on cell phone'
4. Input command 'L' or 'K' to pause or stop audio stream, will report Could not perform command

**Expected behavior**
expect that command could pause or stop audio stream

**HCI Packet Logs**
[hci_dump.pklg.zip](https://github.com/bluekitchen/btstack/files/5118174/hci_dump.pklg.zip)

**Environment:  (please complete the following information):**
 - Current BTstack branch: master
 - Bluetooth Controller [CSR 8510]
 - Remote device: [android 10 MI8]

**Additional context**
Add any other context about the problem here.

jsplyy, 2020-08-25:
Hi @mringwal, do you know this issue?

mringwal, 2020-08-25:
hi @jsplyy I'm not aware of the issue and only had time for a quick look at the log. It's still in the queue. The log mentions a 'wrong packet size', which might indicate the problem.

milamikica, 2020-08-26:
Hi, we updated develop branch, could you please try it again.

jsplyy, 2020-08-28:
Hi @mringwal , I try again with latest commits on develop branch, 
and now most commands could work, but some commands still show "Could not perform command, status 0x c"

mringwal, 2020-09-01:
Thanks for retesting. We'll check and get back to you.

mringwal, 2020-09-05:
That was a tricky one: the 'track changed' notification reset the state and caused the 'command disallowed'. We're reworked the notification handling and can use fast forward / stop fast forward now. Please confirm.

mringwal, 2020-09-09:
ping?

mringwal, 2021-07-02:
Merged onto master / no further feedback.

closed 2021-07-02 by mringwal