bluekitchen/btstack #315: Can't compile any of the examples with C++

aethersis, 2020-08-26:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version v4
- Module or chip used:  ESP32-WROOM-32
- IDF version 4.2
- Build System:         idf.py
- Compiler version 8.2.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: No
- Power Supply: Battery with switch down regulator, also tried USB from PC and USB from power supply

## Problem Description
I need to use C++ for my project. I tried the simplest thing which was to modify the CMakeLists.txt file and the examples extension (tried gap_inquiry) to cpp. 

PS
All the demos compile successfuly in C so the setup is correct for C on my side.

### Expected Behavior
The code compiles and links, everything works

### Actual Behavior
The code compiles, however the linker fails with 
c:\users\mw\esp\btstack\port\esp32\example\gap_inquiry\build/../main/main.cpp:59: undefined reference to `btstack_init()'

PS
When I remove btstack_init(), the code compiles and runs, but since the stack is not initialized, the code crashes on the board as expected. There is something wrong with btstack_init() when compiling C++ - it's the only symbol that results in unresolved external for some reason. Other symbols are fine.

### Steps to reproduce
1. Run integrate_btstack.py
2. Run create_examples.py
3. cd to btstack\port\esp32\example\gap_inquiry\main
4. rename files with *.c extension to *.cpp
5. edit CMakeLists.txt and rename *.c to *.cpp 
6. cd ..
7. idf.py clean (also tried deleting contents of build folder to ensure clean state)
8. idf.py build

The wiring is just the devboard plugged to USB

### Code to reproduce this issue
Other than changing the extensions to .cpp, only the CMakeLists.txt file for gap_example was modified as follows:

idf_component_register(
        SRCS "gap_inquiry.cpp" "main.cpp"
        INCLUDE_DIRS "${CMAKE_CURRENT_BINARY_DIR}")

## Debug Logs
C:\Users\mw\esp\btstack\port\esp32\example\gap_inquiry>idf.py build
Checking Python dependencies...
Python requirements from C:\Users\mw\Desktop\esp-idf\requirements.txt are satisfied.
Executing action: all (aliases: build)
Running ninja in directory c:\users\mw\esp\btstack\port\esp32\example\gap_inquiry\build
Executing "ninja all"...
[1/6] cmd.exe /C "cd /D C:\Users\mw\esp\btstack\port\esp32...*********************************************************"
Partition table binary generated. Contents:
*******************************************************************************
 Espressif ESP32 Partition Table
 Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[2/6] Performing build step for 'bootloader'
ninja: no work to do.
[3/4] Linking CXX executable gap_inquiry.elf
FAILED: gap_inquiry.elf
cmd.exe /C "cd . && C:\Users\mw\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address  -nostdlib @CMakeFiles\gap_inquiry.elf.rsp  -o gap_inquiry.elf  && cd ."
c:/users/mw/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.cpp.obj):(.literal.app_main+0x0): undefined reference to `btstack_init()'
c:/users/mw/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/main/libmain.a(main.cpp.obj): in function `app_main':
c:\users\mw\esp\btstack\port\esp32\example\gap_inquiry\build/../main/main.cpp:59: undefined reference to `btstack_init()'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

mringwal, 2020-08-26:
BTstack is in use by other C++ projects, just not by us.
btstack_init() was missing the C++ protection. please try the updated version on develop and let us know.

aethersis, 2020-08-26:
Amazing! It worked like charm! What is the change? Are there plans to merge it to master so I can have it on a stable branch?

mringwal, 2020-08-27:
Well, I've added the cpp guard: https://github.com/bluekitchen/btstack/commit/29f287b5e3a9748cea306f62747a0fdddf2315ca

We regularly merge develop onto master. You can stick with the  version that works for you now on develop and switch  back to master after the next merge - usually at the beginning of the month.

closed 2020-08-27 by mringwal