bluekitchen/btstack #318: [FreeRTOS] No BT main loop triggering in A2DP mode

SP2FET, 2020-09-22:
When I run a2dp_source_demo on an embedded platform with FreeRTOS, the connection is established but no stream is started. I've found out that the timer related to playback callback is being started but it's not triggering the main loop after adding the timer, so the callback is never executed. I fixed it up with adding trigger call in timer adding method

mringwal, 2020-09-23:
Thanks for reporting. To recap, you are suggesting to add a call to btstack_run_loop_freertos_trigger() at the end of btstack_run_loop_freertos_add_timer of the btstack_run_loop_freertos.c, correct?

I'm wondering why this is needed. Would you have a backtrace for the call to btstack_run_loop_freertos_add_timer()?

In A2DP Source Demo, the audio playback starts as a result of AVDTP packet being received. This means, it is called from the first part in the while(1) in btstack_run_loop_freertos_execute():
https://github.com/bluekitchen/btstack/blob/master/platform/freertos/btstack_run_loop_freertos.c#L231

Assuming it sets a new timer, this timer is already in the list of the third block and there should be no need to call btstack_run_loop_freertos_trigger():
https://github.com/bluekitchen/btstack/blob/master/platform/freertos/btstack_run_loop_freertos.c#L245

Any ideas? :)

SP2FET, 2020-09-23:
> Thanks for reporting. To recap, you are suggesting to add a call to btstack_run_loop_freertos_trigger() at the end of 
> btstack_run_loop_freertos_add_timer of the btstack_run_loop_freertos.c, correct?
Exactly, this fixed issue in my case.

I can try to recreate this case once again if it would help and then we can discuss with some logs provided :)

mringwal, 2020-09-23:
I'm definitely interested in understanding (and properly fixing this). The only idea I have would be that a call to trigger run lop is needed if add timer is called from interrupt context (but that's not supported and hell would freeze over...)

SP2FET, 2020-09-23:
The timer is added during normal code execution (not from ISR). I'm recreating the case right now, I'll send here logs in few minutes

SP2FET, 2020-09-23:
hmm... strange. Probably I didn't handle AVDTP properly? When I tried to recreate the case, now it's working flawlessly... So I think the issue is solved.

mringwal, 2020-09-23:
Weird. If you're using our examples, there shouldn't be anything for you to break. Ok. I'm closing this now, but please report back if there's something else.

closed 2020-09-23 by mringwal