bluekitchen/btstack #321: HCI_INIT_SEND_READ_LOCAL_NAME timeout witch BT860 module

argh35, 2020-09-30:
Hi guys,
I am not sure this is a stack issue, but I hope that someone can help me.

I am using a STM32F4 microcontroller, communicating through UART with a BCM module, so I am using the following platform instances:
- btstack_run_loop_freertos (adapted to be used with keil RTX OS)
- btstack_uart_block_freertos
- btstack_chipset_bcm

I am using the master version [f0565ba] committed 31/07/2020.

The stack init procedure usually goes well and the stack goes up.
Sometimes, however, the stack initialization stops after the HCI_INIT_SEND_READ_LOCAL_NAME command.

From my analysis, it seems that if the reply to the command HCI_INIT_SEND_RESET arrives 'late' (after around 140 ms) everything is ok; if it arrives 'too early' (after around 7 ms) then I do not receive the second command reply.

Please see attached file for log (I do not have a free UART to be used as console, so sorry for my not cool log format):
[hci_dump_ble_reset_timeout_ok.txt](https://github.com/bluekitchen/btstack/files/5304216/hci_dump_ble_reset_timeout_ok.txt)
[hci_dump_ble_ko.txt](https://github.com/bluekitchen/btstack/files/5304217/hci_dump_ble_ko.txt)

To try to solve the issue, I configured a timer also for this command, as for the reset one, with a timeout of 50 ms and, when it expires, I power-cycle the module and then restart the procedure.
This often solve the issue after 2-3 retries, but the timeout can expires also several times, that is very annoying.

Any tip or help is appreciated.

Than you,
regards

Arianna

mringwal, 2020-09-30:
Hi Arianna. From your description, it sounds like your loosing bytes in the UART driver. Please verify that this does not happen, e.g. using a logic analyser: compare what you get on the UART and what you get from the UART driver. If it differs there's a bug (and it cannot be fixed at a higher layer).

Actually, it sounds like a bug that we also run into with the STM32 F4 Cube HAL software. Please see this blog post incl. suggested fix here: https://bluekitchen-gmbh.com/btstack-port-for-stm32-f4-discovery-board-with-cc256x/

argh35, 2020-10-01:
Hi, 
thank you for the reply.
I already faced the DMA issue and I used a different approach: I configured the DMA in circular mode with a 128 byte buffer, so I am receiving all the bytes the module sends to me, and I am giving them to the stack when it requests them.
Do you see any problem using this approach?
(By the way, I tried also your approach and I get the same results)
When the timeout occurs, I see in the DMA buffer only 7 bytes (DMA still waiting 121 bytes), that is the reply to the reset command: "0x04 0x0E 0x04 0x01 0x03 0x0C 0x00".

Regards

Arianna

mringwal, 2020-10-01:
Hi Arianna. Using a circular buffer with DMA should work as well. It must be large enough that you can guarantee to raise RTS before it overflows though. For other MCUs without a proper CTS/RTS, circular DMA is the best option I've seen so far .

I've looked at your logs and the last command in both is HCI_READ_LOCAL_VERSION (not HCI_INIT_SEND_READ_LOCAL_NAME).

Also, there are no timeouts on HCI, I'd suggest to disable any timer there until it's fixed (and you hopefully won't need it later, too).

It's hard to tell from the HCI logs. Please use a logic analyzer and compare the bytes in each.
BTstack does three reads per packet. The first packets are short, only the response to  read local name is long (255 bytes + header). Please try to find out, why DMA is waiting for 121 bytes.

Ah... something else.  Some other CYW controllers have a special mode that does not answer to any commands besides reset. Maybe you've run into  this one. It enters this state depending on the controllers CTS (your RTS) line during reset. See here: https://community.cypress.com/message/201375#201375

argh35, 2020-10-08:
Hello,
I think you can be right, because I see a different behaviour at BT860 start-up when everything is working and when it replies only to the reset command: in this case, the CTS goes low only 6 ms after releasing the RESET, while in 'normal' condition it goes low much more later.

However I tried playing with CTS pulling it up and down during the reset but nothing did resolve the issue, and the module datasheet does not speak about the double mode..

By the way now I am confident that it is not a stack issue, so I am closing it.

Thank you for your precious help,
regards

Arianna

closed 2020-10-08 by argh35

mringwal, 2020-10-08:
Thanks for the info. Do you refer to the Controller CTS (which is your RTS that you  control)? I think CTS should stay high during power cycle/reset to get the normal mode. 

Please let me know if/how you've resolved this.

argh35, 2020-10-08:
If I keep the STM32 RTS high during module power cycle/reset I often have another problem: the module starts as 'broadcom' (manufacturer code 0x000F) instead of as 'cypress' (0x0131), so the init procedure goes fine but then the bluetooth is not really working!

The Lairdtech support said me that it is possible if the module is not loading the firmware from the EEPROM but it is booting up from internal ROM..

So I had not really resolved the issue, I am keeping the timeout on the second message and then I restart from the reset if it expires.

mringwal, 2020-10-08:
Well, it's Broadcom tech, just with a new marketing name :)  But yes, if you get Broadcomm as manufacturer code, it didn't load the internal config to make it a Cypress module. So, you make sure that STM32 RTS is low during power cycle, but it sometimes jumps into the special mode and doesn't respond to 'HCI Read Local Version Information'.

argh35, 2020-10-08:
Yes, exactly!