bluekitchen/btstack #325: ESP IDF - btstack_audio_esp32.c - 'I2S_COMM_FORMAT_STAND_I2S' undeclared

jan-sen, 2020-11-29:
Trying to build (_idf.py all_) any project that has  _#include "btstack.h"_ in it fails with the following error:
______
[...]
D:/espressif/esp-idf/components/btstack/btstack_audio_esp32.c: In function 'btstack_audio_esp32_sink_init':
D:/espressif/esp-idf/components/btstack/btstack_audio_esp32.c:140:45: error: 'I2S_COMM_FORMAT_STAND_I2S' undeclared (first use in this function); did you mean 'I2S_COMM_FORMAT_I2S'?
                     .communication_format = I2S_COMM_FORMAT_STAND_I2S,
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~
                                             I2S_COMM_FORMAT_I2S
D:/espressif/esp-idf/components/btstack/btstack_audio_esp32.c:140:45: note: each undeclared identifier is reported only once for each function it appears in
[913/1234] Building C object esp-idf/coap/CMakeFiles/__idf_coap.dir/libcoap/src/resource.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
_____

And indeed, there is no I2S_COMM_FORMAT_STAND_I2S declared in _esp-idf\components\driver\include\driver\i2s.h_, where I assume this is supposed to be located.
Replacing I2S_COMM_FORMAT_STAND_I2S in _esp-idf\components\btstack\btstack_audio_esp32.c_ with I2S_COMM_FORMAT_I2S (which __is__ in i2s.h) does allow successful building.

I did not actually use **btstack_audio_esp32** in any of my tests so I don't know wether the manual change done to _i2s.h_ has any repercussions.

My build environment:
- esp-idf downloaded/installed from https://dl.espressif.com/dl/esp-idf-tools-setup-2.3.exe on 2020-11-26, apparently giving me v4.1
- btstack-master downloaded as zip-file on 2020-11-29
- btstack has been setup within esp-idf with 'port/esp32/integrate_btstack.py'
- the examples have been created by using 'port/esp32/create_examples.py'

Please let me know if you need any other information.

mringwal, 2020-11-29:
Hi. We've recently updated that code and following 'deprectation' markings in the esp-idf docu.

There's a `I2S_COMM_FORMAT_STAND_I2S ` in https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html#_CPPv417i2s_comm_format_t

It is defined here: https://github.com/espressif/esp-idf/blob/master/components/hal/include/hal/i2s_types.h#L63

Could you try to switch to esp-idf master branch?

jan-sen, 2020-12-01:
Thanks, switching to esp-idf master did solve the issue for me!

Just an FYI, you're probably aware of this, I do get a warning now:
_____
[983/1402] Building C object esp-idf/btstack/CMakeFiles/__idf_btstack.dir/3rd-party/lwip/dhcp-server/dhserver.c.obj
D:/esp/esp-idf/components/btstack/3rd-party/lwip/dhcp-server/dhserver.c: In function 'dhserv_init':
D:/esp/esp-idf/components/btstack/3rd-party/lwip/dhcp-server/dhserver.c:325:16: warning: passing argument 2 of 'udp_recv' from incompatible pointer type [-Wincompatible-pointer-types]
  udp_recv(pcb, udp_recv_proc, NULL);
                ^~~~~~~~~~~~~
In file included from D:/esp/esp-idf/components/btstack/3rd-party/lwip/dhcp-server/dhserver.h:39,
                 from D:/esp/esp-idf/components/btstack/3rd-party/lwip/dhcp-server/dhserver.c:25:
D:/esp/esp-idf/components/lwip/lwip/src/include/lwip/udp.h:128:67: note: expected 'udp_recv_fn' {aka 'void (*)(void *, struct udp_pcb *, struct pbuf *, const struct ip_addr *, short unsigned int)'} but argument is of type 'void (*)(void *, struct udp_pcb *, struct pbuf *, const struct ip4_addr *, u16_t)' {aka 'void (*)(void *, struct udp_pcb *, struct pbuf *, const struct ip4_addr *, short unsigned int)'}
 void             udp_recv       (struct udp_pcb *pcb, udp_recv_fn recv,
                                                       ~~~~~~~~~~~~^~~~
_____

Not sure if this is at all related to switching the esp-idf branch.

mringwal, 2020-12-07:
We've update the dhcp-server as well. I think there was a problem with different lwip versions (the one in btstack/3rd-party vs. the one in esp-idf), but the definition that caused the warning is the same in both, so we've fixed it.

closed 2020-12-07 by mringwal