bluekitchen/btstack #326: printf of uint32_t

dlech, 2020-12-01:
In the code, `%u` is used for the `printf()` format for the `uint32_t`. However on 32-bit Arm platforms, GCC defines `uint32_t` as `unsigned long` rather than `unsigned int`. This causes the compiler to complain about using the incorrect format specifier (should be `%lu` in this case).

We could fix this by using `PRIu32` from `inttypes.h` instead of `%u` or we could cast `uint32_t` to `unsigned int` when used in `printf()`.

mringwal, 2020-12-02:
Thanks for reporting. We know ... :) and just recently fixed a few of these on develop - cast to unsigned int for simple numbers, PRIu32 for actual 32bit values. 

Could you check the develop branch and report missing ones?

dlech, 2020-12-02:
It looks like you got the ones I ran into in https://github.com/bluekitchen/btstack/commit/97d2cfbc6fcf2d7b97d50845edc9a3056528f809.

I will reopen if I find more.

closed 2020-12-02 by dlech