bluekitchen/btstack #327: btstack_audio_esp32.c fails to compile with ESP-IDF 4.1

ecc1, 2020-12-09:
master branch (currently at v1.2-21-gcbf5eb7c7) fails to build btstack_audio_esp32.c with ESP-IDF v4.1, with the following error:

/opt/esp/idf/components/btstack/btstack_audio_esp32.c: In function 'btstack_audio_esp32_sink_init':
/opt/esp/idf/components/btstack/btstack_audio_esp32.c:140:45: error: 'I2S_COMM_FORMAT_STAND_I2S' undeclared (first use in this function); did you mean 'I2S_COMM_FORMAT_I2S'?
                     .communication_format = I2S_COMM_FORMAT_STAND_I2S,
                                             ^~~~~~~~~~~~~~~~~~~~~~~~~
                                             I2S_COMM_FORMAT_I2S

mringwal, 2020-12-09:
please use develop branch until next master update which should happen later this week.

mringwal, 2020-12-09:
(or if I got this wrong, use esp-idf master branch)

mringwal, 2021-07-08:
It's been on master for a while now... :)

closed 2021-07-08 by mringwal