bluekitchen/btstack #328: Port BtStack to STM32L45

hananabilabd, 2020-12-11:
Hello
Currently, I trying to modify stm32f4discovery-cc256x port to work with STM32L45+CC2564B
I've done the compiling successfully eventually. after I saw [blog ](https://bluekitchen-gmbh.com/btstack-port-for-stm32-f4-discovery-board-with-cc256x/)explanation.

- I just wanna know try a simple SPP example that sends android packets to STM32L45, it is written in the blog we can use spp_and_le_counter.c which doesn't exist anymore. So which is the most simple SPP example to use ? 

- in stm32-f4discovery-cc256x\port\port.c  what is meant by these defines ? I can see in the linker script there is no extra section created.
```
    #define HAL_FLASH_BANK_0_ADDR 0x080C0000
    #define HAL_FLASH_BANK_1_ADDR 0x080E0000
    #define HAL_FLASH_BANK_0_SECTOR FLASH_SECTOR_10
    #define HAL_FLASH_BANK_1_SECTOR FLASH_SECTOR_11
```

mringwal, 2020-12-12:
Hi. Feel free to ask on the mailing list: https://groups.google.com/group/btstack-dev

The simplest SPP demo is `spp_counter`

The `HAL_FLASH_xxx` define which are of the Flash is used to store bonding information.  Here, the last two 128 kB 'banks' of the 1MB Flash was used. (a bit large, but hey, thee's still 768 kB left for your app).

Good point about the linker scripts. It would be good to reserve this area, and then even use the defines from the linker script.

closed 2020-12-12 by mringwal

hananabilabd, 2020-12-12:
Thanks for the clarification.
regarding the **FLASH_SECTOR_10** which is defined in stm32f4xx_hal_flash_ex.h, it has no corresponding in stm32l4xx_hal_flash_ex.h and there are no section, would it matter if I just put it with any value ? 

- Also **_close _isatty _lseek _fstat**   functions give compilation error cause they are already defined in syscalls.c

mringwal, 2020-12-12:
I've written about the Flash stuff in another blog post. http://bluekitchen-gmbh.com/storing-link-keys-in-flash-memory/
If you have a different MCU, you might need to adapt the flash read/write implementation.

Which toolchain do you use? Which syscalls.c? With or without ENABLE_SEGGER_RTT?
The F4 port compiles with only minor warnings and no error with the provided Makefile. The libc functions are not used, they are only there to get it to compile, so feel free to remove them for your setup.