bluekitchen/btstack #331: Unable To Build Renamed Example

H0r53, 2021-01-29:
**Describe the bug**
I'm interested in creating a very simple BT utility based on combining some of the examples. I've successfully built examples by first navigating to `port/libusb` and executing `make <example_name>`, where <example_name> is the name of any given file in the example folder. For example: `make gap_inquiry` builds without any issues and runs fine.

I started by simply copying `example/gap_inquiry.c` to a new file name, let's say `example/gap_inquiry2.c`, and changing `#define BTSTACK_FILE__ "gap_inquiry.c"` to `#define BTSTACK_FILE__ "gap_inquiry2.c"`. I looked through the rest of the source in this file and nothing else appears to be filename specific. However, when I execute `make gap_inquiry2` as before I receive the following errors:

```
$ make gap_inquiry2
cc -I. -I../../src/ble -I../../src/classic -I../../src/mesh -I../../src -I../../3rd-party/bluedroid/decoder/include -I../../3rd-party/bluedroid/encoder/include -I../../3rd-party/hxcmod-player -I../../3rd-party/lwip/core/src/include/ -I../../3rd-party/lwip/dhcp-server -I../../3rd-party/md5 -I../../3rd-party/micro-ecc -I../../3rd-party/yxml -I../../platform/lwip -I../../platform/lwip/port -g -std=c99 -Wall -Wmissing-prototypes -Wstrict-prototypes -Wshadow -Wunused-parameter -Wredundant-decls -Wsign-compare -I../../platform/posix -I../../platform/embedded -I../../3rd-party/tinydir -I../../3rd-party/rijndael -I../../chipset/zephyr -I/usr/include/libusb-1.0 -pthread -DHAVE_PORTAUDIO  -lm -lusb-1.0 -lportaudio -lasound -lm -lpthread  ../../example/gap_inquiry2.c   -o gap_inquiry2
/usr/bin/ld: /usr/lib/gcc/x86_64-linux-gnu/10/../../../x86_64-linux-gnu/Scrt1.o: in function `_start':
(.text+0x20): undefined reference to `main'
/usr/bin/ld: /tmp/ccCJ8XMj.o: in function `gap_event_inquiry_result_get_bd_addr':
/home/kali/Repos/btstack/port/libusb/../../src/btstack_event.h:3224: undefined reference to `reverse_bytes'
/usr/bin/ld: /tmp/ccCJ8XMj.o: in function `gap_event_inquiry_result_get_class_of_device':
/home/kali/Repos/btstack/port/libusb/../../src/btstack_event.h:3242: undefined reference to `little_endian_read_24'
/usr/bin/ld: /tmp/ccCJ8XMj.o: in function `gap_event_inquiry_result_get_clock_offset':
/home/kali/Repos/btstack/port/libusb/../../src/btstack_event.h:3251: undefined reference to `little_endian_read_16'
/usr/bin/ld: /tmp/ccCJ8XMj.o: in function `getDeviceIndexForAddress':
/home/kali/Repos/btstack/port/libusb/../../example/gap_inquiry2.c:82: undefined reference to `bd_addr_cmp'
/usr/bin/ld: /tmp/ccCJ8XMj.o: in function `start_scan':
/home/kali/Repos/btstack/port/libusb/../../example/gap_inquiry2.c:91: undefined reference to `gap_inquiry_start'
/usr/bin/ld: /tmp/ccCJ8XMj.o: in function `do_next_remote_name_request':
/home/kali/Repos/btstack/port/libusb/../../example/gap_inquiry2.c:108: undefined reference to `bd_addr_to_str'
/usr/bin/ld: /home/kali/Repos/btstack/port/libusb/../../example/gap_inquiry2.c:109: undefined reference to `gap_remote_name_request'
/usr/bin/ld: /tmp/ccCJ8XMj.o: in function `packet_handler':
/home/kali/Repos/btstack/port/libusb/../../example/gap_inquiry2.c:181: undefined reference to `bd_addr_to_str'
/usr/bin/ld: /home/kali/Repos/btstack/port/libusb/../../example/gap_inquiry2.c:212: undefined reference to `reverse_bd_addr'
/usr/bin/ld: /tmp/ccCJ8XMj.o: in function `btstack_main':
/home/kali/Repos/btstack/port/libusb/../../example/gap_inquiry2.c:253: undefined reference to `hci_set_inquiry_mode'
/usr/bin/ld: /home/kali/Repos/btstack/port/libusb/../../example/gap_inquiry2.c:256: undefined reference to `hci_add_event_handler'
/usr/bin/ld: /home/kali/Repos/btstack/port/libusb/../../example/gap_inquiry2.c:259: undefined reference to `hci_power_control'
collect2: error: ld returned 1 exit status
make: *** [<builtin>: gap_inquiry2] Error 1
```

Why is this happening? It seems like creating your own utilities should be as simple as renaming an example and running with it, but in this case the compiler is throwing several symbol errors. How can this be fixed and what is the best way to create your own derivative of an example?

**To Reproduce**
Steps to reproduce the behavior:
1. Copy `example/gap_inquiry.c` to `example/gap_inquiry.c`
2. In `example/gap_inquiry2.c` change `#define BTSTACK_FILE__ "gap_inquiry.c"` to `#define BTSTACK_FILE__ "gap_inquiry2.c"`
3. In `port/libusb` execute `make gap_inquiry2`

**Expected behavior**
Make produces the executable `gap_inquiry2`

H0r53, 2021-01-29:
After reviewing the make output of `gap_inquiry` it looks like I can compile `gap_inquiry2` with `cc main.o btstack_stdin_posix.o btstack_tlv_posix.o btstack_memory.o btstack_linked_list.o btstack_memory_pool.o btstack_run_loop.o btstack_util.o hci_transport_h2_libusb.o btstack_run_loop_posix.o le_device_db_tlv.o btstack_link_key_db_tlv.o wav_util.o btstack_network_posix.o btstack_audio_portaudio.o btstack_chipset_zephyr.o rijndael.o ad_parser.o hci.o hci_cmd.o hci_dump.o l2cap.o l2cap_signaling.o btstack_audio.o btstack_tlv.o btstack_crypto.o uECC.o sm.o sdp_util.o gatt_sdp.o spp_server.o rfcomm.o bnep.o sdp_server.o device_id_server.o ../../example/gap_inquiry2.c -I. -I../../src/ble -I../../src/classic -I../../src/mesh -I../../src -I../../3rd-party/bluedroid/decoder/include -I../../3rd-party/bluedroid/encoder/include -I../../3rd-party/hxcmod-player -I../../3rd-party/lwip/core/src/include/ -I../../3rd-party/lwip/dhcp-server -I../../3rd-party/md5 -I../../3rd-party/micro-ecc -I../../3rd-party/yxml -I../../platform/lwip -I../../platform/lwip/port -g -std=c99 -Wall -Wmissing-prototypes -Wstrict-prototypes -Wshadow -Wunused-parameter -Wredundant-decls -Wsign-compare -I../../platform/posix -I../../platform/embedded -I../../3rd-party/tinydir -I../../3rd-party/rijndael -I../../chipset/zephyr -I/usr/include/libusb-1.0 -pthread -DHAVE_PORTAUDIO -lm -lusb-1.0 -lportaudio -lasound -lm -lpthread -o gap_inquiry2`.  Thus, the issue must be related to the Makefile and how it finds programs. What is the best way to enable make support for new examples?

H0r53, 2021-01-29:
Editing `Makefile.inc` inside of the example folder enabled support for additional examples.

closed 2021-01-29 by H0r53

mringwal, 2021-01-29:
> Editing `Makefile.inc` inside of the example folder enabled support for additional examples.

That's correct. Sorry, the build systems (there are also some CMakeLists.txt) is a mess. There's a chance it will be cleaned up eventually, but for now adding a new target to example/Makefile.inc or port/*/Makefile is the easiest way.

btw. I think the CMake project in libusb treats all files in example/*.c as targets (minus a few filtered files), so that might work as well.