bluekitchen/btstack #332: SPP counter customization

hananabilabd, 2021-01-30:
I am using SPP counter example and hope to customize it just to have a simple SPP that is able to receive bytes of data. 
while debugging it always enters HAL_UART_RxCpltCallback 
[image]
I see that you are registering a timer  in one_shot_timer_setup , Does removing this line will make it act like normal SPP ?

mringwal, 2021-02-01:
Hi

This is not an issue of the Bluetooth stack. Please post general question on the community mailing list instead. https://groups.google.com/group/btstack-dev

SPP is the profile between two Bluetooth devices. How this data is used, e.g. forwarded to a local UART, is up to the application. In the SPP Counter, the demo app just sends a heartbeat to the remote side. It does not use a local UART.

closed 2021-02-01 by mringwal

hananabilabd, 2021-02-07:
to what remote side does it send to ? 
Is there any documentation for this example ?

mringwal, 2021-02-07:
Well, it does send the heartbeat counter to the remote device that connects to it.
Documentation (extracted from the comments in the C file): https://bluekitchen-gmbh.com/btstack/examples/examples/index.html#spp-server-heartbeat-counter-over-rfcomm