bluekitchen/btstack #334: buffer overflow in pb_adv_handle_transaction_start

hac425xxx, 2021-02-10:
`pb_adv_handler`  read 1 byte to `length`, then pass it to `pb_adv_handle_transaction_start`

```
static void pb_adv_handler(uint8_t packet_type, uint16_t channel, uint8_t *packet, uint16_t size){

    switch(packet[0]){
        case GAP_EVENT_ADVERTISING_REPORT:
            // data starts at offset 12
            data = &packet[12];
            // PDB ADV PDU
            length = data[0];

            // validate length field
            if ((12 + length) > size) return;

            switch (generic_provisioning_control_format){
                case MESH_GPCF_TRANSACTION_START:
                    pb_adv_handle_transaction_start(transaction_nr, &data[7], length-6);
```

`pb_adv_handle_transaction_start` could use size to copy data to `pb_adv_msg_in_buffer`

```

static void pb_adv_handle_transaction_start(uint8_t transaction_nr, const uint8_t * pdu, uint16_t size){

        // store payload
        uint16_t payload_len = size - 4;
        (void)memcpy(pb_adv_msg_in_buffer, &pdu[4], payload_len);
```

the size of pb_adv_msg_in_buffer is 100 bytes.

if `size` is `3` ,  `payload_len`  could be `0xffff`, then memcpy could lead buffer overflow

mringwal, 2021-02-10:
Thanks! Fixed on master 41e943d and develop 205dc73.

Why did you find these issues? Looks like you're writing an analysis tool? :)

closed 2021-02-10 by mringwal

hac425xxx, 2021-02-11:
I am a security researcher， could you please request a cve for this. 

It will be easy to follow the github process

https://docs.github.com/en/github/managing-security-vulnerabilities/publishing-a-security-advisory