bluekitchen/btstack #335: Out-of-Bounds Read  in bnep_l2cap_packet_handler

hac425xxx, 2021-02-10:
`bnep_l2cap_packet_handler` could check size must large than `2`, but then it could copy more bytes from `packet` when size is small.

```
static int bnep_l2cap_packet_handler(uint16_t l2cap_cid, uint8_t *packet, uint16_t size)
{

    if (size < 2) {
        return rc;
    }
    
    bnep_type = BNEP_TYPE(packet[pos]);
    bnep_header_has_ext = BNEP_HEADER_HAS_EXT(packet[pos]); 
    pos ++;
    switch(bnep_type) {
        case BNEP_PKT_TYPE_GENERAL_ETHERNET:
            bd_addr_copy(addr_dest, &packet[pos]);
            pos += sizeof(bd_addr_t);
            bd_addr_copy(addr_source, &packet[pos]);
            pos += sizeof(bd_addr_t);
            network_protocol_type = big_endian_read_16(packet, pos);
            pos += 2;
            break;
```

if size = 3, it could Out-of-Bounds Read and then `pos > size`.

mringwal, 2021-02-10:
Thanks for reporting! Fixed in 99cff9e on master and b85ad95 on develop.

closed 2021-02-10 by mringwal

hac425xxx, 2021-02-11:
I am a security researcher， could you please request a cve for this.

It will be easy to follow the github process

https://docs.github.com/en/github/managing-security-vulnerabilities/publishing-a-security-advisory