bluekitchen/btstack #337: Multiple vulnerabilities in btstack

hac425xxx, 2021-02-11:
## pbap_client_process_vcard_listing out of bound read and stack overflow

`pbap_client_process_vcard_listing` first use `obex_iterator_init_with_response_packet` to parse  `packet` to `obex_iterator_t` , then it could use `obex_iterator_t` to read data from `packet` .

```
static void pbap_client_process_vcard_listing(uint8_t *packet, uint16_t size){
    obex_iterator_t it;
    for (obex_iterator_init_with_response_packet(&it, goep_client_get_request_opcode(pbap_client->goep_cid), packet, size); obex_iterator_has_more(&it) ; obex_iterator_next(&it)){
        uint8_t hi = obex_iterator_get_hi(&it);
        if ((hi == OBEX_HEADER_END_OF_BODY) ||
            (hi == OBEX_HEADER_BODY)){
            uint16_t     data_len = obex_iterator_get_data_len(&it);
            const uint8_t  * data =  obex_iterator_get_data(&it);
            // now try parsing it
            yxml_init(&pbap_client->xml_parser, pbap_client->xml_buffer, sizeof(pbap_client->xml_buffer));
            int card_found = 0;
            int name_found = 0;
            int handle_found = 0;
            char name[PBAP_MAX_NAME_LEN];
            char handle[PBAP_MAX_HANDLE_LEN];
            while (data_len--){
            	yxml_ret_t r = yxml_parse(&pbap_client->xml_parser, *data++);
            	..................
                ..................
            }
```

first it use `obex_iterator_get_data_len` to read some data to `data_len` ，then it could read data  according to `data_len` .

when r is YXML_ATTRVAL , it could copy `xml_parser.data` to `name` , and it could check `strlen(name)`.

```
static void pbap_client_process_vcard_listing(uint8_t *packet, uint16_t size){
    
            uint16_t     data_len = obex_iterator_get_data_len(&it);
            const uint8_t  * data =  obex_iterator_get_data(&it);
            yxml_init(&pbap_client->xml_parser, pbap_client->xml_buffer, sizeof(pbap_client->xml_buffer));
          
            char name[PBAP_MAX_NAME_LEN];
            char handle[PBAP_MAX_HANDLE_LEN];
            while (data_len--){
                yxml_ret_t r = yxml_parse(&pbap_client->xml_parser, *data++);
                
                case YXML_ATTRVAL:
                    if (name_found) {
                        if ((strlen(name) + 4 + 1) >= sizeof(name)) break;
                        strcat(name, pbap_client->xml_parser.data);
                        break;
                    }
                    if (handle_found) {
                        if ((strlen(handle) + 4 + 1) >= sizeof(handle)) break;
                        strcat(handle, pbap_client->xml_parser.data);
                        break;
                    }
                    break;
```

`name` is 32 bytes, so the maximum length of `name` that can enter `strcat` is `26` , the define of `pbap_client->xml_parser.data` 

```
typedef struct {
    
    .....
    .....
	char data[8];  
} yxml_t;
```

so the max length of `pbap_client->xml_parser.data` is  `7`,  so after call `strcat`, the max length of data write to name is `8+26=34` .

There are similar problems when handle the `handle` stack buffer.

## Integer overflow in obex_iterator_init_with_xxx_packet lead oob read

The `obex_iterator_init_with_request_packet` function code is as follows

```
void obex_iterator_init_with_request_packet(obex_iterator_t *context, const uint8_t * packet_data, uint16_t packet_len){
    int header_offset = obex_packet_header_offset_for_opcode(packet_data[0]);
    obex_iterator_init(context, header_offset, packet_data, packet_len);
}
```

packet_data is the received packet, the function first obtains `header_offset` according to `packet_data[0]`, and then passes in the `obex_iterator_init` function.

```
typedef struct obex_iterator {
    const uint8_t * data;
    uint16_t  offset;
    uint16_t  length;
} obex_iterator_t;

static void obex_iterator_init(obex_iterator_t *context, int header_offset, const uint8_t * packet_data, uint16_t packet_len){
    memset(context, 0, sizeof(obex_iterator_t));
    context->data   = packet_data + header_offset;
    context->length = packet_len  - header_offset;
}
```

if `packet_len < header_offset`,  for example `packet_len==header_offset + 1`, then length could be `0xffff`, then it could cause out-of-bounds reading in the subsequent processing flow.

## pbap_packet_handler_goep  oob read

```
        case PBAP_W4_GET_PHONEBOOK_SIZE_COMPLETE:
            uint16_t     data_len = obex_iterator_get_data_len(&it);
            const uint8_t  * data =  obex_iterator_get_data(&it);
            unsigned int i = 0;
            while (i<data_len){
                uint8_t tag = data[i++];
                uint8_t len = data[i++];
				
                i+=len;
            }
```

it first get `data_len` by `obex_iterator_get_data_len`, `data_len` is controllable, if  `data_len`  is large it could lead oob read.

## avrcp_browsing_controller_packet_handler oob read and Integer overflow

```
                case AVRCP_PDU_ID_SET_BROWSED_PLAYER:
                    browsing_connection->uid_counter =  big_endian_read_16(packet, pos);
                    pos += 2;
                    // num_items
                    pos += 4;
                    // charset
                    pos += 2;
                    folder_depth = packet[pos++];

                    for (i = 0; i < folder_depth; i++){
                        uint16_t folder_name_length = big_endian_read_16(packet, pos);
                        pos += 2;
                        packet[pos-1] = AVRCP_BROWSING_MEDIA_ROOT_FOLDER;
                        pos += folder_name_length;
                    }
```

it get `folder_depth` from `packet`, then access the data according to `folder_depth`， if `folder_depth` is large , it could lead to **oob read**.

**int overflow**

```
                case AVRCP_PDU_ID_GET_FOLDER_ITEMS:{
                    switch (avctp_packet_type){
                        case AVRCP_SINGLE_PACKET:
                        case AVRCP_START_PACKET:
                            avrcp_parser_reset(browsing_connection);
                            browsing_connection->uid_counter =  big_endian_read_16(packet, pos);
                            pos += 2;
                            browsing_connection->num_items = big_endian_read_16(packet, pos); //num_items
                            pos += 2;
                            avrcp_browsing_parse_and_emit_element_attrs(packet+pos, size-pos, browsing_connection);
                            break;
```

when `pos > size`, `size-pos` could be `0xfffx`  and pass to `avrcp_browsing_parse_and_emit_element_attrs`, which could lead **oob read** after.

`oob read in report_gatt_services`

```
static void report_gatt_services(gatt_client_t * gatt_client, uint8_t * packet, uint16_t size){
    uint8_t attr_length = packet[1];
    uint8_t uuid_length = attr_length - 4u;
    
    int i;
    for (i = 2; i < size; i += attr_length){
        uint16_t start_group_handle = little_endian_read_16(packet,i);
        uint16_t end_group_handle   = little_endian_read_16(packet,i+2);
        uint8_t  uuid128[16];
        uint16_t uuid16 = 0;

        if (uuid_length == 2u){
            uuid16 = little_endian_read_16(packet, i+4);
            uuid_add_bluetooth_prefix((uint8_t*) &uuid128, uuid16);
        } else {
            reverse_128(&packet[i+4], uuid128);
        }
        emit_gatt_service_query_result_event(gatt_client, start_group_handle, end_group_handle, uuid128);
    }
}

```

when `size=3` and `uuid_length=4`, it could call `reverse_128` to parse `16` byte data from `packet+ 2 + 4` ,  which could lead to `oob read`.

## int overflow and oob in gatt_client_att_packet_handler

```
                case P_W4_READ_BY_TYPE_RESPONSE: {
                    uint16_t pair_size = packet[1];
                    uint16_t offset;
                    uint16_t last_result_handle = 0;
                    for (offset = 2; offset < size ; offset += pair_size){
                        uint16_t value_handle = little_endian_read_16(packet, offset);
                        report_gatt_characteristic_value(gatt_client, value_handle, &packet[offset + 2u], pair_size - 2u);
```

if `pair_size = 1` , then `pair_size - 2 = 0xffff` and pass to `report_gatt_characteristic_value`, it could lead oob read after.

oob 

```
        case ATT_READ_RESPONSE:
            switch (gatt_client->gatt_client_state){
                case P_W4_INCLUDED_SERVICE_UUID_WITH_QUERY_RESULT: {
                    uint8_t uuid128[16];
                    reverse_128(&packet[1], uuid128);
```

if `length of packet < 16` ,  `reverse_128` could oob read.

hac425xxx, 2021-02-11:
I am a security researcher， could you please help me request some cve for this,  thanks a lot!!!

It will be easy to follow the github process

https://docs.github.com/en/github/managing-security-vulnerabilities/publishing-a-security-advisory

mringwal, 2021-02-26:
We've added a fuzzer for the obex_iterator at test/fuzzer, and then re-wrote it. The new version also fixes the 'pbap_packet_handler_goep oob read' OOB as the returned buffer is now guaranteed to be within the original buffer.

All other issues have been fixed on the develop branch as well.

Thanks again for reporting!

closed 2021-02-26 by mringwal

hac425xxx, 2021-03-09:
could you please help me request some cve for this, thanks a lot!!!

It will be easy to follow the github process

https://docs.github.com/en/github/managing-security-vulnerabilities/publishing-a-security-advisory

@mringwal