bluekitchen/btstack #338: Does the apollo2-em9304 port also work on Apollo3 Blue chip?

idea--list, 2021-02-20:
This is rather a question than a feature request.
I can see this BT stack has been also ported to Apollo2 Blue that features an EM9304 to implement BLE.
Question is if the Apollo2 port also works on Apollo3 Blue, which also carries an on-chip EM9304 like the Apollo2 does?

mringwal, 2021-02-20:
I don't know. We don't have an Apollo3 Blue. If it also has an EM9304, the apollo2 port might just work. You probably should check if the UART config matches the one from the Apollo2.

idea--list, 2021-02-21:
I followed the instructions for the Apollo2 found [here](https://bluekitchen-gmbh.com/btstack/ports/existing_ports/index.html#sec:apollo2-em9304Port). By running `./create_examples.py` i could generate all the example folders, however after navigating to the gcc folder of the examples and trying to run `$ run make` i get this message: `bash: run: command not found`

Being a windows user i do not know how to solve that and googling on that gives only tons irrelevant results.
Any help would be much appreciated!

mringwal, 2021-02-21:
I see, it should just be
`make` (the 'run' was from "just run 'make' ...")

idea--list, 2021-02-21:
Thanks, however by typing `make` i get this response:
`make: *** No rule to make target `bin/btstack_em9304_spi_embedded.o', needed by `bin/gap_le_advertisements.elf'.  Stop.` and i get stuck

mringwal, 2021-02-21:
Bummer.

I just tried with a local copy of the AmbiqSuite v1.2.11. There, the current BTstack master is cloned at third-party.
After running ./create examples.py, I change to the folder `AmbiqSuite/boards/apollo2_evb_am_ble/examples/btstack_gap_le_advertisements` and can run `make` there, which compiles the example.

Are you in a real bash shell (e.g. Linux VM, Windows Subsystem for Liinux) and did you to the git clone from there?
If not, please re-try that (maybe Windows path got in the way).

idea--list, 2021-02-21:
I am on a Win10 x64 desktop with AmbiqSuite 2.5.1, i use [Git](https://git-scm.com/downloads) bash.

mringwal, 2021-02-21:
Just had a look at the Apollo3 Blue Datasheet. It only mentions 'BLE Controller' but not EM9304. As it was used before, there's a good chance that it still is, but it would be good to know. 

Then, in the Apollo2 + EM9304 port, the EM9304 was connected via an HCI SPI interface. In the Apollo3 Blue, as the BLE Controller is inside, an internal FIFO is used for communication. This is different from existing HCI interfaces and would require a custom HCI Transport implementation (`hci_transport.h`) to use it with BTstack.

Why the Apollo3 Blue?

idea--list, 2021-02-21:
_Why the Apollo3 Blue?_
Because that MCU has ultra low power consumption and my goal is to airstream an analog signal sampled at 128Hz for several hours running from battery. Sparkfun has a whole line up of boards based on that chip. My Artemis Thing Plus board consumes only 6 mA while doing some basic task, advertising on BLE and also a status LED is on.
My key problem with the Cordio stack included with Mbed is that people report lower throughput than expected, so i am looking for another stack i could take use of. BTstack seems to be a good choice as it also supports my other board the MAX32630FTHR (Mbed dropped support for it's bluetooth module as of V6.0) so BTstack would solve two issues at once.

You are right, in the datasheet they do not mention which bluetooth chip they use in the Apollo3. I just found some hints in the code base. In `AmbiqSuite\boards\apollo3_evb\examples\ble_freertos_barebone\src\hci_apollo_config.h` we can read definition values like AM_BSP_GPIO_EM9304_POWER, AM_BSP_GPIO_CFG_EM9304_POWER, AM_BSP_GPIO_EM9304_RESET.
This made the impression as if they were using the same em9304 as in the Apollo2

However i digged deeper now and found this file: `AmbiqSuite\third_party\cordio\ble-host\sources\hci\ambiq\apollo3\hci_drv_apollo3.h` the 5th line contains:
//! @brief Support functions for the Nationz BTLE radio in Apollo3.

So it turns out the Apollo3 uses a different Bluetooth chip. :confused:
Thanks for the help anyway

mringwal, 2021-02-24:
Thanks for sharing your project. Is the power consumption with BLE on the Apollo3 lower than, e.g. an nRF5x? 

Interesting find about Nationz. As there's no company with that name, I found a press release by CEVA mentioning Nations Technologies Inc., to license BLE IP. On the other hand, it would make sense for Ambiq to directly license BLE IP such as CEVAs for this if they can create a single chip.

Anyway. Given the provided HCI interface and the code in the Cordio stack, it should be possible to port BTstack to the Apollo3, but it will be some work. Someone got BTstack running on an STM32WB55 which also has a dual-core setup where one core implements the BLE Controller.

idea--list, 2021-02-24:
This is interesting. In another press release i have read Ambiq has bought BLE IP from CEVA.
 
`Is the power consumption with BLE on the Apollo3 lower than, e.g. an nRF5x?`
Good question. I do not have an nrf5x based board to test but i found [consumption figures](https://docs.particle.io/datasheets/wi-fi/argon-datasheet) of the WiFi capable Particle Argon added just recently to the doc. That states the board typically consumes 32 mA while transmitting. This sounds too low for WiFi, but much higher than 16mA for BLE @8dbm found in the nrf52840 datasheet.

The Apollo3 based board consumes only 6mA while Tx and doing some work despite also the LED is on by HW design. Though the consumption&range seems to be the same at 0dbm and 4dbm...so i guess power regulation does not work for this board in Mbed Cordio stack.