bluekitchen/btstack #339: Command Identifier is not increased when sending echo request.

jurek4321, 2021-02-21:
**Describe the bug**
I had a problem with sending echo request to android device, I am using test/pts/l2cap_test.c and when im sending echo request to android device only first packet is responded;

**To Reproduce**
Steps to reproduce the behavior:
1. send echo request to android device using l2cap_send_echo_request(handle, (uint8_t *)  "Hello World!", 13);
2. Android device will respond to only first packet.

**Expected behavior**
Android device should respond to every echo request. 
Sending echo request to the same android using l2ping is working fine.

**HCI Packet Logs**
For all Bluetooth issues, please provide a full packet log that includes initial pairing. You need to zip the packet log as GitHub does not know how to handle .pklg files. 

How to get packet log depends on the port:
- For desktop ports, full packet logging is enabled by default and the packet log is available as `/tmp/hci_dump.pklg` (Mac/Linux), or `hci_dump.pklg' (Windows).
- For embedded ports, you can enable #define LOG_INFO in btstack_config.h and uncomment the call to 'hci_dump_open(..)' in the main() function to enable logging over UART Console or Segger RTT. Please capture text output and process using `btstack/tool/create_packet_log.py`. The tool will generate the text file into a .pklg file which can be analysed with Wireshark or Apple's PacketLogger tool.
- For ESP32: BTstack's `port/esp32/integrate.py` script copies the current version of BTstack into the esp-idf folder. To enable packet logging, you need to edit esp-idf/components/btstack/main.c

**Environment:  (please complete the following information):**
 - Current BTstack branch: master
 - Bluetooth Controller ESP32
 - Remote device: sony z3 compact 

**Additional context**
Add any other context about the problem here.

mringwal, 2021-02-22:
Hi. Thanks for reporting. The l2cap_send_echo_request was indeed using a fixed signaling identifier (0x77...).
I've update the develop branch to use a new sig id each time. Could you test it?

mringwal, 2021-07-02:
Closing - merged onto master in the meantime.

closed 2021-07-02 by mringwal