bluekitchen/btstack #340: ESP32 PAN lwIP http server

dr-gino, 2021-02-23:
I can successfully run the PAN lwIP http server demo on the ESP32. But this limits communication between the ESP32 and the bluetooth device (an iPhone in my case). What I want is for the iPhone to act as a gateway to the internet and forward ESP32's IP packets to the its mobile data connection. The PANU demo in client mode seems what this is for but it requires a TAP interface, which the ESP32 doesn't support. 

Is the only way to get this working to implement a TAP like interface on the ESP32 myself? I am asking because since the HTTP demo works fine, IP packets must be getting to the iPhone correctly, it only seems a small step for the iPhone to forward those to the data connection? Why would a full blown TAP interface be required on the ESP32 for this to work? Am I overlooking something.

mringwal, 2021-02-23:
The PAN lwIP http server already uses a full (user land) TCP/IP stack, there's no need for an additional TUN/TAP interface.
We don't have an example for what you're looking for but would be happy to have one as well :0)

Starting from the http server demo, you could setup a PANU client and register such an SDP record as well as with the BNEP adapter. For outgoing connections, you'd look for a NAP server instead of a PANU service. When the BNEP connection is up, Ethernet frames can travel both ways. You then can start a DHCP Client to get an IP address and then try to connect to the Internet.

dr-gino, 2021-02-24:
Thanks for the guidance. I got something very rudimentary going on with the PANU client example, a simple HTTP request is now working.

There's a bit of a practical problem though. I have a small LCD (128x32) connected to my ESP32, and no input buttons.  So there's no way of selecting a phone to use as the NAP. I was thinking about having the phone initiate the pairing so the ESP32 learns the MAC address that it can query in the future.  But since the ESP32 provides no services itself, it just needs internet access, the iPhone doesn't seem to list the ESP32 as a device it is able to pair with?

mringwal, 2021-02-24:
The iPhone filters devices in the Bluetooth settings based on their Class of Device. Just use the CoD for a headset.
Also, the panu_demo does not make itself discoverable, so try this

    gap_discoverable_control(1);
    gap_set_class_of_device(0x200408);

dr-gino, 2021-02-24:
Thanks for the quick response.
I set the CoD and made it discoverable like you suggested, it shows up in the list on the iPhone but when tapping the name it says it failed and the '<Local Name Of ESP32> is not supported.' I was expecting it to pair?

Logs show this while tapping the name:
```
[00:05:34.716] LOG -- hci.c.2404: Connection_incoming: 64:70:33:1F:E3:D2, type 1
[00:05:34.716] LOG -- hci.c.194: create_connection_for_addr 64:70:33:1F:E3:D2, type fd
[00:05:34.716] LOG -- hci.c.4146: sending hci_accept_connection_request
[00:05:34.736] LOG -- hci.c.2428: Connection_complete (status=0) 64:70:33:1F:E3:D2
[00:05:34.736] LOG -- hci.c.2448: New connection: handle 128, 64:70:33:1F:E3:D2
[00:05:34.736] LOG -- hci.c.5018: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1
[00:05:34.756] LOG -- hci.c.1968: Remote features 03, bonding flags f8
[00:05:34.766] LOG -- l2cap.c.1989: create channel 0x3ffbb1bc, local_cid 0x0041
[00:05:34.766] LOG -- hci.c.5191: gap_request_security_level requested level 0, planned level 0, current level 0
[00:05:34.776] LOG -- hci.c.5049: hci_emit_security_level 0 for handle 80
[00:05:34.776] LOG -- l2cap.c.2337: l2cap - security level update for handle 0x0080
[00:05:34.786] LOG -- l2cap.c.2347: channel 0x3ffbb1bc, cid 0041 - state 5: actual 0 >= required 0?
[00:05:34.796] LOG -- l2cap.c.1084: L2CAP_EVENT_INCOMING_CONNECTION addr 64:70:33:1F:E3:D2 handle 0x80 psm 0x1 local_cid 0x41 remote_cid 0x109
[00:05:34.806] LOG -- l2cap.c.2600: L2CAP_ACCEPT_CONNECTION local_cid 0x41
[00:05:34.816] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x41
[00:05:34.816] LOG -- l2cap.c.1251: l2cap_start_rtx for local cid 0x41
[00:05:34.846] LOG -- l2cap.c.2809: L2CAP signaling handler code 4, state 11
[00:05:34.846] LOG -- l2cap.c.2656: Remote MTU 256
[00:05:34.846] LOG -- l2cap.c.2809: L2CAP signaling handler code 5, state 11
[00:05:34.846] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x41
[00:05:34.856] LOG -- l2cap.c.2740: l2cap_signaling_handle_configure_response
[00:05:34.866] LOG -- l2cap.c.1050: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr 64:70:33:1F:E3:D2 handle 0x80 psm 0x1 local_cid 0x41 remote_cid 0x109 local_mtu 1691, remote_mtu 256, flush_timeout 0
[00:05:34.926] LOG -- l2cap.c.2809: L2CAP signaling handler code 6, state 12
[00:05:34.926] LOG -- l2cap.c.1078: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x41
[00:05:34.926] LOG -- l2cap.c.1995: free channel 0x3ffbb1bc, local_cid 0x0041
[00:05:34.936] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x41
[00:05:35.986] LOG -- hci.c.950: Connection closed: handle 0x80, 64:70:33:1F:E3:D2
[00:05:35.986] LOG -- hci.c.5018: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0
```

Also, after doing this a couple of times on the iPhone (and iPad) the device disappears from the list entirely and it takes a reboot of the iOS device for it to show up again. A USB Bluetooth dongle picks up the ESP32 just fine which makes me think iOS hides Bluetooth devices that it thinks are behaving badly? Are you familiar with such behavior?

mringwal, 2021-02-24:
I would suggest to add an SDP record for a PANU client, if you haven't.

dr-gino, 2021-02-24:
> 
> 
> I would suggest to add an SDP record for a PANU client, if you haven't.

It was already in there, I then changed it to NAP instead and it paired successfully. Adding an SDP for A2DP sink also worked. Apparently PANU client isn't satisfying the iPhone when pairing. It feels kind of hacky though to add fake services. Is this the conventional thing to do for devices that don't have any keyboard/input buttons?

dr-gino, 2021-02-25:
I have a weird situation going on: Code that I have been working on runs fine on one ESP32, but when uploading the exact same binary to another ESP32, it doesn't. Could this be hardware related?
My code just wait for a phone to pair and then tries to access the pone's NAP. This works on one ESP32 but on another I can't even get the phone to pair, the phone says it can't pair an I need to make sure the device is switched on an within range. Trying to pair from my laptop gives a similar result.

This is the log from the ESP32 while trying to pair with it from my iPhone:
```
[00:00:00.922] LOG -- hci.c.2117: Manufacturer: 0x0060
[00:00:00.922] LOG -- hci.c.2023: local name:
[00:00:00.922] LOG -- hci.c.1702: Received local name, need baud change 0
[00:00:00.932] LOG -- hci.c.2135: Local supported commands summary bf - 1f
[00:00:00.932] LOG -- hci.c.2078: Local Address, Status: 0x00: Addr: B4:E6:2D:CD:EF:0B
[00:00:00.942] LOG -- hci.c.2040: hci_read_buffer_size: ACL size module 1021 -> used 1021, count 9 / SCO size 255, count 4
[00:00:00.952] LOG -- hci.c.2103: Packet types cc18, eSCO 1
[00:00:00.962] LOG -- hci.c.2106: BR/EDR support 1, LE support 1
[00:00:00.972] LOG -- hci.c.5074: BTSTACK_EVENT_DISCOVERABLE_ENABLED 1
[00:00:00.972] LOG -- hci.c.2060: hci_le_read_buffer_size: size 251, count 10
[00:00:00.982] LOG -- hci.c.1810: Supported commands 30
[00:00:00.982] LOG -- hci.c.2067: hci_le_read_maximum_data_length: tx octets 251, tx time 2120 us
[00:00:00.992] LOG -- hci.c.2073: hci_le_read_white_list_size: size 12
[00:00:01.002] LOG -- hci.c.1560: hci_init_done -> HCI_STATE_WORKING
[00:00:01.002] LOG -- hci.c.4949: BTSTACK_EVENT_STATE 2
[00:00:01.012] LOG -- main.c.346: transport: configure SCO over HCI, result 0x0000
[00:00:47.722] LOG -- hci.c.2404: Connection_incoming: 64:70:33:1F:E3:D2, type 1
[00:00:47.722] LOG -- hci.c.194: create_connection_for_addr 64:70:33:1F:E3:D2, type fd
[00:00:47.732] LOG -- hci.c.4146: sending hci_accept_connection_request
[00:00:47.792] LOG -- hci.c.2428: Connection_complete (status=0) 64:70:33:1F:E3:D2
[00:00:47.792] LOG -- hci.c.2448: New connection: handle 128, 64:70:33:1F:E3:D2
[00:00:47.792] LOG -- hci.c.5018: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1
[00:00:47.852] LOG -- hci.c.1968: Remote features 03, bonding flags f8
[00:00:53.862] LOG -- l2cap.c.1989: create channel 0x3ffbb230, local_cid 0x0041
[00:00:53.862] LOG -- hci.c.5191: gap_request_security_level requested level 0, planned level 0, current level 0
[00:00:53.872] LOG -- hci.c.5049: hci_emit_security_level 0 for handle 80
[00:00:53.882] LOG -- l2cap.c.2337: l2cap - security level update for handle 0x0080
[00:00:53.882] LOG -- l2cap.c.2347: channel 0x3ffbb230, cid 0041 - state 5: actual 0 >= required 0?
[00:00:53.892] LOG -- l2cap.c.1084: L2CAP_EVENT_INCOMING_CONNECTION addr 64:70:33:1F:E3:D2 handle 0x80 psm 0x1 local_cid 0x41 remote_cid 0x708
[00:00:53.902] LOG -- l2cap.c.2600: L2CAP_ACCEPT_CONNECTION local_cid 0x41
[00:00:53.912] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x41
[00:00:53.922] LOG -- l2cap.c.1251: l2cap_start_rtx for local cid 0x41
[00:00:59.872] LOG -- l2cap.c.1989: create channel 0x3ffbb2a4, local_cid 0x0042
[00:00:59.872] LOG -- hci.c.5191: gap_request_security_level requested level 0, planned level 0, current level 0
[00:00:59.882] LOG -- hci.c.5049: hci_emit_security_level 0 for handle 80
[00:00:59.882] LOG -- l2cap.c.2337: l2cap - security level update for handle 0x0080
[00:00:59.892] LOG -- l2cap.c.2347: channel 0x3ffbb230, cid 0041 - state 11: actual 0 >= required 0?
[00:00:59.902] LOG -- l2cap.c.2347: channel 0x3ffbb2a4, cid 0042 - state 5: actual 0 >= required 0?
[00:00:59.912] LOG -- l2cap.c.1084: L2CAP_EVENT_INCOMING_CONNECTION addr 64:70:33:1F:E3:D2 handle 0x80 psm 0x1 local_cid 0x42 remote_cid 0x808
[00:00:59.922] LOG -- sdp_server.c.573: busy, queing incoming cid 0x0042, now 1 waiting
[00:01:03.932] LOG -- l2cap.c.1226: l2cap_rtx_timeout for local cid 0x41
[00:01:03.932] LOG -- l2cap.c.1050: L2CAP_EVENT_CHANNEL_OPENED status 0x69 addr 64:70:33:1F:E3:D2 handle 0x80 psm 0x1 local_cid 0x41 remote_cid 0x708 local_mtu 1691, remote_mtu 672, flush_timeout 0
[00:01:03.942] LOG -- l2cap.c.1995: free channel 0x3ffbb230, local_cid 0x0041
[00:01:03.952] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x41
[00:01:05.882] LOG -- l2cap.c.1989: create channel 0x3ffbb230, local_cid 0x0043
[00:01:05.882] LOG -- hci.c.5191: gap_request_security_level requested level 0, planned level 0, current level 0
[00:01:05.882] LOG -- hci.c.5049: hci_emit_security_level 0 for handle 80
[00:01:05.892] LOG -- l2cap.c.2337: l2cap - security level update for handle 0x0080
[00:01:05.902] LOG -- l2cap.c.2347: channel 0x3ffbb2a4, cid 0042 - state 9: actual 0 >= required 0?
[00:01:05.902] LOG -- l2cap.c.2347: channel 0x3ffbb230, cid 0043 - state 5: actual 0 >= required 0?
[00:01:05.912] LOG -- l2cap.c.1084: L2CAP_EVENT_INCOMING_CONNECTION addr 64:70:33:1F:E3:D2 handle 0x80 psm 0x1 local_cid 0x43 remote_cid 0x908
[00:01:05.932] LOG -- l2cap.c.2600: L2CAP_ACCEPT_CONNECTION local_cid 0x43
[00:01:05.932] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x43
[00:01:05.942] LOG -- l2cap.c.1251: l2cap_start_rtx for local cid 0x43
[00:01:15.952] LOG -- l2cap.c.1226: l2cap_rtx_timeout for local cid 0x43
[00:01:15.952] LOG -- l2cap.c.1050: L2CAP_EVENT_CHANNEL_OPENED status 0x69 addr 64:70:33:1F:E3:D2 handle 0x80 psm 0x1 local_cid 0x43 remote_cid 0x908 local_mtu 1691, remote_mtu 672, flush_timeout 0
[... cut ...]
[00:01:15.962] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x43
[00:01:18.322] LOG -- l2cap.c.1078: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x42
[00:01:18.322] LOG -- l2cap.c.1995: free channel 0x3ffbb2a4, local_cid 0x0042
[00:01:18.332] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x42
[00:01:18.332] LOG -- hci.c.950: Connection closed: handle 0x80, 64:70:33:1F:E3:D2
[00:01:18.342] LOG -- hci.c.5018: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0
```

Any idea what might be going on? I'll try to get a hold of yet another ESP32 to test on as well.

mringwal, 2021-02-26:
Is the log from the working or the non-working one?
There are 2 Clock options for ESP32. My SparkFun Thing uses 24 Mhz, most other have a different Clock.

dr-gino, 2021-02-27:
That is the log from the not working one.  Both logs are actually similar up to the point in the last line of the log where it says:

```hci.c.950: Connection closed: handle 0x80, 64:70:33:1F:E3:D2```

In the working log, instead of connection closed, it says:

```hci.c.2578: IO Capability Request received, stack bondable 1, io cap 3
hci.c.5163: gap_mitm_protection_required_for_security_level 0```

Is it the phone that decides to close the connection?

Are you talking about the actual xtal used? Both ESP32's run at 160Mhz, one works and one doesn't If i remember correctly they both have an an internal 40Mhz.

dr-gino, 2021-02-27:
I got another ESP32 dev. board and it works fine. So it now works on 2/3 dev. boards. On the 3rd dev. board even the Arduino Bluetooth Serial example which uses the Bluedroid stack, gets stuck during pairing. So safe to say it's totally not btstack related.
But boy would that have been a debugging nightmare if that were my only dev. board!

Still somewhat curious what might cause Bluetooth to malfunction like that, the WiFi part works fine on that particular dev. board.

closed 2021-02-28 by dr-gino

ei-ke, 2021-05-24:
@dr-gino I've found this issue while searching for a way to connect an ESP32 to the Internet via an mobile phone.
Would you be willing to share your example code with me?