bluekitchen/btstack #341: Possible values for: sdp_event_query_complete_get_status

dr-gino, 2021-02-28:
Where can I find a list of possible status code this function returns. I get that 0 means success and from the sdp query example 4 means the device didn't respond/wasn't in range. But I have also seen it return 0x67, what is the meaning of that particular code, and possibly others?

mringwal, 2021-03-01:
Most of all 'status' codes use the Bluetooth enumeration, with BTstack adding custom values.
https://github.com/bluekitchen/btstack/blob/master/src/bluetooth.h#L146

A good suggestion how we could have avoided you asking about the status codes resp. where could we have added a comment?

closed 2021-03-01 by mringwal

dr-gino, 2021-03-01:
> 
> 
> Most of all 'status' codes use the Bluetooth enumeration, with BTstack adding custom values.
> https://github.com/bluekitchen/btstack/blob/master/src/bluetooth.h#L146
> 
> A good suggestion how we could have avoided you asking about the status codes resp. where could we have added a comment?

In the examples often the return value isn't explicitly compared against one of the error constants you just showed me. e.g.:
```
if (sdp_event_query_complete_get_status(packet)){
    printf("SDP query failed 0x%02x\n", sdp_event_query_complete_get_status(packet));
    break;
} 
````
I think if instead it read something like:
```
if (sdp_event_query_complete_get_status(packet) != ERROR_CODE_SUCCESS) {
...
}
```

I would have been able to look up ERROR_CODE_SUCCESS from within the IDE/Editor and see the other possible return values.

mringwal, 2021-03-03:
Thanks for your feedback. We've started to use ERROR_CODE_SUCCESS instead of 0 more in the code, but making sure the defines are used in the examples is a good idea.