bluekitchen/btstack #342: LE Connection 0401: disconnect, reason 13

ahsankaukab3, 2021-03-09:
Hi @mringwal ,

I am running gatt_streamer_server example with bluetooth stack (STM32-F411) and i am getting the following issue:

LE Connection 0401: connected - connection interval 48.75 ms, latency 0
LE Connection 0401: request 15 ms connection interval
[00:00:00.000] LOG -- hci.c.856: acl_handler called with non-registered handle 1025!
LE Connection 0401: connection update - connection interval 7.50 ms, latency 0
[00:00:00.000] LOG -- hci.c.856: acl_handler called with non-registered handle 1025!
[00:00:00.000] LOG -- hci.c.856: acl_handler called with non-registered handle 1025!
LE Connection 0401: disconnect, reason 13

//////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

My port_main is as follow:

void port_main(void){
 btstack_memory_init();
    btstack_run_loop_init(btstack_run_loop_embedded_get_instance());
    // init HCI
    hci_init(hci_transport_h4_instance(btstack_uart_block_embedded_instance()), (void*) &config);
    hci_set_chipset(btstack_chipset_cc256x_instance());
    // setup TLV Flash Sector implementation
    const hal_flash_bank_t * hal_flash_bank_impl = hal_flash_bank_stm32_init_instance(
    		&hal_flash_bank_context,
    		HAL_FLASH_BANK_SIZE,
			HAL_FLASH_BANK_0_SECTOR,
			HAL_FLASH_BANK_1_SECTOR,
			HAL_FLASH_BANK_0_ADDR,
			HAL_FLASH_BANK_1_ADDR);
    const btstack_tlv_t * btstack_tlv_impl = btstack_tlv_flash_bank_init_instance(
    		&btstack_tlv_flash_bank_context,
			hal_flash_bank_impl,
			&hal_flash_bank_context);
     // // setup global tlv
     btstack_tlv_set_instance(btstack_tlv_impl, &btstack_tlv_flash_bank_context);
    // setup Link Key DB using TLV
    const btstack_link_key_db_t * btstack_link_key_db = btstack_link_key_db_tlv_get_instance(btstack_tlv_impl, 
   &btstack_tlv_flash_bank_context);
    hci_set_link_key_db(btstack_link_key_db);
    // setup LE Device DB using TLV
    le_device_db_tlv_configure(btstack_tlv_impl, &btstack_tlv_flash_bank_context);
    #ifdef HAVE_HAL_AUDIO
   // setup audio 
   btstack_audio_sink_set_instance(btstack_audio_embedded_sink_get_instance());
    btstack_audio_source_set_instance(btstack_audio_embedded_source_get_instance());
    #endif
    hci_set_chipset(btstack_chipset_cc256x_instance());
     // hand over to btstack embedded code
    btstack_main(0, NULL);
    // go
    btstack_run_loop_execute();
}

mringwal, 2021-03-09:
This is the issue tracker and this not an issue with BTstack. That's why I've suggested the mailing list.

Is this a CC2564?

ahsankaukab3, 2021-03-09:
Yes you are right its CC2564MODA

ahsankaukab3, 2021-03-09:
what can be done to solve this issue?

mringwal, 2021-03-09:
It looks like that's an issue of the CC2564 when changing the connection params.
Please try to comment theses lines in the example:

                    // request min con interval 15 ms for iOS 11+ 
                    printf("- LE Connection %04x: request 15 ms connection interval\n", con_handle);
                    gap_request_connection_parameter_update(con_handle, 12, 12, 0, 0x0048);

ahsankaukab3, 2021-03-09:
The same issue retained. 
The output logs are as follow:

[00:00:00.000] LOG -- btstack_chipset_cc256x.c.167: cc256x: using default init script
[00:00:00.000] LOG -- btstack_tlv_flash_bank.c.443: found bank 0
[00:00:00.000] LOG -- btstack_tlv_flash_bank.c.211: test erased: bank 0, offset 56
[00:00:00.000] LOG -- btstack_tlv_flash_bank.c.472: Flash clean after last found tag
[00:00:00.000] LOG -- btstack_tlv_flash_bank.c.487: write offset 56
[00:00:00.000] LOG -- btstack_tlv.c.48: TLV Instance 0x802a090
[00:00:00.000] LOG -- le_device_db_tlv.c.159: num valid le device entries 0
[00:00:00.000] LOG -- btstack_chipset_cc256x.c.167: cc256x: using default init script
[00:00:00.000] LOG -- sm.c.4351: sm: generate new ec key
[00:00:00.000] LOG -- att_db.c.209: att_set_db 0x8029e79
[00:00:00.000] LOG -- hci.c.1138: Num LE Peripheral roles: 0 of 1
[00:00:00.000] LOG -- hci.c.3539: hci_power_control: 1, current mode 0
[00:00:00.000] LOG -- btstack_chipset_cc256x.c.167: cc256x: using default init script
[00:00:00.000] LOG -- hci.c.4949: BTSTACK_EVENT_STATE 1
[00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] LOG -- hci.c.2117: Manufacturer: 0x000d
[00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] LOG -- hci.c.2023: local name:
[00:00:00.000] LOG -- hci.c.1702: Received local name, need baud change 0
[00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] L [... cut ...]
[00:00:00.000] CMD => [00:00:00.000] EVT <= [00:00:00.000] EVT <= [00:00:00.000] ACL <= [00:00:00.000] LOG -- hci.c.856: acl_handler called with non-registered handle 1025!
[00:00:00.000] EVT <= - LE Connection 0401: connection update - connection interval 7.50 ms, latency 0
[00:00:00.000] ACL <= [00:00:00.000] LOG -- hci.c.856: acl_handler called with non-registered handle 1025!
[00:00:00.000] ACL <= [00:00:00.000] LOG -- hci.c.856: acl_handler called with non-registered handle 1025!
[00:00:00.000] EVT <= - LE Connection 0401: disconnect, reason 13

mringwal, 2021-03-09:
hm.. maybe your port is not correct

this is suspicious 
[00:00:00.000] LOG -- hci.c.856: acl_handler called with non-registered handle 1025!

Did you set MAX_NR_HCI_CONNECTIONS to 0?

Also: please implement the hal_get_time_ms() function to get time stamps and also find a way to get proper log output: the actual data of the packets is not printed and the newline is missing.

mringwal, 2021-07-02:
Closing - no further information from author.

closed 2021-07-02 by mringwal