bluekitchen/btstack #344: Callback when notifications are disable on Nordic SPP

dlech, 2021-03-12:
**Is your feature request related to a problem? Please describe.**

We need to be able to detect when notifications are disabled on the Nordic UART characteristic. We need to be able to disable and re-enable notifications without disconnecting the device due to an edge case of Chromium + BlueZ.

**Describe the solution you'd like**

The existing callback could be modified to not ignore this case.

**Describe alternatives you've considered**

I think a change like this could be made with minimal impact on existing users. 

```diff
diff --git a/src/ble/gatt-service/nordic_spp_service_server.c b/src/ble/gatt-service/nordic_spp_service_server.c
index a08016430..3d4a9470c 100644
--- a/src/ble/gatt-service/nordic_spp_service_server.c
+++ b/src/ble/gatt-service/nordic_spp_service_server.c
@@ -89,9 +89,7 @@ static int nordic_spp_service_write_callback(hci_con_handle_t con_handle, uint16
        }
        if (attribute_handle == nordic_spp_tx_client_configuration_handle){
                nordic_spp_tx_client_configuration_value = little_endian_read_16(buffer, 0);
-               if (nordic_spp_tx_client_configuration_value){
-                       client_callback(con_handle, NULL, 0);
-               }
+               client_callback(nordic_spp_tx_client_configuration_value ? con_handle : HCI_CON_HANDLE_INVALID, NULL, 0);
        }
 
        return 0;
```

```diff
diff --git a/example/nordic_spp_le_counter.c b/example/nordic_spp_le_counter.c
index f5b02ee29..1310fcbb2 100644
--- a/example/nordic_spp_le_counter.c
+++ b/example/nordic_spp_le_counter.c
@@ -125,9 +125,12 @@ static void heartbeat_handler(struct btstack_timer_source *ts){
 /* LISTING_END */
 
 static void nordic_data_received(hci_con_handle_t tx_con_handle, const uint8_t * data, uint16_t size){
-    if (size == 0 && con_handle == HCI_CON_HANDLE_INVALID ){
+    if (tx_con_handle == HCI_CON_HANDLE_INVALID) {
+        printf("Notifications disabled for handle 0x%04x\n", con_handle);
+        con_handle = HCI_CON_HANDLE_INVALID;
+    } else if (size == 0 && con_handle == HCI_CON_HANDLE_INVALID ){
         con_handle = tx_con_handle;
-        printf("Connected with handle 0x%04x\n", con_handle);        
+        printf("Connected with handle 0x%04x\n", con_handle);
     } else {
         printf("RECV: ");
         printf_hexdump(data, size);
```

**Additional context**

I can make a pull request if this seems reasonable.

mringwal, 2021-03-16:
Hi! Thanks for the observation and the suggestions. I think our idea to signal the 'notifications enabled' via the data received callback wasn't a good one and I wouldn't try to extend it even more. A better way could be to provide events for notifications enabled/disabled. The open question would be if the current callback should be dropped or not (while a 'compatibility' mode could still be provided). The same applies to the u-blox service (which is more or less the same).

I can try to implement this in the next days.

dlech, 2021-03-16:
That is actually what I did with a custom service that started as a copy of the UART service. I just added a 2nd callback for "client configuration" that is registered in the init function, then the appropriate callback is called depending on the attribute...

```C
static int pybricks_service_write_callback(hci_con_handle_t con_handle, uint16_t attribute_handle, uint16_t transaction_mode, uint16_t offset, uint8_t *buffer, uint16_t buffer_size) {
    UNUSED(transaction_mode);
    UNUSED(offset);
    UNUSED(buffer_size);

    if (attribute_handle == pybricks_value_handle) {
        client_callback(con_handle, &buffer[0], buffer_size);
    }
    else if (attribute_handle == pybricks_client_configuration_handle) {
        pybricks_client_configuration_value = little_endian_read_16(buffer, 0);
        client_configuration_callback(con_handle, pybricks_client_configuration_value);
    }

    return 0;
}
```

mringwal, 2021-03-22:
Hi David. Please check out the new version I've pushed on the develop branch and let me know if it works for you.

dlech, 2021-03-23:
Hi Matthias, I've integrated the changes into our code and it is working. Thanks for the fix.

mringwal, 2021-03-23:
Excellent. I'm glad the API is better now (no hacks).

closed 2021-03-23 by mringwal