bluekitchen/btstack #349: ESP32: spp streamer/client hang while connecting, time out after a min

panthony22, 2021-04-08:
Hello,
I'm not sure if this is a bug or something I'm doing wrong, but I seem to be having trouble trying to get two ESP32 devices to connect using the SPP streamer implementation. Both devices are ESP32 DevkitC boards. After programming one with the streamer example and the other with the client example, they appear to find one another and open a communication channel. However, it seems they get stuck on communicating their MTU and eventually just close the channel. Once in a great while they manage to complete the connection process and begin transmitting to one another, but it is inconsistent. I've attached logs that were printed to gtkterm. Is there any way I can remedy this problem?
[spp_streamer_client_log.txt](https://github.com/bluekitchen/btstack/files/6276421/spp_streamer_client_log.txt)
[spp_streamer_log.txt](https://github.com/bluekitchen/btstack/files/6276423/spp_streamer_log.txt)

mringwal, 2021-04-09:
Hi. The SPP Streamer demo did work the last time I've tried.
In the log, you get error 105 = 0x69 -> see src/bluetooth.h -> L2CAP_CONNECTION_RESPONSE_RESULT_RTX_TIMEOUT

Could you enable HCI packet logging in their respective main() and post new log files?

panthony22, 2021-04-10:
Thank you for replying. Here are the HCI packet logs. I think I should mention that I was having issues programming the boards in the first place.

After running integrate_btstack.py, I would run "make" and "make flash" but the terminal output would end with an error about multiple undefined references to esp_core_dump functions. For example:
`
/home/peter/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/peter/btstack/btstack-master/port/esp32/example/spp_streamer_client/build/espcoredump/libespcoredump.a(core_dump_common.o):(.literal.esp_core_dump_switch_task_stack_to_isr+0x4): undefined reference to 'esp_core_dump_get_isr_stack_end' `
...
```
collect2: error: ld returned 1 exit status
make: *** [/home/peter/esp/esp-idf/make/project.mk:564: /home/peter/btstack/btstack-master/port/esp32/example/spp_streamer_client/build/spp_streamer_client.elf] Error 1
```

To get around this, I ran idf.py from ESP-IDF to build and flash to the boards after running "make flash." This let me program the boards, but then there are issues connecting like I described earlier.

[hci_dump_spp_streamer_client.txt](https://github.com/bluekitchen/btstack/files/6289121/hci_dump_spp_streamer_client.txt)
[hci_dump_spp_streamer.txt](https://github.com/bluekitchen/btstack/files/6289122/hci_dump_spp_streamer.txt)

mringwal, 2021-04-10:
Hi. Well, the compile errors are suspicious, but should not cause problems later. Maybe update to the latest version of esp-idf and btstack (master) and start over.

In the logs, there's just an Link Layer disconnect: the connection breaks without a reason. There are config options for the clock or such. Please check there. Otherwise, please try to run some Bluetooth examples from esp-idf and post on their issue tracker. There's nothing wrong in HCI logs.

mringwal, 2021-07-02:
Closing - no further information from issue author.

closed 2021-07-02 by mringwal