bluekitchen/btstack #350: hid_keyboard_demo barely functional when connected to Windows 10

sidit77, 2021-04-18:
**Describe the bug**
The connection between hid_keyboard_demo and Windows 10 seems to be extremely slow. For example if I quickly type "Hello Word from hid_keyboard_demo" there is a delay of ~1s between the arrival of each keystroke (Full text takes >10s to arrive)
[hci_dump.zip](https://github.com/bluekitchen/btstack/files/6331136/hci_dump.zip)
. This delay is very inconsistent and feels a lot like a traffic jam for the lack of a better word.

**To Reproduce**
Steps to reproduce the behavior:
1. Run example 'hid_keyboard_demo'
2. Connect to example from device with Windows 10
3. Start typing quickly

**Expected behavior**
Keystrokes arrive in a timely manner and with somewhat constant latency.
[hci_dump.zip](https://github.com/bluekitchen/btstack/files/6331139/hci_dump.zip)

**Environment:  (please complete the following information):**
 - Current BTstack branch: [master]
 - Bluetooth Controller [WinUSB]
 - Remote device: [Laptop with Windows 10]

**Additional context**
I tested the demo in various configurations:
| Keyboard Device      | Remote Device           | Status |
| -------------------- | ----------------------- | ------ |
| Desktop (Windows 10) | Laptop1 (Windows 10)    | Bad    |
| Desktop (Windows 10) | Laptop2 (Windows 10)    | Bad    |
| RaspberryPi Zero W   | Laptop1 (Windows 10)    | Bad    |
| Desktop (Windows 10) | OnePlus 8T (Android 11) | Good   |
| Desktop (Windows 10) | Laptop1 (Ubuntu 20.04)  | Good   |

Additionally I also did some testing with [another keyboard emulator](https://github.com/quangthanh010290/keyboard_mouse_emulate_on_raspberry) which uses BlueZ:
| Keyboard Device      | Remote Device           | Status |
| -------------------- | ----------------------- | ------ |
| RaspberryPi Zero W   | Laptop1 (Windows 10)    | Good   |
| RaspberryPi Zero W   | Laptop1 (Ubuntu 20.04)  | Good   |

mringwal-logitech, 2021-04-19:
Thanks for reporting. We mainly test against iOS and Mac, but it should work with Win 10 as well :)
From the logs, it's unclear what's going on. We'll have a look but first need to find a Windows 10 machine/setup to test with.

mringwal, 2021-04-19:
Thanks for reporting. We mainly test against iOS and Mac, but it should work with Win 10 as well :)
From the logs, it's unclear what's going on. We'll have a look but first need to find a Windows 10 machine/setup to test with.

mringwal, 2021-04-20:
Just got a chance to try this with a Win 10 VM. Good news: it reproduces here as well, characters are received slower than 1 character/s.

I've tried to disable Sniff mode, but no change. Could you try to get an HCI dump from the BlueZ emulator for comparison? (really, there's nothing unexpected in BTstack's HCI log, but there must be a difference).

sidit77, 2021-04-20:
I'll try to capture hci log as soon as possible. Something I've noticed white looking for the difference is that the hid packages look different:
BTStack: [0xa1, mod, 0, 0, kc1, kc2, ...]
BlueZ: [0xa1, 0x1, mod, 0, kc1, kc2, ...]
I'm not yet very familiar with bluetooth, but as far as I could determine the second format is used when the "boot protocol" is enabled (?). Could this be reason? I've seen that there is a variable in the hid_demo for turning in boot protocol but it seemed to have no effect.

mringwal, 2021-04-21:
It's possible that the HID reports look different (HID is extremely flexible), but Windows can correctly decode it. It does not explain how we get to almost 1000 ms latency per key pressed. It must be queued somewhere, and it is not in the stack. (the log shows reasonable delta). I didn't look at the 'num completed packets" event as this indicates how many packets have been acknowledged by the remote side.

sidit77, 2021-04-21:
Ok, I've captured the BlueZ log (`tcpdump -i bluetooth0 -w bluetooth0_capture_12_12.pcap` hope that's the right command).
[bluetooth0_capture_12_12.zip](https://github.com/bluekitchen/btstack/files/6350460/bluetooth0_capture_12_12.zip)

mringwal, 2021-04-21:
The capture only shows HCI Events, but neither data nor HCI Commands. There's a Command Complete for Exit Sniff Mode, which indicates that a HCI Exit Sniff Mode was sent before :)

That makes sense, but I also got the high latency with Sniff mode being disabled.

sidit77, 2021-04-21:
I've finally figured out why `hcidump` failed to capture all packets: I forgot to run it as root :/.
That being said here is the full log: 
[capture.zip](https://github.com/bluekitchen/btstack/files/6353991/capture.zip)

mringwal, 2021-04-26:
Thanks for the log. In there, I don't see any HCI commands that could be related to this latency. In this case, there must be a difference in the data provided by our demo compared to the other keyboard emulator.

In the SDP Record, the other keyboard emulator has set the sniff (sub)rating params. 

On the develop branch, the sniff params are set by BTstack as well. Feel free to give it a quick try.

sidit77, 2021-05-03:
I am sorry for the delay,  the samples of the develop branch segfault for me when I try to run them so I had to get them working first. Unfortunately I experience the same latency problems on the develop branch as on the master branch.

mringwal, 2021-05-03:
Could you post the location / backtrace of the segfault? It shouldn't :) and/or what you've needed to fix.

sidit77, 2021-05-03:
This is the gdb output:
```
Thread 1 received signal SIGSEGV, Segmentation fault.
0x00007ff7d126210d in hci_init (transport=0xffffffffd129b340, config=0x0)
    at ../../src/hci.c:3194
3194        transport->register_packet_handler(&packet_handler);
(gdb) backtrace
#0  0x00007ff7d126210d in hci_init (transport=0xffffffffd129b340, config=0x0)
    at ../../src/hci.c:3194
#1  0x00007ff7d1251853 in main (argc=1, argv=0x28c111a1a00) at main.c:142
```

Unfortunately I couldn't find the issue and resorted to compiling the qt-usb ports instead.

mringwal, 2021-05-03:
Thanks, will have a look at this later and at least fix the seg fault on windows.

mringwal, 2021-07-12:
Minor update. I've verified with an USB protocol analyzer between the WIndows VM and the USB dongle that the Bluetooth communication is flawless - the key presses are received without any latency. So, the latency is caused by the Windows Bluetooth/HID driver. The question is still, why this happens. It does not happen with e.g. a Logitech Ultrathin iPad keyboard I had lying around.

mringwal, 2021-07-12:
Found it!

The keycode was reported at the second position of the 6 available positions for key codes. Apparently, Windows behaves weird (introducing about 1 s latency per character) when the keycode is reported there instead of in the first position.

Fixed on develop, please try and report back.

sidit77, 2021-07-12:
Works perfectly now.

 Respect that you were able to figure it out, issues like this are among the most rage/despair inducing things when writing software.

closed 2021-07-12 by sidit77

mringwal, 2021-07-12:
Thanks! The annoying thing was the knowledge that unless someone at Microsoft wanted to annoy me, there must be a simple explanation for this... but what is it...