bluekitchen/btstack #351: L2CAP Data Channel Packets Stop

meavydev, 2021-04-23:
We are using the L2CAP Data Channel Server on a Raspberry Pi to Android 11 using a createInsecureL2capChannel().
After a period of time the sending just stops and the channel->send_sdu_buffer isn't getting set to NULL.

**To Reproduce**
Run a L2CAP Data Channel stream and keep going until it stops.

**Expected behavior**
It should keep sending packets.

**HCI Packet Logs**
[hci_dump.pklg.zip](https://github.com/bluekitchen/btstack/files/6364616/hci_dump.pklg.zip)

**Environment:  (please complete the following information):**
 - Current BTstack branch: Master
 
 - Bluetooth Controller : 
 H4, 3000000, BT_REG_EN at GPIO 128, Reset Controller
Local name: BCM4345C0

 - Remote device: Android 11 Pixel 3A XL

mringwal, 2021-04-23:
Hi. Did you look at the hci_dump file in Wireshark? There, it does not stop sending until the end of the file.
You could try to turn of the other device to check if you get a Disconnect event in the log.

meavydev, 2021-04-23:
There is no disconnect. 
The BTstack side just stops sending and doesn't get a L2CAP_EVENT_LE_PACKET_SENT / L2CAP_EVENT_LE_CAN_SEND_NOW.
The reason the WireShark log stops is that the BTStack stops sending ;)

meavydev, 2021-04-23:
If you look at the previous frame, you see a:
Frame 22149: 59 bytes on wire (472 bits), 59 bytes captured (472 bits)
Bluetooth
PacketLogger Note
    Type: Note (0xfc)
    Info: l2cap.c.3932: len 175, pos 2 => payload 175, credits 61722
I'm assuming we would expect to see this as a response to one of these:
 22143	121.621269	Raspberr_27:3f:11 (NavenioPeripheral)	7e:a0:63:6d:5a:6c ()	L2CAP	196	Sent Connection oriented channel
So at the end of the Wireshark log, you can see the "Sent Connection oriented channel" but no L2CAP "credits" line.
Without the credits line, it appears to stop, as no more packets can be sent, as the channel->send_sdu_buffer isn't set to NULL and no events are generated to allow the next packet to be sent.

meavydev, 2021-04-23:
Just one more thing.
We have more or less the same Pi code to generate samples with Bluez L2CAP CoC and this works.
The reason we like BTStack is that it allows the LE Advertising to work up until the point where the L2Cap socket is created from Android, whereas with Bluez it stops when you start opening the L2CAP socket.

mringwal, 2021-04-26:
Thanks for the additional comments. Could you try to disconnect the remote side to see if the interface to the Bluetooth Controller is well?

In the log for the last send command, there's a 0x7d (L2CAP_EVENT_LE_PACKET_SENT) followed by a 0x7c (L2CAP_EVENT_LE_CAN_SEND_NOW). however, there's no 0x6e (HCI_EVENT_TRANSPORT_PACKET_SENT), which indicates that the ACL packet was successfully sent to the Bluetooth Controller. That's with Apple's PacketLogger tool.

With Wireshark, an error is reported about an invalid large packet. There, the 0x6e HCI_EVENT_TRANSPORT_PACKET_SENT is shown, followed by an event with code 0x00 (haven't seen that one yet).

If the stack gets the 'trasnport sent' and the 'hci num complete' - which apparently it does - there's no reason to stop sending.

To find out if this is related to the used Bluetooth Controller or the stack, could you try to run the port/libusb on the Raspi with an USB dongle, or, the libusb port on a desktop/laptop computer?

meavydev, 2021-04-27:
I'm on holiday this week, but will have a look next week.
Just to confirm the same Android code works with a Bluez L2CAP CoC connection without stopping, so it is only failing with BTStack.
We have switched to Bluez now, as advertising is working with Bluez on the latest Pi update, so I will only be checking given we are not planning on using BTStack any more.

mringwal, 2021-04-27:
Enjoy your vacation!

I understand that it works with bluez l2cap coc, but would like to figure out why it stops with BTstack in your setup which includes the Raspi and your test app on Anroid.

If I run the le_data_channel_client/server examples on my mac with two USB dongles, it doesn't stop. That's why I'm wondering about the root of this. E.g. if you like, could you run the GATT Streamer server on the Pi? You can connect with a GATT Browser and just enable notifications.

meavydev, 2021-05-06:
I'm sorry, but for some reason BTStack has stopped talking to the Pi.
I have done some updates on the Pi, but it's just not initialising any more in BTStack, so LE Advertising isn't turning on.
My guess is something in the raspi port is no longer working, but I haven't been allocated any time to look at BTStack (given Bluez is working)...
So I guess you can close this issue, until someone starts trying L2Cap with a Pi to Android.

meavydev, 2021-05-07:
Ah actually I forgot to turn off the Bluez services, which is why it wasn't starting.
I'm afraid I won't be able to spend any more time investigating, but here is your .pklg and also a WireShark BTSniffer capture showing the transfer just stopping. 
Android is still connected and the same code works fine with Bluez.
[WireShark.zip](https://github.com/bluekitchen/btstack/files/6440215/WireShark.zip)

mringwal, 2021-05-09:
Thanks for the traces. I'll try the LE Data Channel example when I have an Rasperry Pi connected the next time.

closed 2021-05-09 by mringwal