bluekitchen/btstack #352: `gap_connect_cancel()` does not cleanly disconnect le direct connections

eldonrivers, 2021-04-25:
**Overview**
A new connection started with `gap_connect()` using an `BD_ADDR_TYPE_LE_PUBLIC`, then cancelled with `gap_connect_cancel()` does not reset `hci_stack->le_connecting_request`. As a result additional attempts to open a new connection with `gap_connect()` will fail and the error "le connection already active" is logged.

I'm wondering if when the `ERROR_CODE_UNKNOWN_CONNECTION_IDENTIFIER` case is handled in `event_handle_le_connection_complete` if `hci_stack->le_connecting_request` should be set to `LE_CONNECTING_IDLE`?

**To Reproduce**
1. Initiate a new connection with `gap_connect()` to a known `BD_ADDR_TYPE_LE_PUBLIC` device address.
2. Immediately cancel with `gap_connect_cancel()`
3. Retry the connection with `gap_connect()`
4. `gap_connect()` returns immediately and the error "le connection already active" is logged.

**Expected behavior**
After cancelling a connect it should be possible to start a new connection.

**Environment**
 - Current BTstack branch: 1.3.2
 - Bluetooth Controller [CSR 4.0 USB]
 - Remote device: [TI CC2640]

**Additional context**
We are using a connection attempt timeout and retry logic that happens to make this case especially painful. I suspected for some time (and am still wondering) that I'm doing something wrong how I'm attempting to disconnect in this case (ie: "`gap_connect_cancel()` is the wrong choice here") but no alternative is clear.

mringwal, 2021-04-26:
Hi! Thanks for reporting and the preliminary analysis. You're totally right that `hci_stack->le_connecting_request` should be set the to `LE_CONNECTING_IDLE` on connection complete with `ERROR_CODE_UNKNOWN_CONNECTION_IDENTIFIER`.

Could you re-test with the current version on develop branch?

In general, there's rarely a need to cancel an outgoing connection to a specific device to only try again later. If you have more than one devices you'd like to connect, you can use the whitelist to connect to one of them (call `gap_whitelist_add` for all of them, then call `gap_connect_with_whitelist`).

eldonrivers, 2021-04-27:
Something still isn't working. I'm reluctant to debug into this code because I'm unfamiliar and don't want to make bad assumptions. I noticed when looking into the `hci_emit_le_connection_complete(...)` called from `gap_connect_cancel()` that the local hci event handler doesn't end up getting called to handle this event at all? My local client handler (registered through `hci_add_event_handler`) does get called with the expected `HCI_SUBEVENT_LE_CONNECTION_COMPLETE` event, but the `hci.c` code that resets `hci_stack->le_connecting_request` never seems to happen. I have previously seen this code run as a result of `gap_connect_cancel()`, but maybe the timing was different and the connection was in a different state?

I'm supporting a device test fixture and being able to quickly respond to connect and disconnect requests would be ideal, including specifically interrupting/disconnecting from an in-progress connection handshake. For now my strategy is to only run `gap_connect()` once the device is found advertising, then if a disconnect is required before we have a connection handle I'm just waiting for the connection to complete before disconnecting. It would be great if it was possible to support immediate shutdown even if the connection hasn't been fully established.

eldonrivers, 2021-04-27:
Ok, so the new change seems to fix the `SENT_CREATE_CONNECTION` case in `gap_connect_cancel()`, but the `SEND_CREATE_CONNECTION` case still needs to end up having `hci_stack->le_connecting_request` set back to `LE_CONNECTING_IDLE`. I'm unsure if it makes sense to do it right in `gap_connect_cancel()` or if anything else needs to happen first. Putting it there seems to work?

mringwal, 2021-04-27:
Hi again. Good catch about the state reset if gap_connect_cancel is called right after gap_connect (before the stack got a chance to send the hci le create connection command). Yes, `hci_stack->le_connecting_request` should be reset to `LE_CONNECTING_IDLE` in this case.

LE Connection setup is rather minimal. The Peripheral sends and Advertisements and waits a bit for a Connection Indication. When the Central sends the Connection Ind, the Controller reports Connection Complete. ON the Peripheral, Connection Complete is reported when the Connection Ind is received.

Please let me know if you can start/stop outgoing le connections as you like, or if there's a sequence left that doesn't work.

eldonrivers, 2021-05-26:
I've managed to get this deployed and can confirm that (at least in our use cases so far) this works well now. Thanks :)

closed 2021-05-26 by eldonrivers

mringwal, 2021-05-26:
Thanks for the thumbs up.