bluekitchen/btstack #356: a2dp_source_demo can't run in background on Ubuntu 16.04.6

lnjin, 2021-05-12:
**Describe the bug**
a2dp_source_demo can't run in background on Ubuntu 16.04.6

**To Reproduce**
Steps to reproduce the behavior:
1. try to run a2dp_source_demo in backgournd with the following command:
 ./a2dp_source_demo &   or  ./a2dp_source_demo 2>&1 /tmp/a2dp.log &

root@lina-OptiPlex-790:/home/lina/Git/btstack/port/libusb# ./a2dp_source_demo 2>&1 /tmp/bt.log &[3] 16955
root@lina-OptiPlex-790:/home/lina/Git/btstack/port/libusb# Packet Log: /tmp/hci_dump.pklg

[3]+  Stopped                 ./a2dp_source_demo /tmp/bt.log 2>&1

**Expected behavior**
a2dp_source_demo can run in background successfully

**HCI Packet Logs**
[hci_dump.zip](https://github.com/bluekitchen/btstack/files/6462902/hci_dump.zip)

**Environment:  (please complete the following information):**
 - Current BTstack branch: [develop]
 - Bluetooth Controller Choseal on Ubuntu 16.04.6
 - 

**Additional context**
Add any other context about the problem here.

mringwal, 2021-05-12:
No idea. But your output says "Stopped" and the hci dump basically is empty. Maybe it hangs because there's no stdio available. Why do you need to run it in background? You could use a separate tab or something like 'screen' btw.

closed 2021-05-12 by mringwal

lnjin, 2021-05-12:
Hi, mringwal,

Thanks for your feedback, I want to start the a2dp_source_demo in script but without blocking other process, so I need it to run in background. Could you help provide any idea on such request?

mringwal, 2021-05-12:
Well, it looks like the process stops when you put it into background. You can check /tmp/hci_dump.pklg to see if it runs at all. If I run your commands, I get "[2]  + 21525 suspended (tty output)". So it's not an issue with the BTstack example, only a how to use my operating system.

Maybe you can run it as a daemon, e.g. using supervisord?
We have port/daemon, which basically is a BTstack application that provides a TCP Socket. This one can run in background.

hm.. you could try to remove HAVE_BTSTACK_STDIN from btstack_config.h as that removes the STDIN processing.