bluekitchen/btstack #357: sdp query crash when called on GAP_EVENT_DEDICATED_BONDING_COMPLETED event on ESP32

dr-gino, 2021-05-16:
BTStack: 1.3.2
ESP-IDF: 4.2

I'm trying to bond with a device using `gap_dedicated_bonding` like so:

```
void hci_packet_handler(uint8_t packet_type, uint16_t channel, uint8_t *packet, uint16_t size)
{
    switch (hci_event_packet_get_type(packet)) {    
    case BTSTACK_EVENT_STATE:
        if (btstack_event_state_get_state(packet) == HCI_STATE_WORKING) {
            gap_dedicated_bonding(remote_device_, 1);
        }
        break;
    case GAP_EVENT_DEDICATED_BONDING_COMPLETED:
        if (gap_event_dedicated_bonding_completed_get_status(packet) == 0) {
            sdp_client_query_uuid16(&handle_sdp_client_query_result, remote_device_, BLUETOOTH_SERVICE_CLASS_NAP);
        }
        break;
    default:
        break;
    }
}
```

This should start an SDP query after the bonding completed successfully, however this causes a crash when initiating the query.
Here's the stack trace:

```
0x4008325a: lock_acquire_generic at newlib/locks.c:142
0x4008fcf7: panic_abort at esp_system/panic.c:330
0x40090395: esp_system_abort at esp_system/system_api.c:106
0x4009428a: abort at newlib/abort.c:46
0x4008325a: lock_acquire_generic at newlib/locks.c:142
0x4008337d: _lock_acquire_recursive at newlib/locks.c:170
0x40134cca: _puts_r at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/puts.c:89 (discriminator 2)
0x40134d2f: puts at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/puts.c:129
0x400e18b2: report_recv_called_from_isr at btstack/btstack_port_esp32.c:104
0x400e18c3: host_recv_pkt_cb at btstack/btstack_port_esp32.c:128
0x401552fa: vhci_send at ??:?
0x40146e3a: r_eif_send at ??:?
0x4001791d: ?? ??:0
0x401475a5: hci_tx_start at hci_tl.c:?
0x40147682: r_hci_tl_send at ??:?
0x400188f5: ?? ??:0
0x40147037: r_hci_send_2_host_hack at ??:?
0x400175ee: ?? ??:0
0x40017688: ?? ??:0
0x4008e4ee: vhci_recv_end at vhci.c:?
0x4008e6cd: r_vhci_isr at ??:?
0x40146de7: r_eif_isr at ??:?
0x4008e6e4: r_eif_isr_wrapper at intc.c:?
0x40082cc1: _xt_lowint1 at freertos\xtensa/xtensa_vectors.S:1105
0x400876e2: cause_sw_intr_to_core_wrapper at bt/controller/bt.c:769
0x4008e495: vhci_set_interrupt at vhci.c:?
0x40155499: API_vhci_host_send_packet at ??:?
0x40143860: esp_vhci_host_send_packet at bt/controller/bt.c:1008
0x400e1731: transport_send_packet at btstack/btstack_port_esp32.c:287
0x400db851: hci_send_acl_packet_fragments at btstack\src/hci.c:723
0x400db97b: hci_send_acl_packet_buffer at btstack\src/hci.c:785
0x400df7b7: l2cap_send_signaling_packet at btstack\src/l2cap.c:1287
0x400df8fc: l2cap_run_for_classic_channel at btstack\src/l2cap.c:1517
0x400e0477: l2cap_run at btstack\src/l2cap.c:1874
0x400e1235: l2cap_create_channel at btstack\src/l2cap.c:2060
0x400d93d1: sdp_client_query at btstack\src\classic/sdp_client.c:756
0x400d93fd: sdp_client_query_uuid16 at btstack\src\classic/sdp_client.c:761
0x400d310f: hci_packet_handler at src/bttr.c:420
```

Any idea as to what's going on here?

mringwal, 2021-05-16:
Hi. Thanks for reporting. BTstack's HCI packet handler `host_recv_pkt_cb` is called from Interrupt context, which should not happen (it should only be called from thread context).

Please report this issue on the ESP-IDF Issue tracker together with the back trace above and link the issue here.

dr-gino, 2021-05-17:
As you can see I reported the issue. Looking forward to their response.
However, I have been experimenting with BTstack for a while now and never noticed this issue. The packet handler callback gets called a lot, I would have expected it to at least trigger a crash by now, but it never did until I tried the piece of code shown in my opening post.
Did I just get (un)lucky and found a sequence to reliably trigger it?

dr-gino, 2021-05-17:
Somebody replied to the issue stating that calling `printf` caused the crash, and they are correct. But the call to `printf` from `report_recv_called_from_isr` was merely to show a warning that the packet callback has been unexpectedly called from within an ISR. This doesn't seem to solve the root cause, after all BTstack's packet handler simply returns 0 when it detects it's called from inside an ISR.
I skimmed through the ESP IDF docs regarding `esp_vhci_host_register_callback` but it doesn't mention that the callbacks can be expected to not be called in the context of an ISR?

mringwal, 2021-05-17:
Well, yes, I'd say you've just got (un)lucky with this code. 
I don't remember if the 'not called from ISR context' is documented anywhere, but I had this issue before and so far they have fixed it. 

If that's not possible/Espressiv is not willing to fix it, but documents that it can be called either from ISR or from thread context, we need to work around that / use other data structures for that. (IRQ-safe ringbuffers are possible, but it would be easier / less error-prone if we could avoid that).

mringwal, 2021-05-17:
btw. the previous one was here:
https://github.com/espressif/esp-idf/issues/1322

dr-gino, 2021-05-17:
> Well, yes, I'd say you've just got (un)lucky with this code.
> I don't remember if the 'not called from ISR context' is documented anywhere, but I had this issue before and so far they have fixed it. [...]

I'm not too hopeful of Espressif properly fixing things quickly.
Taking a look at `host_recv_pkt_cb` I notice that it's accessing a ring buffer protected by a mutex, What more would be required other than calling `xSemaphoreGiveFromISR` instead of `xSemaphoreGive` when called from an interrupt? It seems like there's a lot of protection in place already?

mringwal, 2021-05-18:
The host_send_pkt_available_cb would work from IRQ using as it is.  host_recv_pkt_cb uses a semaphore to protect the receive packet ringbuffer. If the BTstack thread has already locked the mutex, xSemaphoreTakeFromISR would fail. You cannot block an interrupt. 

As I said, to make it work from interrupt would require to use a ring-buffer like data structure. Looking at the FreeRTOS docu, there is a (new) stream buffer API, which matches the use case. 

I'd suggest to wait for Espressif to decide if this is a bug in their code, or, if the ESP32 receive code needs to be rewritten.
Btw. the Bluedroid code calls osi_calloc, and I'd be surprised if their allocator can be called safely from ISR context...

dr-gino, 2021-05-19:
I played around with the code some more and noticed that when the `host_rcv_pkt_cb` is called from ISR context it contains an error packet: `0x04 0x10 0x01 0x00`. This is the first and only packet received immediately after calling `sdp_client_query_uuid16`.
This is a hardware error correct? According to this comment https://github.com/espressif/esp-idf/issues/1322#issuecomment-350255260 error events could get submitted from within an ISR. So that issue hasn't been addressed properly but I'm more curious as to why an error would occur in the first place?

mringwal, 2021-05-19:
Good work. Yes, 0x04 0x10 0x01 0x00 is  a hardware error. If we get that there isn't much we can do and it doesn't matter much if that one is called from ISR.

Could you enable packet logging (see app_main() in main.c) and post a packet log here?

dr-gino, 2021-05-20:
That log from above is where I commented out the `if (xPortInIsrContext())` check in the `host_recv_pkt_cb` function and protected the ring buffer with a critical section so it couldn't possibly be interrupted by an interrupt. I wanted to do so testing.

To make sure I didn't accidentally mess anything up I just tried it again with a fresh copy of BTstack and with only the following modifications:
- changed `printf` to `ets_printf` in `report_recv_called_from_isr` so it doesn't crash.
- Rewrote the `if (xPortInIsrContext)` check to the following below:

```
if (xPortInIsrContext()){
    report_recv_called_from_isr();
    // Added by me
    ets_printf("len: %u, data:", len);
    for (uint16_t i = 0; i < len; i++) {
        ets_printf(" %02X", data[i]);
    }
    ets_printf("\n");
    // End Moditications
    return 0;
}
```
No other modications were made and this yields the packet log below.
This seems to be exactly the same (minus the LOG from hci.c because the packet never makes it there since the `return 0` in the ISR check is still in place)

```
Bonding completed, calling sdp_client_query_uuid16.
[00:00:05.245] ACL => 81 00 0C 00 08 00 01 00 02 01 04 00 01 00 41 00 
host_recv_pkt_cb called from ISR!
len: 4, data: 04 10 01 00
[00:00:05.255] EVT <= 70 18 6A D2 EB 1C 33 70 64 81 00 01 00 41 00 00 00 9B 06 A0 02 00 00 00 00 00 
[00:00:05.265] EVT <= 61 01 00 
[00:00:05.265] EVT <= 6E 00
```

Regarding why I'm doing the dedicated bonding: This was a small piece of test code meant to be part of a larger piece.
But the main problem boils down to this:

I have a project that needs internet access, it has no screen or buttons but it needs access to the Internet over Bluetooth.
The problem is how does my ESP32 get to know the MAC of the iPhone it's supposed to connect to? 
I have something working where I call `gap_set_class_of_device` with a headphone CoD to indicate it's a Headphone (which it totally isn't but that's the only way to get it to show up on the iPhone at all). I also add the PANU SDP record but this isn't enough to get the iPhone to pair because when I select it, a few seconds later the iPhone says the device is not supported. It'll only pair if I also add an A2DP record using `a2dp_sink_create_sdp_record`. This is far from ideal but at least now the iPhone pairs with the ESP32. The iPhone then indicates it couldn't connect to the ESP32, this is expected since I'm not actually implementing the A2DP at all, I just need it to pair since I have no other way of learning the iPhone's MAC. Once it's paired the ESP32 then does an SDP query to find out if the iPhone actually is a NAP and gets the L2CAP PSM. (I took this code from the PANU demo).
I then connect by calling `bnep_connect` and set up an ethernet interface on the ESP32 and can do TCP/IP requests. This all works but It feels error prone because of the fake Headphones CoD and the A2DP record that I don;t actually support.

So I wanted to try a test setup to add a small screen and a button to initiate the pairing from the ESP32's side so I don't have to mess around with this fake headphone/A2DP stuff. But when calling `bnep_connect` without bonding first, nothing happens, I kind of expected the iPhone to pop up a message asking me if it was ok for the ESP32 to connect? That's why I started to do the dedicated bonding process. Once that is completed I then always do an SDP query to find the L2CAP PSM for the bnep connection.

It's quite a long story and I'm far from a Bluetooth expert so I'd appreciate your take on this. Ideally I don't want to add a screen and button just for the sake of Bluetooth, but I also don't want to make something hacky that might break in the future because I'm faking a headphone/A2DP support.

mringwal, 2021-05-21:
Thanks for your efforts to get a clean log, appreciated.

As for iPhone bonding without a full UI, I would also recommend to pick a CoD that the iPhone deems worthy of trying to connect and check the SDP records. I'm not sure if you need to setup the BNEP connection. Have a look at the pan_lwip_http_demo - which is a bit the other way round, but works (iPhone connects and pairs, then TCP/IP is working). Please give it a try and let me know, what problems you run into using the same approach for Internet access.

bnep_connect should starting pairing: it uses the current security level from gap_get_security_level() which is initialized to LEVEL_2 (encrypted without MITM protection). If not, please file a new issue and attache the packet log.

mringwal, 2021-05-21:
FWIW, I've fixed the bug creating an outgoing connection without security requirements (SDP) from the HCI Disconnect callback on the develop branch. Feel free to test on ESP32. It works correctly on my desktop setup with an USB Bluetooth dongle.

dr-gino, 2021-05-21:
Thanks for your insights.
I changed the CoD to 0x20300 as per the pan_lwip example and added both the NAP and PANU SDP records. This makes the iPhone pair and connect, but no Internet connection is possible. The iPhone doesn't seem to route anything to the Internet and expects the ESP32 to run a DHCP server while not running on itself so the ESP32 can't even get an IP address. If I then try to establish a PANU connection while the iPhone already has the NAP connection, it seems to refuse it. See log below:

```
[00:00:19.037] LOG -- bnep.c.1609: BNEP_CONNECT addr 64:70:33:1C:EB:D2
[00:00:19.037] LOG -- bnep.c.732: bnep_max_frame_size_for_l2cap_mtu:  1691 -> 1676
[00:00:19.047] LOG -- l2cap.c.2024: L2CAP_CREATE_CHANNEL addr 64:70:33:1C:EB:D2 psm 0xf mtu 1691 -> local mtu 1691, sec level 2
[00:00:19.057] LOG -- l2cap.c.1989: create channel 0x3fff0d7c, local_cid 0x0044
[00:00:19.067] LOG -- l2cap.c.1921: connection complete con_handle 0080 - for channel 0x3fff0d7c cid 0x0044
[00:00:19.077] LOG -- hci.c.5191: gap_request_security_level requested level 2, planned level 0, current level 2
[00:00:19.087] LOG -- hci.c.5049: hci_emit_security_level 2 for handle 80
[00:00:19.097] LOG -- l2cap.c.2337: l2cap - security level update for handle 0x0080
[00:00:19.097] LOG -- l2cap.c.2347: channel 0x3fff07cc, cid 0042 - state 1: actual 2 >= required 0?
[00:00:19.107] LOG -- l2cap.c.2347: channel 0x3fff0840, cid 0043 - state 12: actual 2 >= required 2?
[00:00:19.117] LOG -- l2cap.c.2347: channel 0x3fff0d7c, cid 0044 - state 6: actual 2 >= required 2?
[00:00:19.127] LOG -- l2cap.c.1995: free channel 0x3fff07cc, local_cid 0x0042
[00:00:19.137] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x42
[00:00:19.147] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x44
[00:00:19.147] LOG -- l2cap.c.1251: l2cap_start_rtx for local cid 0x44
[00:00:19.157] LOG -- hci.c.5074: BTSTACK_EVENT_DISCOVERABLE_ENABLED 0
[00:00:19.237] LOG -- l2cap.c.2809: L2CAP signaling handler code 3, state 10
[00:00:19.247] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x44
[00:00:19.247] LOG -- l2cap.c.1050: L2CAP_EVENT_CHANNEL_OPENED status 0x67 addr 64:70:33:1C:EB:D2 handle 0x80 psm 0xf local_cid 0x44 remote_cid 0x0 local_mtu 1691, remote_mtu 672, flush_timeout 0
[00:00:19.257] LOG -- bnep.c.1271: L2CAP_EVENT_CHANNEL_OPENED for BLUETOOTH_PSM_BNEP, status 103
[00:00:19.277] LOG -- bnep.c.116: BNEP_EVENT_CHANNEL_OPENED status 0x67 bd_addr: 64:70:33:1C:EB:D2, handler 0x400d3bbc
[00:00:19.287] LOG -- l2cap.c.1995: free channel 0x3fff0d7c, local_cid 0x0044
[00:00:19.287] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x44
```

If I add the NAP SDP record but don't actually accept the incoming BNEP connection (i.e.: not call `bnep_register_service(bnep_handler, BLUETOOTH_SERVICE_CLASS_NAP, 1691)`) the iPhone will pair but then show a message it couldn't connect. The ESP32 can then immediately connect to the iPhone and access the internet. So that's something but to the user this seems like it failed. I'd rather just have the NAP connection lingering in the background and then initiate a connection from the ESP32 so the "failed to connect" error doesn't show up on the iPhone.

mringwal, 2021-05-26:
Hi. What happens if you only add the PANU SDP record? If the iPhone sees the NAP, it might think it can get internet from your device instead of vice-versa.

dr-gino, 2021-05-26:
> 
> 
> Hi. What happens if you only add the PANU SDP record? If the iPhone sees the NAP, it might think it can get internet from your device instead of vice-versa.

I just tried that. The iPhone discovers the ESP32 but when tapping it, after ~2 seconds it shows a message saying something along the lines of: Connecting failed, Device 'ESP32' is not supported.
The BTstack log shows this:

```
[00:00:20.967] LOG -- hci.c.2404: Connection_incoming: 64:70:33:1C:EB:D2, type 1
[00:00:20.967] LOG -- hci.c.194: create_connection_for_addr 64:70:33:1C:EB:D2, type fd
[00:00:20.977] LOG -- hci.c.4146: sending hci_accept_connection_request
[00:00:20.997] LOG -- hci.c.2428: Connection_complete (status=0) 64:70:33:1C:EB:D2
[00:00:20.997] LOG -- hci.c.2448: New connection: handle 128, 64:70:33:1C:EB:D2
[00:00:20.997] LOG -- hci.c.5018: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 1
[00:00:21.017] LOG -- hci.c.1968: Remote features 03, bonding flags f8
[00:00:21.027] LOG -- l2cap.c.1989: create channel 0x3fff06d4, local_cid 0x0041
[00:00:21.027] LOG -- hci.c.5191: gap_request_security_level requested level 0, planned level 0, current level 0
[00:00:21.037] LOG -- hci.c.5049: hci_emit_security_level 0 for handle 80
[00:00:21.047] LOG -- l2cap.c.2337: l2cap - security level update for handle 0x0080
[00:00:21.047] LOG -- l2cap.c.2347: channel 0x3fff06d4, cid 0041 - state 5: actual 0 >= required 0?
[00:00:21.057] LOG -- l2cap.c.1084: L2CAP_EVENT_INCOMING_CONNECTION addr 64:70:33:1C:EB:D2 handle 0x80 psm 0x1 local_cid 0x41 remote_cid 0x108
[00:00:21.067] LOG -- l2cap.c.2600: L2CAP_ACCEPT_CONNECTION local_cid 0x41
[00:00:21.077] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x41
[00:00:21.087] LOG -- l2cap.c.1251: l2cap_start_rtx for local cid 0x41
[00:00:21.087] LOG -- l2cap.c.2809: L2CAP signaling handler code 4, state 11
[00:00:21.097] LOG -- l2cap.c.2656: Remote MTU 256
[00:00:21.097] LOG -- l2cap.c.2809: L2CAP signaling handler code 5, state 11
[00:00:21.107] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x41
[00:00:21.117] LOG -- l2cap.c.2740: l2cap_signaling_handle_configure_response
[00:00:21.127] LOG -- l2cap.c.1050: L2CAP_EVENT_CHANNEL_OPENED status 0x0 addr 64:70:33:1C:EB:D2 handle 0x80 psm 0x1 local_cid 0x41 remote_cid 0x108 local_mtu 1691, remote_mtu 256, flush_timeout 0
[00:00:21.207] LOG -- l2cap.c.2809: L2CAP signaling handler code 6, state 12
[00:00:21.207] LOG -- l2cap.c.1078: L2CAP_EVENT_CHANNEL_CLOSED local_cid 0x41
[00:00:21.207] LOG -- l2cap.c.1995: free channel 0x3fff06d4, local_cid 0x0041
[00:00:21.217] LOG -- l2cap.c.1242: l2cap_stop_rtx for local cid 0x41
[00:00:22.427] LOG -- hci.c.950: Connection closed: handle 0x80, 64:70:33:1C:EB:D2
[00:00:22.427] LOG -- hci.c.5018: BTSTACK_EVENT_NR_CONNECTIONS_CHANGED 0
```

It looks like the iPhone is connecting, but I'm unsure to which service/what it expects at the other end? BTstack then decides to close the channel?

mringwal, 2021-05-26:
The thing I'm wondering is this: you can share Internet (e.g Cellular) from one iOS with another iOS device. I would assume that an iPod only has PANU and no NAP. Now, the question is if the iPod gets internet when pairing the iPhone to the iPod (compared to the other way around). Btw.. you have enabled Bluetooth Tethering on the iPhone, didn't you?

dr-gino, 2021-05-26:
Yes I do have Bluetooth tethering enabled on the iPhone.
Regarding sharing Internet between two iOS devices: I can not get my iPad to show up in the list of available Bluetooth devices on the iPhone. On the iPad however I can see the iPhone just fine, connect with it and then use it's data connection.

Does the log I posted tell you anything about what the iPhone is trying to achieve? It seems like we're almost there?

mringwal, 2021-05-26:
>  I can not get my iPad to show up in the list of available Bluetooth devices 
Hm... I would assume that the ESP32 is in the same position as the iPad, so if the iPhone doesn't see the iPad that's a problem/

I always/need the log incl. HCI dump. Up there, it looks like the iPhone checked the SDP records and then does nothing. The disconnect is most likely initiated by the remote device. 

I guess you could set up a HFP HF or an A2DP Sink profile, just so the iPhone is happy. So, it would be an internet enabled headphone!

mringwal, 2021-05-26:
Closing this, as the issue with calls from IRQ is at least documented and BTstack tries to avoid it as good as possible.

closed 2021-05-26 by mringwal