bluekitchen/btstack #358: Random timeout during rfcomm connection (New?)

Matheus-Garbelini, 2021-06-01:
### Describe the bug
Sometimes **repeatedly** starting RFCOMM connection gets blue kitchen stuck into an active connection as if BT state machine is idle even after receiving L2CAP Configure Response from the target. The symptom is the user packet handler does not seem to receive RFCOMM_EVENT_CHANNEL_OPENED.

The capture below shows what happens during the hang.
[image]

The over-the-air capture shows as follows:
[image]

### To Reproduce
Steps to reproduce the behavior:
1. Compile the inline example below
2. Start the example with `./sdp_rfcomm_query -u /dev/ttyUSB1 -a 41:42:00:00:00:ab --iocap 3 --authreq 3 --bounding 1`
You may omit `-u` since you don't necessarily need to run this over h4.
3. The example discovers SDP services, connections to each one, and then disconnects.
4. This is repeated in a cycle. But at some point, the connection gets stuck as if BT state machine gets idle.

<details> 
  <summary>Inline BT code: Custom code modified from sdp_rfcomm_query.c</summary>
  
```C
#include "btstack_config.h"

#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include <signal.h>

#include "hci_cmd.h"
#include "btstack_run_loop.h"

#include "btstack.h"

#define NUM_SERVICES 20

int conf_iocap = SSP_IO_CAPABILITY_NO_INPUT_NO_OUTPUT;
int conf_auth_req = SSP_IO_AUTHREQ_MITM_PROTECTION_NOT_REQUIRED_GENERAL_BONDING;
int conf_enable_bounding = 1;
int conf_discoverable = 0;
char conf_pin[5] = {'0', '0', '0', '0', '\x00'};

int flag_bounding_start = 0;
int flag_bounding_done = 0;
int flag_rfcomm_connection_start = 0;
int flag_rfcomm_connection_done = 0;

int var_rfcomm_opened_channels = 0;
int var_rfcomm_closed_channels = 0;
uint16_t var_rfcom_cids[20];

static btstack_timer_source_t timer_rfcomm_query;
static btstack_timer_source_t timer_rfcomm_connection;
static btstack_timer_source_t timer_ssp_pairing;
static bd_addr_t remote_addr;

static struct
{
    uint8_t channel_nr;
    char service_name[SDP_SERVICE_NAME_LEN + 1];
} services[NUM_SERVICES];

static uint8_t service_index = 0;
static uint8_t current_service_index = 0;

static btstack_packet_callback_registration_t hci_event_callback_registration;

static void packet_handler(uint8_t packet_type, uint16_t channel, uint8_t *packet, uint16_t size);

static btstack_context_callback_registration_t handle_sdp_client_query_request;
[... 367 lines cut ...]
### HCI Packet Logs
[debug.zip](https://github.com/bluekitchen/btstack/files/6575703/debug.zip)

This log is **special**. It has the pklg file, but also a real over-the-air capture of LMP packets so you know precisely what is happening over-the-air. Because Wireshark cannot decode this file, I provide a compiled plugin
that you install to Wireshark **C:\Program Files\Wireshark\plugins\3.4\epan**. 
I can compile this plugin for Linux Wireshark if required.
* hci_dump.pklg - Default bluekitchen log file
* capture_bluetooth.pcapng - LMP & L2CAP wireshark capture (requires h4bcm.dll)
* logs.txt - Combined logs of LMP & L2CAP packets & program output (stdout)
* stack.1.txt - Bluekicthen program output (stdout)
* session.1.txt - Only LMP & L2CAP packets
* readme_wireshark_lmp_captures.txt - Guide to install h4bcm on wireshark for windows

### Environment:  (please complete the following information):
 - Current BTstack branch: develop
 - Bluetooth Controller: ESP32 VHCI (It's a custom code that does VHCI callback to UART path, nothing fancy)
 - Remote device: Bluetrum AB32VG1 (but the bug may happen with any target)

### Additional context
I'm trying to perform a full cycle as follows:
1. Query RFCOMM SDP services
2. Perform Pairing
3. Connect to target, start encryption and open all scanned 
4. Close all RFCOMM channels and delete all keys
5. Start again

mringwal, 2021-06-02:
Hi Matheus. Thanks for reporting - and providing the wireshark plug-ins. What would be the original tool to view these? I've got the Frontline Protocol Viewer installed.

In the screenshot, BTstack doesn't receive/log the L2CAP Configure Request from the remote - which is visible in the air trace. Any chance it got lost over the HCI Transport?

Matheus-Garbelini, 2021-06-02:
> In the screenshot, BTstack doesn't receive/log the L2CAP Configure Request from the remote - which is visible in the air trace. Any chance it got lost over the HCI Transport?

Hi @mringwal, thanks for pointing this out. It appears to be some HCI-H4 transmission error from esp32 to pc. I may need to fix this by adding some CRC check & retransmission on my esp32 vhci-uart code.

In any case, is there some L2CAP or RFCOMM timeout event that I can receive in the packet handler whenever such random uart errors occur even if `L2CAP Configure Request` is not received from remote? The only configuration I found is related to l2cap response timeout at [l2cap.c](https://github.com/bluekitchen/btstack/blob/master/src/l2cap.c#L113). I was checking if there is a way to retry `l2cap channel configuration with remote` without forcing disconnection.

> What would be the original tool to view these? I've got the Frontline Protocol Viewer installed.

A patched esp32 firmware that exposes Espressif Internal BT Baseband TX/RX path for sniffing. The received packets are then saved to .pcapng file and can be viewed using the plugin that I've attached. I don't know Frontline Protocol Viewer, but it appears to require expensive hardware.
Hopefully, this esp32 firmware will be released to the public after some embargo process.

Thanks again for your reply.

mringwal, 2021-06-04:
>  It appears to be some HCI-H4 transmission error from esp32 to pc. I may need to fix this by adding some CRC check & retransmission on my esp32 vhci-uart code.

> In any case, is there some L2CAP or RFCOMM timeout event that I can receive in the packet handler whenever such random uart errors occur even if L2CAP Configure Request is not received from remote? [...]

The Bluetooth protocol stack design basically assumes error-free packet transport end-to-end, with a minor exception of dropping ACL packets in the Controller under some conditions. I never actually tried that. I would argue that this is a good decision, as everything else (e.g. handling random errors on the HCI Transport) would let the complexity explode.

To make a long story short, please make sure that you don't flip a single bit between the Controller and the Host, btw. BTstack assumes that it can just pause incoming data via RTS indefinitely, although this should not be a problem when running on a desktop OS with large buffers.

Do you use the build-in HCI H4 UART of the ESP32? It sounds like you've implemented your own transport. An alternative could be to implement the H5 Protocol, which includes the full package (CRC, frame re-sync, windowing, re-transmission...).

(no, there's no retry functionality as packets either are delivered or the connection is disconnected in case of transmission problems).

> A custom esp32 firmware. The received packets are then saved to .pcapng file and can be viewed using the plugin that I've attached. I don't know Frontline Protocol Viewer, but it appears to require expensive hardware.
Hopefully, this esp32 firmware will be released to the public after some embargo process.

Wow! That would be awesome. I don't have a full featured BR/EDR sniffer either. I did implement an LE Sniffer (https://github.com/bluekitchen/raccoon), but LE is rather simple and there are others too. 

I'm closing this ticket, but feel free to contact me directly for news on your sniffer - matthias@bluekitchen-gmbh.ocm

closed 2021-06-04 by mringwal

Matheus-Garbelini, 2021-06-04:
Hi @mringwal thanks for your reply. 
Indeed since it assumes error-free, this issue should not require fixes from your side.
However, I was hoping there were some internal timeouts I could use to work around an H4 error scenario.

> Do you use the build-in HCI H4 UART of the ESP32? It sounds like you've implemented your own transport. An alternative could be to implement the H5 Protocol, which includes the full package (CRC, frame re-sync, windowing, re-transmission...).

It's my custom transport, but it's really just an ESP32 VHCI callback forwarding H4-HCI packets to UART.
I'll then add some CRC check + retry since I don't need anything fancy like H5. 
For now, I'll add a manual timer once an l2cap connection has been established as a timeout workaround for l2cap configuration issues.

> I'm closing this ticket, but feel free to contact me directly for news on your sniffer - matthias@bluekitchen-gmbh.ocm

Sure, I'll let you know. For some reason, BR/EDR vendors may use some obscure or expensive sniffers. 
Not sure why this is the case even in 2021.

mringwal, 2021-06-04:
> It's my custom transport, but it's really just an ESP32 VHCI callback forwarding H4-HCI packets to UART.
I'll then add some CRC check + retry since I don't need anything fancy like H5.
For now, I'll add a manual timer once an l2cap connection has been established as a timeout workaround for l2cap configuration issues.

Hm...how did you connect the ESP32 UART to your PC? If you use a regular dev kit with some USB-2-UART (e.g. FTDI232), the UART traces are a few centimeters and the chance for errors there are close to zero. Also, in the case of random communications errors, they would happen at others times too, e.g. during init. Could you check if your forwarding code can operate full-duplex? L2CAP Config is the first time whee there's full duplex communication usually. (I've seen similar issues with custom BTstack ports to MCUs). A quick check with a logic analyser might be helpful, too.,

> Sure, I'll let you know. For some reason, BR/EDR vendors may use some obscure or expensive sniffers.
> Not sure why this is the case even in 2021. I'm guessing it's due to some licensing issues in releasing LMP tools to the public.

Well, the really good ones are 80 Mhz SDR implementations, which are still not cheap. (I bought a LimeSDR Mini with 30 Mhz bandwidth, but if you don't have the full 80 Mhz, you would need to change the range, which requires FPGA code..). I don't think I've seen a RF chip that implements BR/EDR on the physical level - not like the easy to use nRF5x.

All link layers for regular Bluetooth Controllers are closed-source, and often licensed from a third-party, so there are licensing issues as you point out as well.

Matheus-Garbelini, 2021-09-06:
@mringwal Hi,
Sorry for the late reply. The **_active_** sniffer is now available in the following link:
https://github.com/Matheus-Garbelini/esp32_bluetooth_classic_sniffer

You can sniff Baseband packets directly from ESP32 when connecting (or accepting connection) to a remote BT device.
The project uses _Bluekitchen_ out of the box to guide the BT host protocol exchanges. Plus you can get packets in both Scapy or Wireshark live view.

Let me know what you think :+1:

mringwal, 2021-09-06:
@Matheus-Garbelini Thanks for the hint. I've seen the BrakTooth report and had a look at your repo, which didn't have a Readme. The new Readme looks great, I'll try to try it out soon.. and will provide feedback of course. Thanks!

Matheus-Garbelini, 2021-09-06:
@mringwal thanks. Yes I actually had to rush with the first commit, but now proper documentation with all acknowledgements are there. Didn't test with macOS yet, but my plan is to make sure the sniffer can be used on all platforms including Windows and macOS. Feel free to provide any feedback when also testing under macOS.

Regards.