bluekitchen/btstack #360: Dump BTSnoop format HCI traffic log

loginerrort, 2021-06-06:
**Is your feature request related to a problem? Please describe.**
There are two format that can be damped now that are BlueZ's format and packet logger format.  
But they record not only traffic data but also string log.  
Does it possible to add support for BTSnoop format?  It's purely HCI traffics in it.  
There are tools which can be used to analysis HCI traffic and parsing different layers of protocols, such like Frontline Protocol Analysis or Ellisys bluetooth analyzer.  
Android also can dump BTSnoop log, it's really convenient to debug protocol's bug.

**Describe the solution you'd like**
System which supported Posix file system can dump BTSnoop format log.

**Additional context**
[BT_Snoop_File_Format](https://fte.com/webhelpii/hsu/Content/Technical_Information/BT_Snoop_File_Format.htm)
[RFC 1761](https://datatracker.ietf.org/doc/html/rfc1761)

mringwal, 2021-06-07:
Interesting idea, and not much work either. We've been happy using Apple's PacketLogger and Wireshark for analysis.
Thanks for the BTSnoop format docs from Frontline. Do you know how to log debug messages as well?

loginerrort, 2021-06-11:
Glad to see you responding this feature request.  
BTSnoop format can only record HCI traffic.  
In my mind, I think that BTSnoop is an additional log type. It can be recorded beside origin log.  

The following is a snapshot importing BTSnoop log from Ellisys Bluetooth analyzer.
[image]

Regards

yy1059056584, 2021-06-23:
@mringwal 
Dear mringwal,
    I also have a strong desire to implement HCI logging for BTSnoop format. HCI interactive logging in this format is straightforward and clear for analysis. Here is an example of logging for the Frontline Protocol Comprobe Analyzer System, thanks.

[image]
[image]
[image]

mringwal, 2021-06-23:
Thanks for your feedback. How do I open a BTsnoop file in FrontLine's analyzer? (I found the "import" option in the Ellisys tool, though).

I've been fine with Apple's PacketLogger (the only one with a nice/clean user interface) and Wireshark (for cases where PacketLogger doesn't properly decode the packets, notably SDP queries and AVDTP or AVCDP (one of them doesn't work)).

In what situations would it be useful to use the BTsnoop file? 

Please check out the "hci-dump-btsnoop" branch. It will be merged onto develop after the v1.4.1 release (which will be soon-ish).

yy1059056584, 2021-06-23:
@mringwal 
Dear mringwal,
Q1: How do I open a BTsnoop file in FrontLine's analyzer?
A: BtSnoop file log is better using Frontline's own software to read instead of ellisays. When you install the software on your computer, you simply need to change the suffix of the Btsnoop file to ".cfa".
The following link can download the software.(Frontline Protocol Analysis Software, only support Windows system):
https://fte.com/support/CPAS-download.aspx?demo=BPA%20600&iid=21

Q2: In what situations would it be useful to use the BTsnoop file?
A: Btsnoop file can be used to analyze all the profiles in the bluetooth stack. Most importantly, the interaction between the Host layer and the Controller can be analyzed intuitively and clearly using the Frontline Protocol Analysis Software. It is a nearly perfect bluetooth protocol stack analysis tool.

The following example file is a Btsnoop file log from my Android phone that connects to a Bluetooth speaker,  you can experience it.

[btsnoop_hci_btspeaker.zip](https://github.com/bluekitchen/btstack/files/6701750/btsnoop_hci_btspeaker.zip)

mringwal, 2021-06-25:
Thanks for the help. We have released v1.4.1 and merged BTSnoop support onto develop branch.

closed 2021-06-25 by mringwal