bluekitchen/btstack #361: PBAP-Client chunked data (ESP32 PBAP vCard listing broken)

timkoers, 2021-06-14:
**Describe the bug**
I'm trying to fetch all contacts from a PBAP enabled server (smartphone), and I notice that very few contacts are actually displayed.
I've enabled obex dumps and when viewing them in HxD, I noticed that the incomming datastream is cut off. 
For example take the end of this ACL data dump:

(first packet:) `3C 63 61 72 64 20 68 61 6E 64 6C 65 3D 22 32 37 35 2E 76 63 66` 
(second packet:) `20 6E 61 6D 65 3D 22 56 6F 65 74 74 65 63 68 20 4C 69 73 73 65 22 2F 3E 3C 63 61 72 64 20 68 61 6E 64 6C 65 3D 22 32 39 39 2E 76 63 66 22 20 6E 61 6D 65 3D 22 47 65 6D 65 65 6E 74 65 68 75 69 73 20 48 61 61 72 6C 65 6D 6D 65 72 6D 65 65 72 22 2F 3E 3C 63 61 72 64 20 68 61 6E 64 6C 65 3D 22 33 36 35 2E 76 63 66`
If you paste these two in HxD, you'll see this

(first packet:) `<card handle="275.vcf"`
(second packet:) `name="Voettech Lisse"/><card handle="299.vcf" name="Gemeentehuis Haarlemmermeer"/><card handle="365.vcf`

When pasting the packets after each other, they become valid, but the card with handle 275.vcf doesn't get printed. It goes straight to the handle 299.vcf.

What I do see, is that the packet_handler starts to spit out vCards even during the processing stage, so what I think is happening is that the pbap_client_process_vcard_listing function in pbap_client.c expects a whole, valid vCard xml entry to be present at the start of each packet. If there isn't one, it will proceed to search a little bit further and eventually finds the next vCard xml entry in the packet., ignoring the split up vCard xml entry. Furthermore, after it prints out the vCard entry (handle 299) after the ignored entry (handle 275), it completely skips the rest of the packet all together (handle 365 and on).

This is due to the pbap_client_process_vcard_listing function being called with chunked data.

Dough! I just went scrolling through the code and I found a comment stating // TODO: handle chunked data.

I just felt that this issue should still be posted since other might be having this issue too.

Is there any progress on implementing chunked data?

mringwal, 2021-06-16:
Hi! Thanks for pointing at that TODO. What device/os/stack gave you the chunked data? 
Nobody's working on PBAP chunked data, although we made sure that PBAP Client passes the official tests. I just checked and didn't see a 'receive chunk data' test case. 
What's your use case for pulling the phone book?

timkoers, 2021-06-17:
Hi! I just pulled a whole phonebook from /telecom/pb containing around 340 contacts. I made a temporary (ugly) fix to get it to work. I just extended a buffer everytime new data came in until the PBAP_W4_GET_CARD_LIST_COMPLETE event got an OBEX_RESP_SUCCESS packet, then it will parse the cached buffer in one go, resulting in a valid contact list

mringwal, 2021-06-19:
Awesome! What device/os gives you the chunked data?

timkoers, 2021-06-20:
OnePlus 6T running the latest Oxygen OS. I've implemented a better method for the chunked data, but I'll test it first.

mringwal, 2021-06-25:
Thanks. The web tells me that it is based on Android. Do you know which Android version?

Also, would you have a HCI packet log until I'm able to reproduce it locally? I'm sure I've tried it against an at the time up-to-date Android and I didn't see the chunked data.

As yxml (the underlying xml parser) was designed for stream processing, it might be sufficient to only initialize at the beginning of the phonebook query.

mringwal, 2021-06-29:
Hi again. I've moved the parsing state into the pbap_client context and initiate it only in the initial pbap_pull_vcard_listing function. Could you give the code on the pbap-chunked-vcard-listing a quick try?

mringwal, 2021-06-30:
I did a quick test against PTS tool and vCard listing still works and have merged the changes onto develop. As the parsing state is now preserved across packets, this should be resolved now. Please reopen, if it doesn't work for you.

closed 2021-06-30 by mringwal

timkoers, 2021-07-17:
Thanks @mringwal!
Sorry for not responding for a while, the project got a little less attention as usual. I'll try to see if I can test it without breaking my current fixes

mringwal, 2021-07-18:
You could commit your local fixes to a branch and try with the version on develop...

What other fixes? :) - Feel free to open more issues if needed

timkoers, 2021-12-06:
@mringwal Just noticed my issue and the fact that I haven't responded yet. I'll try to answer you later today, if I don't please reach out to me

mringwal, 2021-12-06:
@timkoers good timing. we rewrote PBAP last week to handle any kind of fragmentation (there are stream parsers for xml phone book, application params, etc called from a stream parser that handles obex message structure). 

Please try the current develop branch.

timkoers, 2021-12-06:
I've created a pull request for my small changes, as of now I am unable to get the development branch to work. But then again I am unable to get my previous idf environment to work, so more debugging is needed for now. I'll keep you up-to-date

timkoers, 2021-12-06:
I've managed to get it to work, I've updated my commit, so that others will not have the same error as I did.

mringwal, 2021-12-07:
Thanks for the update. If you get 308 out of 341, the question is: does the pbap client code drop the missing contacts or are they not sent/requested? Maybe you can do some basic checks like count number of VCard Begin tags or similar. If you don't use any selector, do you get all 341?

timkoers, 2021-12-09:
I came to the conclusion that I was using a mix of your development branch and my personal edits to the code (not the new edits).

Regarding the new PBAP client, I would preffer to have the raw vCard data be marked with a  seperate event type (eg HCI_EVENT_PACKET -> HCI_EVENT_PBAP_META -> PBAP_SUBEVENT_VCARD_ENTRY_RESULT)

I'll have this edit in my pull request.
I haven't had the chance to count the resulted items yet.

timkoers, 2021-12-09:
@mringwal do you have discord?