bluekitchen/btstack #362: Abort during HSP AG Demo execution

jimmorrisson, 2021-06-17:
Assertion `next != ts' failed while trying to run hsp_ag_demo using raspberry pi port.

Steps to reproduce the behavior:
1. Run example hsp_ag_port while SCO_DEMO_MODE is defined as SCO_DEMO_MODE_SINE
2. Connect to device JBL CLIP 2
3. Start action: Establish audio connection

**Expected behavior**
Audio should be established.

**Logs**
[hsp_issue.zip](https://github.com/bluekitchen/btstack/files/6671440/hsp_issue.zip)

**Environment:  (please complete the following information):**
 - Branch: master v1.3.2 924daf9e4265a6459f04c3acfdb0b615fe2854e4
 - Bluetooth Controller Raspberry Pi 3 Model B+
 - Remote device: JBL CLIP 2

mringwal, 2021-06-18:
Hi. I've just tried the current develop version (shouldn't make a different) and did not get an assertion / received audio.
Could you select SCO_DEMO_MODE_SINE as SCO_DEMO_MODE in sco_demo_util.c and try to run the demo in gdb to get a backtrace? if the assert hits, just enter 'bt' and send the gdb output.

mringwal, 2021-06-18:
Alternatively, please also try to run the demo on a linux/mac system from the port/libusb or windows from port/windows-winusb for comparison.

jimmorrisson, 2021-06-21:
Hi, following your suggestions there's gdb bt (unfortunetely i have no bt dongle to test it on my mac):
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
50	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x76e82230 in __GI_abort () at abort.c:79
#2  0x76e8fbb8 in __assert_fail_base (fmt=0x76f966b0 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=0x445390 "next != ts", assertion@entry=0x76ff9cc0 "", file=0x445370 "../../src/btstack_run_loop.c", 
    file@entry=0x4454bc <__PRETTY_FUNCTION__.6750> "btstack_run_loop_base_add_timer", line=95, line@entry=1996056240, function=function@entry=0x4454bc <__PRETTY_FUNCTION__.6750> "btstack_run_loop_base_add_timer") at assert.c:92
#3  0x76e8fc6c in __GI___assert_fail (assertion=0x76ff9cc0 "", file=0x4454bc <__PRETTY_FUNCTION__.6750> "btstack_run_loop_base_add_timer", line=1996056240, 
    function=0x4454bc <__PRETTY_FUNCTION__.6750> "btstack_run_loop_base_add_timer") at assert.c:101
#4  0x0040b36c in btstack_run_loop_base_add_timer (ts=0x469df0) at ../../src/btstack_run_loop.c:95
#5  0x0040b8ce in btstack_run_loop_add_timer (ts=0x469df0) at ../../src/btstack_run_loop.c:239
#6  0x004127d8 in sco_schedule_tx (conn=0x469d98) at ../../src/hci.c:3033
#7  0x004128ec in sco_handler (packet=0x45faa3 <hci_packet_with_pre_buffer+15> "\006", size=63) at ../../src/hci.c:3080
#8  0x0041297a in packet_handler (packet_type=3 '\003', packet=0x45faa3 <hci_packet_with_pre_buffer+15> "\006", size=63) at ../../src/hci.c:3116
#9  0x00405052 in hci_transport_h4_packet_complete () at ../../src/hci_transport_h4.c:224
#10 0x00405234 in hci_transport_h4_block_read () at ../../src/hci_transport_h4.c:296
#11 0x00403bd8 in btstack_uart_block_posix_process_read (ds=0x45f7f8 <transport_data_source>) at ../../platform/posix/btstack_uart_posix.c:157
#12 0x004048ac in hci_uart_posix_process (ds=0x45f7f8 <transport_data_source>, callback_type=DATA_SOURCE_CALLBACK_READ) at ../../platform/posix/btstack_uart_posix.c:626
#13 0x00403246 in btstack_run_loop_posix_execute () at ../../platform/posix/btstack_run_loop_posix.c:209
#14 0x0040b9fc in btstack_run_loop_execute () at ../../src/btstack_run_loop.c:269
#15 0x00408352 in main (argc=1, argv=0x7efff6a4) at main.c:444

mringwal, 2021-06-21:
Thanks for the backtrace, now we know which timer was already active when it was added again. The one in sco_schedule_tx.
That's the good news and we'll fix that one. The bad news is that we haven't figured out how SCO packets (for HSP/HFP) are sent over HCI/UART connection of Broadcom/Cypress/Infineon controllers. There most likely will be period clicks.

There are no such issues when sending audio via A2DP, e.g. using A2DP Source Demo. Alternatively, this works fine with an USB Bluetooth dongle.

mringwal, 2021-07-08:
Hi again. I didn't run into this assert with another Controller. Anyway, I've added a one line fix to first remove the timer before adding it, which will avoid the assert that you've seen. The assert might fire when two SCO packets are received before the timer fires.

Did you try the USB dongle? Could you update to current develop branch and check if the assert is fixed? How good is outgoing voice? (sco_demo_util allows to let the demo send a sine wave which is perfect to check if sending(.

jimmorrisson, 2021-07-12:
Hi, i get no assert on the newest develop, but at the moment that i used to get an assert it stops sending (i can't hear sine anymore). Still haven't tried with the dongle (there were some weird things happening to my delivery and i couldn't get one :( )

mringwal, 2021-07-12:
Could you post the log again?
(If an USB dongle could be an option, that's the easiest way)

jimmorrisson, 2021-07-16:
Hi, here is the log
[hci_dump.zip](https://github.com/bluekitchen/btstack/files/6831171/hci_dump.zip)
I also managed to get dongle and it works on current master, on develop i had some issues building.

mringwal, 2021-07-16:
Thanks for the log. Looks like this doesn't work yet. Anyway, we haven't been able to send SCO data via HCI with Broadcom controllers from a POSIX system yet. If you can use an USB dongle, that's the simple way forward. Please post a new issue if you have issues compiling develop branch.

closed 2021-07-16 by mringwal