bluekitchen/btstack #363: bond data isn't parsed correctly (when having more than one device) causing repair requests.

dit8, 2021-06-20:
used tag v1.4 in reproduce

**To Reproduce**
you can use sm_pairing_peripheral test to reproduce. just use following:
sm_set_io_capabilities(IO_CAPABILITY_NO_INPUT_NO_OUTPUT);
sm_set_authentication_requirements(SM_AUTHREQ_SECURE_CONNECTION | **SM_AUTHREQ_BONDING**); 

Steps to reproduce the behavior:
1. connect (bond) to ble device1 and then disconnect
2. connect (bond) to ble device2 and then disconnect
3. reconnect to both devices works fine at this point
4. exit test and re-run the test (which caused btstack_at_{my_mac}_le_device_db.txt to be read)
5. attempt to reconnect to device2 fails with LL_REJECT_IND "pin or key missing" (device1 reconnects fine).

**Expected behavior**
5. we should be able to reconnect to both device1 and device2, but reconnect works only to device1.

**HCI Packet Logs**
[repro_info_bond.zip](https://github.com/bluekitchen/btstack/files/6682140/repro_info_bond.zip)
attached

**Environment:  (please complete the following information):**
 - Current BTstack branch: [e.g. develop]
 - Bluetooth Controller exynos
 - Remote device: iPhone SX with iOS 14.4, iphone6S IOS 14.4 (reproduces regardless of device model)

**Additional context**
issue is in file le_device_db_fs.c, function le_device_db_read, line 248
```
            // read newline
            fgetc(wFile);
```
this actually reads "," but not the "\n" of the end of the line thus parsing all the lines but the first line incorrectly.
 (see hci_dump line 162-171)

mringwal, 2021-06-21:
Thanks for the report and the investigation. In general, le_device_db_fs.c as well as btstack_link_key_db_fs.c are deprecated. Instead you could use le_device_db_tlv and link_key_db_tlv together with bstack_tlv_posix.c.

Nevertheless, we'll fix le_device_db_fs.c anyway.

dit8, 2021-06-21:
I appreciate your reply. how should I save the credentials? can you please specify an example or reference in documentation.
also, where can I see that this file is depricated? 
thanks

mringwal, 2021-06-21:
hi. to save the bonding information on a posix system, please configure btstack_tlv_posix, and then use the le_device_db_tlv on top of it, as shown here: https://github.com/bluekitchen/btstack/blob/master/port/libusb/main.c#L126

Sorry, I guess the use of the generic TLV storage over the specific implementations for le device db as well as link key db isn't really documented. Usually, people start with one of the provided ports and move on from there. In them, le_device_db_fs.c. Anyway, no big deal and we'll fix it soon anyway.

mringwal, 2021-06-21:
Here's the fixed version: https://github.com/bluekitchen/btstack/commit/21a309a4d79a7150c9635bf2e88a0b47447fd8db

closed 2021-06-21 by mringwal