bluekitchen/btstack #366: WinUsb not starting device

mmartinm1992, 2021-07-08:
**Describe the bug**
This might not be a btstack bug but it is related with the winUsb usage. I am using btstack in Windows to make a Desktop Application that comunnicates via BLE but since May I can See that the WinUsb driver is unable to initialize my Intel Bluetooth device, therefore I have to reinstall the provider driver and reinstall the WinUSB. 

This can be seen in the following image. And due to that the BLE communication is not working fine.
[image]

Any clue on how to fix this?

**To Reproduce**
Steps to reproduce the behavior:
1. Run any example on Windows after installing the WinUSB driver.
2. Restart your PC
3. Run the same example, the BTstack will return local addr 00:00:00:00:00:00 and won't initialize correctly

**Expected behavior**
Do not need to reinstall WinUsb everytime the PC is restarted.

**HCI Packet Logs**
None

**Environment:  (please complete the following information):**
 - Current BTstack branch: Does not matter it is WinUSB thing.
 - Bluetooth Controller : Intel Wireless bluetooth.
 
[image]

 - Remote device: [e.g. iPhone X with iOS 13.3]

**Additional context**

mringwal, 2021-07-08:
HI there. Is this with an internal Intel Wireless 8260/8265 Controllers or similar?

These controllers require a firmware to load, which is done by your OS when the driver is installed. The firmware stays active until you power down your PC. If you install the WinUSB driver, the OS will not load the firmware.

Please try port/windows-winusb-intel, which supports loading the firmware.
If this doesn't work, please also upload the file hci_dump.pklg from the current directory.

mmartinm1992, 2021-07-08:
Thank you for your quick answer, thats it, the Bluetooth device of the laptop is the 8087. I will try with windows-winusb-intel Makefile. Also the final application will be running on windows 7 PCs, does the dongle work fine with Windows 7 for BLE communication.

Can you advice any commercial usb dongle?

mmartinm1992, 2021-07-08:
I have tried it compiled in windows-winusb-intel. At first it was successfull (I didn't restart the PC), but after restarting it it was not working: 
[image]
[hci_dump.zip](https://github.com/bluekitchen/btstack/files/6785964/hci_dump.zip)

mringwal, 2021-07-08:
Do you have the file ibt-19-2.sfi in your local directory? It looks like it's not there but it is the firmware file that's needed for this controller.

hm... looks like that's missing in the Windows makefile. Could you try to run 'make all-intel' to see if the files get downloaded, and then try again?

mringwal, 2021-07-08:
Ah... this file is not in the list. Please find it in the linux source tree. The other ones are fromL  https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/

mringwal, 2021-07-08:
Well. No ibt-19-2.sfi there.The code in chipset/intel/btstack_chipset_intel_firmware.c uses the hw_variant field from the hci_intel_read_version (19, seems to be ok) and the device revision from the hci_intel_read_secure_boot_params, which seems to be 0x0002, 
Ah...on the linux kernel repo, the devices with hw_variant 19 have a three number scheme, which means the name is composed diferently. You can either try to find the code in BlueZ, or, ... try all provides files that start with 19...

mmartinm1992, 2021-07-12:
I have successfully load the first one on the linux kernel web page (ibt-19-0-0.sfi ). Thank you so much for your help!!!

closed 2021-07-12 by mmartinm1992