bluekitchen/btstack #367: Can't connect to Creative Muvo 1c using A2DP

jimmorrisson, 2021-07-16:
Hi, I can't connect to the Creative Muvo 1c while trying to run a2dp_source_demo using libusb port.

Steps to reproduce the behavior:
1. Run example a2dp_source_demo
2. Connect to remove device Creative Muvo 1c
3. Start action A2DP Source create connection to addr or AVRCP create connection to addr

**Expected behavior**
Audio should be established.

**HCI Packet Logs**
[hci_dump.zip](https://github.com/bluekitchen/btstack/files/6831006/hci_dump.zip)

**Environment:  (please complete the following information):**
 - Current BTstack branch: master
 - Bluetooth Controller Asus BT400
 - Remote device: Creative Muvo 1c

mringwal, 2021-07-16:
Hi. Your speaker connects to the A2DP Source demo before it is started. Please turn your speaker off, start the demo, then only turn the speaker on. Does this help?

jimmorrisson, 2021-07-16:
Hi. No it doesn't help i've tried couple of times, but can't get correct A2DP connection. Checked HFP and it works great on that device ;)

mringwal, 2021-07-16:
Could you send a new log with that?

jimmorrisson, 2021-07-16:
Sure 
[hci_dump.zip](https://github.com/bluekitchen/btstack/files/6831509/hci_dump.zip)

mringwal, 2021-07-16:
In the log, BTstack sends an SDP query for "AVDTP" and the remote answer "nope, don't have that".
Are you sure it supports A2DP Sink?

jimmorrisson, 2021-07-16:
Spec says that it has support for A2DP https://www.flipkart.com/creative-muvo-1c-bluetooth-speaker/p/itm86f4844f05414 . I've just tried connecting with android and it connected using SBC :)

mringwal, 2021-07-18:
Weird. If it supports A2DP Sink, but don't report an SDP Record for "AVDTP", I would consider this to be a bug in their Bluetooth SDP Server implementation.

Out of curiosity, could you replace BLUETOOTH_PROTOCOL_AVDTP with  BLUETOOTH_SERVICE_CLASS_AUDIO_SINK in src/classic/avrcp.c:289

-        sdp_client_query_uuid16(&avdtp_handle_sdp_client_query_result, (uint8_t *) connection->remote_addr, BLUETOOTH_PROTOCOL_AVDTP);

+        sdp_client_query_uuid16(&avdtp_handle_sdp_client_query_result, (uint8_t *) connection->remote_addr, BLUETOOTH_SERVICE_CLASS_AUDIO_SINK);

and try + send logs?

mringwal, 2022-01-28:
HI @jimmorrisson did you get this to work? I've seen this issue with other speakers, esp. ones with an Actions (https://www.actions-semi.com) Bluetooth Controller chip. There, using the BLUETOOTH_SERVICE_CLASS_AUDIO_SINK fixes it.

mringwal, 2023-02-21:
Closing due to inactivity.

closed 2023-02-21 by mringwal