bluekitchen/btstack #370: Change bonding / connection restrictions.

satirebird, 2021-07-19:
I want to implement a LE peripheral device which has no capabilities for the user to confirm something during paring / bonding. Due to the limitation I want to limit the bonding time after reset. 

- The user should be able to bond the device within the first few minutes after reset
- It should not possible to connect to the device without bonding after the "initialization" period.
- (It should not possible to use the device without bonding)

How can I implement this behavior using the btstack API? I have read the documentation but I couldn't find any useful information about that - especially how to limit the access for none bonded central devices.

satirebird, 2021-07-20:
After some research I figured out that the event SM_EVENT_IDENTITY_RESOLVING_FAILED is a good point disconnect the central if the device is not in initialization mode.

closed 2021-07-20 by satirebird

mringwal, 2021-07-21:
Hi. This would be a perfect question for the mailing list.

You cannot reject an incoming connection as Peripheral. The only thing you can do is to disconnect if you don't like the other. If the GATT Services requires encryption, this should not be a problem

SM_EVENT_IDENTITY_RESOLVING_FAILED is not directly related to "are paired - although it might relect the bonding state well. Other things to check is, e.g. try to request security and if that doesn't happen, you can safely disconnect.

Pairing mode/non-pairing mode: you'll get an event for pairing request that you can deny, if not in pairing mode by your application specification.