bluekitchen/btstack #371: How can I assure that classic HID Bluetooth keyboard works?

AlexanderSveshnikov, 2021-07-26:
Hello.
 I'm trying to create my device on a base of "HID Keyboard example" using ESP32 SoC (Bluekitchen Classic bluetooth HID example).  I'm using "Manhattan bluetooth micro adapter Hi speed USB2.0 Class 2 + EDR (Dongle) connected to Windows7 (or Windows10) computer.

I expect that when I connect (pair) my device with "HID Keyboard example" (with HAVE_BTSTACK_STDIN disabled) downloaded into it and open text editor (notepad) I'll see the string  demo_text = "\n\nHello World!\n\nThis is the BTstack HID Keyboard Demo running on an Embedded Device.\n\n" output discontinuously. But nothing happens.  I tried Classic bluetooth HID mouse example, and when paired it with computer I saw my mouse pointer movement across the screen. So a desided "it works!" How can I assure that keyboard example works?
I found  that I need to set to set IO capability to  SSP_IO_CAPABILITY_DISPLAY_YES_NO, so I added 
gap_ssp_set_io_capability(SSP_IO_CAPABILITY_DISPLAY_YES_NO); to the btstack_main function, but this doesnt help.
Then I tried to output packet_type in packet_handler function, and I found that after HCI_EVENT_USER_CONFIRMATION_REQUEST there's no HCI_EVENT_HID_META with either HID_SUBEVENT_CONNECTION_OPENED or HID_SUBEVENT_CAN_SEND_NOW. But I found a lot of other HCI events,  packets with ID's 0x36, 0F, 0E, 6E, 05, 61, 04, 03, 0B... What am I doing wrong? How to make keyboard example work in my case?
Regards
Alexander Sveshnikov

AlexanderSveshnikov, 2021-07-26:
Here is the log with HCI commands that I got during keyboard "session"
Got hci type 6E
Got hci type 0E
Got hci type 6E
BTstack up and running at 84:0D:8E:20:1E:9A
Got hci type 60
Got hci type 0E
Got hci type 6E
Got hci type 04
Got hci type 0F
Got hci type 6E
Got hci type 61
Got hci type 03
Got hci type 0F
Got hci type 0B
Got hci type 6E
Got hci type 6E
Got hci type 1B
Got hci type 6E
Got hci type 6E
Got hci type 6E
Got hci type 0F
Got hci type 23
Got hci type 13
Got hci type 6E
Got hci type 6E
Got hci type 13
Got hci type 6E
Got hci type DF
Got hci type 32
Got hci type 31
Got hci type 0E
Got hci type 13
Got hci type 6E
Got hci type 33
SSP User Confirmation Request with numeric value '580635'
SSP User Confirmation Auto accept
Got hci type 0E
Got hci type 6E
Got hci type E0
Got hci type 36
Got hci type 18
Got hci type 08
Got hci type 0E
Got hci type 6E
Got hci type 0F
Got hci type 17
Got hci type 6E
Got hci type 0E
Got hci type 6E
Got hci type 6E
Got hci type 6E
Got hci type D8
Got hci type 06
Got hci type 6E
Got hci type 6E
Got hci type 13
Got hci type 30
Got hci type 6E
Got hci type 6E
Got hci type 6E
Got hci type 13
Got hci type 6E
Got hci type 13
Got hci type 13
Got hci type 05
Got hci type 61
Got hci type 04
Got hci type 0F
Got hci type 6E
Got hci type 61
Got hci type 03
Got hci type 0F
Got hci type 0B
Got hci type 6E
Got hci type 6E
Got hci type 1B
Got hci type 0F
Got hci type 23
Got hci type 17
Got hci type 0E
Got hci type 6E
Got hci type FF
Got hci type 08
Got hci type 0E
Got hci type 6E
Got hci type 0F
Got hci type 17
Got hci type 6E
Got hci type 0E
Got hci type 6E
Got hci type 6E
Got hci type 6E
Got hci type D8
Got hci type 06
Got hci type 6E
Got hci type 13
Got hci type 30
Got hci type 13
Got hci type 05
Got hci type 61
Got hci type 04
Got hci type 0F
Got hci type 6E
Got hci type 61
Got hci type 03
Got hci type 0F
Got hci type 0B
Got hci type 6E
Got hci type 6E
Got hci type 1B
Got hci type 0F
Got hci type 23
Got hci type 17
Got hci type 0E
Got hci type 6E
Got hci type FF
Got hci type 08
Got hci type 0E
Got hci type 6E
Got hci type 0F
Got hci type 17
Got hci type 6E
Got hci type 0E
Got hci type 6E
Got hci type 6E
Got hci type 6E
Got hci type D8
Got hci type 06
Got hci type 6E
Got hci type 13
Got hci type 30
Got hci type 13
Got hci type 05
Got hci type 61

mringwal, 2021-07-26:
Please enable packet logger in app_main() of main/main.c

Also, please test with a different computer/phone/tablet. Windows might cache the HID description too long and get confused if a device changes its Bluetooth profile.

AlexanderSveshnikov, 2021-07-28:
After a number of experiments I found that after ~30sec of disconnection (hci type 05) device reconnected and starting it's timer that handles embedded output. But I found this behaviour is only if Windows 7 with dongle  used as host. Also I found that output to notepad delayed significantly and letters repeated. So I have two questions:
- I want my device to be connected immediately after disconnection ( "HCI_EVENT_DISCONNECTION_COMPLETE") As I found, I need to use hid_device_connect(device_addr, &hid_cid); function. But how can I get bluetooth (MAC) address using Bluekitchen stack? (esp_bt_dev_get_address() crashes the device, because have to be used with bluedroid stack, as I found)
- What is the cause of this delays and letter repeat?
I'd be thankful if you could answer me.

AlexanderSveshnikov, 2021-07-28:
Second question answered. Code error

mringwal, 2021-07-29:
gap_local_bd_addr() gives you the Bluetooth address of your device. To connect, you need the address of the remote device, which you get, e.g. in the HID_SUBEVENT_CONNECTION_OPENED event.

Slow typing: this is a very weird behaviour of Windows when the keycode isn't send in the first position of the HID report. If you update to the current develop version, it will behave correctly (as we've recently figured out the problem).

Please let me know, if this works and you've got everything working now.

AlexanderSveshnikov, 2021-07-29:
As I understood, the "slow typing" problem  arised because of wrong report length definition in function send_report(...). You defined this length = 8 bytes in descriptor (...
// Keycodes

    0x95, 0x06,                    //   Report Count (6)
    0x75, 0x08,                    //   Report Size (8)
...) and made this length 9 bytes in report. When I removed 1 byte, Windows typing speed increased significantly. Moreover, with this correction I can  connect to old Tablet with android 5.1 & can see typing on it's screen (I was not able to made connection with this Tablet before this correction)

mringwal, 2021-07-29:
Hm. EIther that, or the wrong position. Did you try the version on the develop branch? Does it work for you as well?

AlexanderSveshnikov, 2021-07-29:
Ok, I'll try to change keycode position. Please wait awhile

mringwal, 2021-07-29:
Not sure what that means. please just try the version on develop :)

AlexanderSveshnikov, 2021-07-29:
That means: I'll change string uint8_t report[] = {0xa1, modifier, 0, keycode, 0, 0, 0, 0, 0}; to 
string uint8_t report[] = {0xa1, modifier, 0, 0, 0, keycode, 0, 0, 0}; (for examp[le)

mringwal, 2021-07-29:
I see. Current code on develop is this

    uint8_t report[] = { 0xa1, REPORT_ID, modifier, 0, keycode, 0, 0, 0, 0, 0};

Report Data, Report ID, modifiers, reserved, 6 x keycodes

AlexanderSveshnikov, 2021-07-29:
Output speed remains fast when  keycode position  changes. Problem was in report length. I was right.

mringwal, 2021-07-29:
Excellent. So, I've fixed the actual problem by accident. good enough :)
Can we close this issue now?

AlexanderSveshnikov, 2021-07-29:
Ok

closed 2021-07-29 by AlexanderSveshnikov

mringwal, 2021-07-29:
Great!