bluekitchen/btstack #374: Failed to connect to a2dp_source_demo

alisagao123, 2021-08-06:
**Describe the bug**
AuthenticationTimeout to connect to a2dp_source_demon sometimes, as below:
Attempting to pair with 00:1A:7D:DA:71:11
Failed to pair: org.bluez.Error.AuthenticationTimeout

**To Reproduce**
Steps to reproduce the behavior:
1. Run example 'a2dp_source_demo'
2. Connect from remote device(linux system) with bluetoothctl app
***Note*** there are two "a2dp_source_demo" running in one LAB room, but they run on different PCs(Linux system)
one is : 00:1A:7D:DA:71:1; another is "00:1A:7D:DA:71:13". not sure if this is the reason.
********************************

**Expected behavior**
it should connect to a2dp source demo even several times tries.

**HCI Packet Logs**
spawn bluetoothctl
Waiting to connect to bluetoothd...
Agent registered
Controller 20:4E:F6:BB:0F:BA Pairable: yes
# paired-devices
[bluetooth]
Device D0:A5:AE:CE:52:8B nut
Device 10:D2:5A:69:2D:4F nut
Device 3B:7A:5A:69:2B:40 nut
Device 1E:CF:5A:69:1B:82 nut
Device 20:43:5A:69:29:2B nut
Device 37:86:5A:69:2D:4E nut
Device 0E:F9:36:81:7D:93 nut
Device 66:BA:5A:69:23:79 nut
Device 78:80:5A:69:27:0B nut
Device 5A:D0:5A:69:2D:4A nut
Device 28:94:5A:69:11:60 nut
Device 61:9E:EC:F5:A9:46 61-9E-EC-F5-A9-46
# scan on
Discovery started
Controller 20:4E:F6:BB:0F:BA Discovering: yes
Device 0E:F9:36:81:7D:93 nut
Device 61:9E:EC:F5:A9:46 61-9E-EC-F5-A9-46
Device 78:80:5A:69:27:0B nut
Device 28:94:5A:69:11:60 nut
Device 61:9E:EC:F5:A9:46 ManufacturerData Key: 0x004c
Device 00:1A:7D:DA:71:13 A2DP Source 00:1A:7D:DA:71:13
Device 00:1A:7D:DA:71:11 A2DP Source 00:1A:7D:DA:71:11
# scan off
Device 00:1A:7D:DA:71:11 RSSI is nil
Device 00:1A:7D:DA:71:13 RSSI is nil
Device 3B:7A:5A:69:2B:40 RSSI is nil
Device 78:80:5A:69:27:0B RSSI is nil
Device 61:9E:EC:F5:A9:46 RSSI is nil
Device 0E:F9:36:81:7D:93 RSSI is nil
Controller 20:4E:F6:BB:0F:BA Discovering: no
Discovery stopped
# pair 00:1A:7D:DA:71:11
Attempting to pair with 00:1A:7D:DA:71:11
[bluetooth][0m# 
Failed to pair: org.bluez.Error.AuthenticationTimeout
Device 00:1A:7D:DA:71:11 A2DP Source 00:1A:7D:DA:71:11
Device 20:43:5A:69:29:2B nut
Device 28:94:5A:69:11:60 nut
 Device 61:9E:EC:F5:A9:46 61-9E-EC-F5-A9-46
# trust 00:1A:7D:DA:71:11
Device 00:1A:7D:DA:71:11 not available

*********TRY again*****************
+ bt_connect.sh 00:1A:7D:DA:71:11
spawn bluetoothctl
Waiting to connect to bluetoothd...
# Controller 20:4E:F6:BB:0F:BA Pairable: no
# scan on
# Device 00:1A:7D:DA:71:13 A2DP Source 00:1A:7D:DA:71:13
# Device 00:1A:7D:DA:71:13 A2DP Source 00:1A:7D:DA:71:13
# Device 00:1A:7D:DA:71:11 A2DP Source 00:1A:7D:DA:71:11
# scan off
# Device 00:1A:7D:DA:71:11 RSSI is nil
# Device 00:1A:7D:DA:71:13 RSSI is nil
# Device 10:D2:5A:69:2D:4F RSSI is nil
# Controller 20:4E:F6:BB:0F:BA Discovering: no
# Discovery stopped
# pair 00:1A:7D:DA:71:11
Attempting to pair with 00:1A:7D:DA:71:11
#Failed to pair: org.bluez.Error.AuthenticationTimeout
# Device 00:1A:7D:DA:71:11 A2DP Source 00:1A:7D:DA:71:11
# trust 00:1A:7D:DA:71:11
Device 00:1A:7D:DA:71:11 not available
....................................

**Environment:  (please complete the following information):**
1. Current BTstack branch: [develop]
2. Bluetooth Controller [biaze D27]
3. Remote device: embedded device running Linux 5.10

mringwal, 2021-08-08:
Hi. Please enable hci packet logging and convert the text output into a .pklg file which can be viewed with Wireshark.
See: https://bluekitchen-gmbh.com/btstack/#how_to/#bluetooth-hci-packet-logs

Do you have a datasheet for the [biaze D27] Bluetooth Controller?

alisagao123, 2021-09-09:
@mringwal , thanks for your reply, I will get the packet as your request.
and I found another issue:
when I plug-out/plug-in the BT dongle, and use command "hciconfig -a", I can see the interface up and running, 
but after I run the "a2dp_source_demo", and then this interface disappear in command "hciconfig -a", do you known why or is it expected?

(base) elbert@shrek-ThinkCentre-M83:~$ hciconfig
hci0:	Type: Primary  Bus: USB
	BD Address: 00:1A:7D:DA:71:11  ACL MTU: 310:10  SCO MTU: 64:8
	UP RUNNING 
	RX bytes:646 acl:0 sco:0 events:42 errors:0
	TX bytes:2175 acl:0 sco:0 commands:42 errors:0

(base) elbert@shrek-ThinkCentre-M83:~$ sudo a2dp_source_demo 
[sudo] password for elbert: 
Packet Log: /tmp/hci_dump.pklg
USB Path: 01-03
Local version information:
- HCI Version    0x0006
- HCI Revision   0x22bb
- LMP Version    0x0006
- LMP Subversion 0x22bb
- Manufacturer 0x000a
BTstack up and running on 00:1A:7D:DA:71:11.

[2]+  Stopped                 sudo a2dp_source_demo
(base) elbert@shrek-ThinkCentre-M83:~$ hciconfig
(base) elbert@shrek-ThinkCentre-M83:~$

mringwal, 2021-09-09:
When you use BTstack with the USB Bluetooth dongle, you should not use it with BlueZ at the same time. I don't know BlueZ very well, in the libusb/Readme.md, I've added notes on how to prevent BlueZ / Kernel from using the USB Dongle. 

Ah... so after running BTstack it is not bad that it does not appear in hciconfig. (you shouldn't even use it with BlueZ at all)

mringwal, 2022-01-17:
No response, closing this.

closed 2022-01-17 by mringwal