bluekitchen/btstack #382: How could get hfp sco packets over HCI/UART connection of Broadcom/Cypress/Infineon controllers

yajunzhi, 2021-10-09:
Hi, 
I found the issue that described the audio sco packet issue. Ref: "Abort during HSP AG Demo execution #362" 
I want to know whether the project could receive/send sco packet from cpyress CYW20704/20706 controllers or not.
In the project, there are vendor configuration commands, but it looks running not success, is that true?

mringwal, 2021-10-09:
For Broadcom-based Controllers, we don't know how to send SCO packets over HCI correctly. However, newer controllers (incl. CYW20704) provide an internal Wide-Band Speech (mSBC) codec and the SCO audio can be routed over the PCM/I2S interface, see ENABLE_SCO_OVER_PCM + ENABLE_BCM_PCM_WBS.

mringwal, 2021-10-09:
I've used one of the Laird Technologies BT-850 or BT-860 Dev Kits. I somehow missed the difference between 850 and 860 - both contain an CYW20704.

What do your mean by "image version of the controller"? I didn't use/have a PatchRAM file for that one. If there are, please let me know where they are available.

With  ENABLE_SCO_OVER_PCM + ENABLE_BCM_PCM_WBS, the stack configures it for 8 or 16 kHz, 16 bit, I2S Master. I think I just connected a generic I2S to speaker board from Adafruit.

yajunzhi, 2021-10-11:
Thanks for your info.
I have a cypress CYBT-343026-EVAL board, and flash images throgth WICD-studio. The PatchRAM files are provided by the studio project. The controller is connected with a CSR8675 audio board througth I2S interface.  But not sure the PCM/SCO settings correct or not.

yajunzhi, 2021-11-12:
Hi, sorry for long time to confirm.
At last, the controller could not output any clk no matter how to try, I guess the pins settings is not correct. 
We changed the controller board, choose TI controller. And it the sco is ok.
Thanks for your support.

mringwal, 2021-12-21:
Glad you got it working.

closed 2021-12-21 by mringwal