bluekitchen/btstack #386: doc/manual build error

Roger-Hoo3, 2021-11-05:
**Describe the bug**
error occur in Makefile:69 when make in doc/manual

**To Reproduce**
make in doc/manual

**Expected behavior**
no error

**HCI Packet Logs**
NO

**Environment:  (please complete the following information):**
NO

**Additional context**
Tables is chaotic in btstack_gettingstarted.pdf, I think it is related with building error
[image]

milamikica, 2021-11-05:
Hi,

thank you for reporting this, I’ll take a look at it - for now please use the HTML version. 

Best regards,
Mila

Roger-Hoo3, 2022-02-16:
rootcause: columnwidth is mistakenly set for longtable during converting from btstack_generated.md to btstack_final.tex by pandoc, as follows：
[image]

if set columnwidth as ' {\raggedright\arraybackslash}p{0.4231\columnwidth }',  table isnot chaotic.

Now i cannot provide solution to fix this bug, perhaps it is ok which add --filter option to modify columnwidth setting during using pandoc...

Roger-Hoo3, 2022-02-25:
this issue is related to build environment, so close it.

closed 2022-02-25 by Roger-Hoo3