bluekitchen/btstack #388: Can't see my device changed name in advertisement report

tamarva, 2021-11-30:
Hi,
When I scan in the first time I can see my device. Implementing FOTA - when I write to a certain characteristic - the device that does the advertise changes it's name and address. When I look at application LightBlue **I can see the new name** of the device and it's **new address** but the in the advertisement_report_contains_name method I can't see the new name of the device but only the old name (Although it clearly changed as I can see in lightblue)

Your help is much appreciated

tamarva, 2021-11-30:
I fixed this by changing in the scan parameters from passive scanning to active scanning

tamarva, 2021-11-30:
I'm not sure why but it got fixed by changing in the scan parameters from passive scanning to active scanning

closed 2021-11-30 by tamarva

mringwal, 2021-11-30:
In BLE, the device name can be provide in the advertisement and as a GATT Characteristic of the GAP Service. Smartphone usually will first use the name from the adv if available and on connect, read the GAP Name Characteristic. Maybe your smartphone did delete the cached entry