bluekitchen/btstack #391: make l2cap_get_channel_for_local_cid public

versaloon, 2021-12-07:
**Is your feature request related to a problem? Please describe.**

I'm implementing a device/driver framework extension for btstack, which will make it easier to implement drivers.
To implement this, I need to get the hci_con_handle_t in btstack_packet_handler.
Most HCI packets has that parameter, but L2CAP_DATA_PACKET don't.
There is a private API l2cap_get_channel_for_local_cid can be used to get the l2cap_channel structure, and con_handle is in this structure.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

To implement a device/driver framework extension, I need to call different driver packet_handlers with device struct.

static void btstack_packet_handler(uint8_t packet_type, uint16_t channel, uint8_t *packet, uint16_t size)
{
    hci_con_handle_t con_handle = HCI_CON_HANDLE_INVALID;

    switch (packet_type) {
    case HCI_EVENT_PACKET:
        switch (hci_event_packet_get_type(packet)) {
        case BTSTACK_EVENT_STATE:
            ......
            break;
        case HCI_EVENT_CONNECTION_COMPLETE:
            if (!hci_event_connection_complete_get_status(packet)) {
                con_handle = hci_event_connection_complete_get_connection_handle(packet);
                ......
            }
            break;
        case L2CAP_EVENT_INCOMING_CONNECTION:
            con_handle = l2cap_event_incoming_connection_get_handle(packet);
            break;
        case L2CAP_EVENT_CHANNEL_OPENED:
            con_handle = l2cap_event_channel_opened_get_handle(packet);
            break;
        case HCI_EVENT_DISCONNECTION_COMPLETE:
            con_handle = hci_event_disconnection_complete_get_connection_handle(packet);
            break;
        }
        break;
    case L2CAP_DATA_PACKET: {
            extern l2cap_channel_t * l2cap_get_channel_for_local_cid(uint16_t local_cid);
            l2cap_channel_t * l2cap_channel = l2cap_get_channel_for_local_cid(channel);
            if (l2cap_channel != NULL) {
                con_handle = l2cap_channel->con_handle;
            }
        }
        break;
    }

    if (con_handle != HCI_CON_HANDLE_INVALID) {
        btstack_dev_t *dev = btstack_get_dev_by_con_handle(con_handle);
        if (dev != NULL) {
            if (dev->op->packet_handler != NULL) {
                dev->op->packet_handler(dev, packet_type, channel, packet, size);
            }
            ......
        }
    }
}

Code above will try to get the con_handle and find the corresponding device structure, and than call packet_handler in the driver op.
The only modification to btstack is to make l2cap_get_channel_for_local_cid public, which is a static API.

Or, is there any other public API can do this?

mringwal, 2021-12-09:
Hi @versaloon. Could you explain a bit more on your architecture? why do you need the con handle for l2cap channels?

The basic idea in BTstack's design is that there's a single identifier that is used between layers, so l2cap uses the hci con handles for hci, rfomm uses the l2cap cids, etc.. So, the hci con handle is not needed for the l2cap api.

In BTstack's daemon (client/server), the server needs to keep track which connection was created by which client, so it needs to keep a separate list. With your device driver, it looks like you want to have one struct per remote device, but do you need that aggregated information versus e.g. all management data per l2cap/rfcomm/bnep ... channel?

versaloon, 2021-12-16:
For example, I use btstack to connect an earphone, a mouse, a game controller.
The driver I plemented 
```c
void btstack_host_hid_base_packet_handler(btstack_dev_t *dev, uint8_t packet_type, uint16_t channel, uint8_t *packet, uint16_t size)
{
    btstack_host_hid_base_t *hid_dev = (btstack_host_hid_base_t *)btstack_host_get_priv_dev(dev);

    if (HCI_EVENT_PACKET == packet_type) {
        switch (hci_event_packet_get_type(packet)) {
        case L2CAP_EVENT_INCOMING_CONNECTION: {
                uint_fast16_t cid = l2cap_event_incoming_connection_get_local_cid(packet);
                switch (l2cap_event_incoming_connection_get_psm(packet)) {
                case PSM_HID_CONTROL:
                    BTSTACK_HOST_HID_TRACE("%d: incoming control channel\r\n", dev->con_handle);
                    hid_dev->l2cap_control_cid = cid;
                    break;
                case PSM_HID_INTERRUPT:
                    BTSTACK_HOST_HID_TRACE("%d: incoming interrupt channel\r\n", dev->con_handle);
                    hid_dev->l2cap_interrupt_cid = cid;
                    break;
                default:
                    BTSTACK_ASSERT(false);
                    return;
                }
                l2cap_accept_connection(cid);
            }
            break;
        case L2CAP_EVENT_CHANNEL_OPENED:
            if (l2cap_event_channel_opened_get_status(packet) != 0) {
                break;
            }

            uint_fast16_t psm = l2cap_event_channel_opened_get_psm(packet);
            if (PSM_HID_CONTROL == psm) {
                BTSTACK_HOST_HID_TRACE("%d: control channel connected\r\n", dev->con_handle);
                btstack_l2cap_create_channel(dev, PSM_HID_INTERRUPT, l2cap_max_mtu(), &hid_dev->l2cap_interrupt_cid);
            } else if (PSM_HID_INTERRUPT == psm) {
                BTSTACK_HOST_HID_TRACE("%d: interrupt channel connected\r\n", dev->con_handle);
            }
            break;
        case HCI_EVENT_DISCONNECTION_COMPLETE:
            BTSTACK_HOST_HID_TRACE("%d: disconnected\r\n", dev->con_handle);
            break;
        }
    }
}
```

``` c
const btstack_host_drv_op_t btstack_host_joycon_left_drv = {
    .cod            = 0x002508,
    .name           = LEFT_JOYCON_NAME,
    .connect        = btstack_host_hid_base_connect,
    .disconnect     = btstack_host_hid_base_disconnect,
    .packet_handler = btstack_host_joycon_packet_handler,
};

const btstack_host_drv_op_t btstack_host_joycon_right_drv = {
    .cod            = 0x002508,
    .name           = RIGHT_JOYCON_NAME,
    .connect        = btstack_host_hid_base_connect,
    .disconnect     = btstack_host_hid_base_disconnect,
    .packet_handler = btstack_host_joycon_packet_handler,
};
```

btstack_host_joycon_left_drv/btstack_host_joycon_right_drv is the driver for 2 devices.
btstack_host_hid_base_packet_handler is used for all hid devices, with the first parameter btstack_dev_t *, which is pointer to the device structure in the extension.
I register one common handler for HCI, l2cap, etc. So I need to get the con_handle in the common callback, and get the btstack_dev_t struct if it's for a dedicated device(according to the con_handle), and then call the handler in the driver of the device.

mringwal, 2021-12-21:
I see. When you get the L2CAP_EVENT_INCOMING_CONNECTION event, you already store (have to) store the l2cap_cid. Would it be possible to look up the btstack_dev_t via this l2cap_cid?

Vice versa, if you would have to support multiple services on the same remote device, you would not be able to match them with the hci con handle alone.

versaloon, 2021-12-25:
You mean I store both the con_handle and local_cid?
For HCI_EVENT_PACKET while issuing l2cap connectiong, con_handle is used because local_cid is not available.
For L2CAP_DATA_PACKET, local_cid is used.

Yeah, it's a solution based on current code.
If you recommend this, I will modify the extension code.

versaloon, 2021-12-25:
And not save one local_cid, because one device can have more than one l2cap connection, so need to store all the local_ids bound to the device.

mringwal, 2021-12-25:
Well. To use L2CAP channels, you need to store all their l2cap cids anyway, so please change your code to lookup a device based on these.

I don't know more about your architecture/design. Maybe you can have one object per profile or such. Then, you could register a packet handler per profile type and that profile could have a list of currently active instances. There, each would have e.g. L2CAP cids for HID.

Btw. We have HID Host and HID Device. Are you missing anything from these?

versaloon, 2021-12-29:
Can HID in btstack connecting more than one HID devices?
As I remember, HID host in btstack is singleton pattern.
I want to connect 2 HID devices(2 joycon from switch) at the same time.
So I implement this extension to use a device/driver model.
As the code I show above, btstack_host_joycon_left_drv and btstack_host_joycon_right_drv is the driver for the left and right joycon controller(HID device), which has no limitation on the number of devices connected.

versaloon, 2021-12-29:
I'm just preparing to make the extension open source. Maybe you can give some suggestions if any possible improvements after the code is released.

mringwal, 2021-12-29:
Sure. I'm curious to see your project. I'm closing this for now. Cheers

closed 2021-12-29 by mringwal