bluekitchen/btstack #393: esp32: fails to compile when using -O2

ricardoquesada, 2021-12-15:
**Describe the bug**
btstack fails to compile due to a warning that is treated as error when "Compiler Options" -> "Optimization Level" -> "Optimize for performance (-O2)" is chosen.

**To Reproduce**
Steps to reproduce the behavior:
1. Use ESP-IDF v4.4
2. `idf.py menuconfig`
3. Select "Compiler Options" -> "Optimization Level" -> "Optimize for performance (-O2)"
4. Compile: `idf.py build`

It fails with the following error:
```
/home/r/bin/esp/esp-idf/components/btstack/src/classic/hfp.c: In function 'parse_sequence':
/home/r/bin/esp/esp-idf/components/btstack/src/classic/hfp.c:1507:13: error: 'strncpy' output may be truncated copying 20 bytes from a string of length 99 [-Werror=stringop-truncation]
             strncpy((char *)hfp_connection->ag_indicators[hfp_connection->parser_item_index].name,  (char *)hfp_connection->line_buffer, HFP_MAX_INDICATOR_DESC_SIZE);
```

**Expected behavior**
It should compile Ok.
In fact, it only fails when using `idf.py build`... but compiles Ok with `make`.

The btstack code looks Ok. But a possible way to ignore this error is  by replacing `strncpy` with `memcpy`

**Environment:  (please complete the following information):**
 - Current BTstack branch: master (0f2a810173d1d70943d1c915bffd6f9b1171e8f6)
 - Bluetooth Controller ESP32
 - Remote device: N/A

**Additional context**
Add any other context about the problem here.

mringwal, 2021-12-21:
@ricardoquesada  thanks for reporting this. After some StackOverflow, it looks like the best approach for us will be to provide a custom strcpy function that can be then safely be used everywhere. CLion tells me there are 17 uses of strncpy in the core library.

solhuebner, 2022-01-07:
@mringwal that would be great as I have the same issue

mringwal, 2022-01-17:
Hi there. We've got a new btstack_strcpy(dest, dest_buffer_size, src) and it's used on the develop branch now.

closed 2022-01-17 by mringwal